realsenseai/realsense-ros #2670: [ROS1] Pointcloud topic not publishing anything ROS melodic

IsaiahOdunze, 2023-03-27:
I am trying to visualize pointcloud data using command "roslaunch realsense2_camera rs_camera.launch filters:=pointcloud" , yet nothing appears. I even try to echo the topic, but it seems like nothing is being published. Nothing shows up in RViz either. I am using a jetson nano to run this ROS package. On my Laptop running Ubuntu 20.04, the same camera works, and it even has the same output. Are there specific parameters needed to run the packages on a jetson nano or in Ubuntu 18.04?

Kernel Version: 4.9.140-tegra

Below is the output when running.

'''
titanauv@titanauv-desktop:/opt/ros/melodic/share/realsense2_camera/launch$ roslaunch rs_camera.launch filters:=pointcloud
... logging to /home/titanauv/.ros/log/b951de90-cce7-11ed-b4e2-00044beabd6b/roslaunch-titanauv-desktop-19325.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://titanauv-desktop:45187/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 83 lines cut ...]
[ INFO] [1679953119.161672574]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1679953124.234398678]: Done Setting Dynamic reconfig parameters.
[ INFO] [1679953124.235833806]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1679953124.238505622]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1679953124.238628020]: setupPublishers...
[ INFO] [1679953124.251494593]: Expected frequency for depth = 30.00000
[ INFO] [1679953124.390698104]: Expected frequency for color = 30.00000
[ INFO] [1679953124.486515096]: setupStreams...
 27/03 14:38:44,719 WARNING [547769409920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1679953124.769968661]: SELECTED BASE:Depth, 0
[ INFO] [1679953124.819065225]: RealSense Node Is Up!
[ WARN] [1679953124.841117162]: 
 27/03 14:38:44,842 WARNING [547769409920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/03 14:38:44,892 WARNING [547769409920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/03 14:38:44,943 WARNING [547769409920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/03 14:38:45,185 WARNING [547769409920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/03 14:38:45,235 WARNING [547769409920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/03 14:38:45,387 WARNING [547769409920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1679953150.160858702]: publishing unordered pointcloud.
[ WARN] [1679953150.813787104]: No stream match for pointcloud chosen texture Process - Color
'''

MartyG-RealSense, 2023-03-28:
Hi @IsaiahOdunze  The warning **No stream match for pointcloud chosen texture Process - Color** means that the color stream is dropping frames.

The warning **control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11** indicates that there is a communication problem between the camera and the computer / computing device.  

There are known issues with using the pointcloud filter on Jetson boards specifically.  A workaround in ROS1 is to use the **rs_rgbd.launch** launch file instead of rs_camera.launch.

You will first need to install support for an RGBD launch in Melodic by using the instruction below in the Ubuntu terminal.

`sudo apt-get install ros-melodic-rgbd-launch`

Then perform the rs_rgbd launch.

**roslaunch realsense2_camera rs_rgbd.launch**

With this launch file there is no need to enable the pointcloud filter with `filters:=pointcloud`.  It will automatically generate a point cloud that is published at **depth_image_proc**

IsaiahOdunze, 2023-03-28:
Thank you, that fixed my problem. I have one more question if you don't mind. Is there a way to configure the launch files to only publish pointcloud data, or do you always have to run the camera to run the point cloud? I see that Linux abstracts the realsense camera into three video files (/dev/video0 , /dev/video1, and /dev/video2) and I was wondering if it was possible to only publish point cloud data but not raw camera feed. I want to use the camera feed in another application and it would be easier if it wasn't publishing here. If this is not possible, then this issues branch can be closed. Thanks again!

MartyG-RealSense, 2023-03-29:
The /video channels will only stream raw camera data, unfortunately.

As you suggested, I will close this issue as only raw data can be provided by the /video channels.  Thanks!

closed 2023-03-29 by MartyG-RealSense