realsenseai/realsense-ros #2675: l515 align depth to color with different resolutions

robopassio, 2023-03-30:
Hi!
I'm using ROS topics to communicate with the camera.
L515 resolutions of rgb and depth channels are:
rgb: 1920x1080, 1280x720, 640x480
depth: 1024x768, 640x480

It seems that only 640x480 matches between the two channels.
If I want to use higher resolution of the rgb (1920x1080 or 1280x720) , any way to align the depth to color? and what is the resolution in the aligned result?

Thanks!

MartyG-RealSense, 2023-03-30:
Hi @robopassio  If different resolutions are used with depth-color alignment via the ROS wrapper's **align_depth** parameter then the RealSense SDK's **align processing block** will automatically adjust for differences in resolution and sensor field of view (FOV) size.

With alignment between depth and another stream type, the depth FOV size is resized to match the target stream's FOV size.  So if depth is aligned to color then the depth FOV takes on the color FOV size.

More information about the adjustments made automatically during alignment can be found at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align#overview

Examples of ROS launch instructions that cause aligned topics to be published are below.

**ROS1**
roslaunch realsense2_camera rs_camera.launch align_depth:=true

**ROS2**
ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true

robopassio, 2023-03-31:
Thanks @MartyG-RealSense!
Since the maximum resolution of the depth is 1024x768, if I select 1920x1080 for the color, does it mean that the depth will align with the color and result in a depth image size of 1920x1080? Please correct me if I am wrong.

MartyG-RealSense, 2023-03-31:
When aligning depth to color, the resultant aligned image depends mainly on the FOV size of the camera's RGB sensor and not the resolution size.   If the RGB sensor has a smaller FOV size than the depth sensor then the outer edges of the depth image can be excluded from the aligned image as though there is a television 'letterbox' around the border.  

L15 has a smaller RGB FOV on the vertical plane and the same horizontal FOV as depth, so I would expect an aligned depth to color image on L515 to have the same width as the depth image but have some of the depth image's vertical detail cut off from the aligned image on the top and bottom of it.

The depth FOV is 70° × 55° and the RGB FOV is 70° × 43°

robopassio, 2023-03-31:
Thanks @MartyG-RealSense, so I just need to run some thing like this
```
roslaunch realsense2_camera rs_rgbd.launch color_width:=1920 color_height:=1080 align_depth:=true
```
Then, I can output both depth and rgb at the same resolution at 1920x1080?

MartyG-RealSense, 2023-03-31:
There is not information available on how the RealSense SDK's **align processing block** makes the automatic adjustments for differences in resolution.  I believe though that it is more accurate to say that it is an adjustment of the pixels of the image and not a literal transformation of the depth data from 1080x720 resolution to 1920x1080.

closed 2023-03-31 by robopassio