realsenseai/realsense-ros #2679: [FATAL] [1680449427.937103336]: Call to publish() on an invalid Publisher

kevinDrawn, 2023-04-02:
when I run "roslaunch realsense2_camera rs_camera.launch",  I got an error like below.

```

[FATAL] [1680449427.936975991]: ASSERTION FAILED
	file = /opt/ros/melodic/include/ros/publisher.h
	line = 106
	cond = false
	message = 
[FATAL] [1680449427.937103336]: Call to publish() on an invalid Publisher
[FATAL] [1680449427.937195577]: 

[camera/realsense2_camera_manager-2] process has died [pid 15061, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/park/.ros/log/456dbffc-d16b-11ed-a6d2-024271229252/camera-realsense2_camera_manager-2.log].
log file: /home/park/.ros/log/456dbffc-d16b-11ed-a6d2-024271229252/camera-realsense2_camera_manager-2*.log

```
My camera model is D405
ROS version is melodic (I solved the compatibility problem with D405 and melodic version) 

How can I settle this problem?? 
This man had same error like me. But I can't figure out any solution.  

https://github.com/IntelRealSense/realsense-ros/issues/1068

MartyG-RealSense, 2023-04-03:
Hi @kevinDrawn  As the D405 does not have an RGB sensor and instead obtains RGB from the depth sensor , you could first test whether the launch is requesting the color stream and failing by setting color to false.

`roslaunch realsense2_camera rs_camera.launch enable_color:=false`

kevinDrawn, 2023-04-04:
Umm... Same error occurred!

`[FATAL] [1680580523.187965528]: ASSERTION FAILED
	file = /opt/ros/melodic/include/ros/publisher.h
	line = 106
	cond = false
	message = 
[FATAL] [1680580523.188092763]: Call to publish() on an invalid Publisher
[FATAL] [1680580523.190435270]: 

[camera/realsense2_camera-3] process has finished cleanly
log file: /home/park/.ros/log/7fea81c4-d29c-11ed-8586-88366cf8d049/camera-realsense2_camera-3*.log
[camera/realsense2_camera_manager-2] process has died [pid 10208, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/park/.ros/log/7fea81c4-d29c-11ed-8586-88366cf8d049/camera-realsense2_camera_manager-2.log].
log file: /home/park/.ros/log/7fea81c4-d29c-11ed-8586-88366cf8d049/camera-realsense2_camera_manager-2*.log
`

MartyG-RealSense, 2023-04-04:
It sounds like the RealSense ROS users in https://github.com/IntelRealSense/realsense-ros/issues/1068 did not find a solution either.

Do you still experience the error if you launch with the **rs_rgbd.launch** launch file instead using the instructions at https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-1468076630

kevinDrawn, 2023-04-07:
oh! That is correct for me. Thank you!

MartyG-RealSense, 2023-04-07:
You are very welcome  :)

closed 2023-04-13 by MartyG-RealSense