realsenseai/realsense-ros #2681: Will not install on Jammy, Humble

Russ76, 2023-04-04:
I used the vcpkg installer on Linux Jammy, on Intel i5 with 8Gig RAM.
Ros2 Humble
I thought Realsense2 installed OK, but Realsense-ros has trouble finding it and compiling.

I added this line to the cmakelists.txt under realsense2_camera
set(realsense2_DIR "~/vcpkg/packages/realsense2_x64-linux/share/realsense2")

greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ colcon build
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [1.19s]                     
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [0.21s]                           
--- stderr: realsense2_camera                             
gmake[2]: *** No rule to make target '/home/greg/vcpkg/packages/realsense2_x64-linux/lib/libusb-1.0.a', needed by 'librealsense2_camera.so'.  Stop.
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/realsense2_camera.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< realsense2_camera [1.91s, exited with code 2]
                                
Summary: 2 packages finished [3.32s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera

Installation was like this:

greg@greg-HP-EliteDesk-800-G2-DM-65W:~$ mkdir ros2_ws
greg@greg-HP-EliteDesk-800-G2-DM-65W:~$ cd ros2_ws
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ mkdir src
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ cd src
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws/src$ git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-development
cd ~/ros2_ws
Cloning into 'realsense-ros'...
remote: Enumerating objects: 14610, done.
remote: Counting objects: 100% (301/301), done.
remote: Compressing objects: 100% (169/169), done.
remote: Total 14610 (delta 183), reused 203 (delta 125), pack-reused 14309
Receiving objects: 100% (14610/14610), 31.98 MiB | 7.62 MiB/s, done.
Resolving deltas: 100% (9187/9187), done.
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ sudo apt-get install python3-rosdep -y
[sudo] password for greg: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-rosdep is already the newest version (0.22.2-1).
The following package was automatically installed and is no longer required:
  libllvm13
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ sudo rosdep init
ERROR: default sources list file already exists:
	/etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/greg/.ros/rosdep/sources.cache
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ rosdep install -i --from-path src --rosdistro humble --skip-keys=librealsense2 -y
#All required rosdeps installed successfully
greg@greg-HP-EliteDesk-800-G2-DM-65W:~/ros2_ws$ colcon build
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [7.92s]                     
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [1.18s]                           
--- stderr: realsense2_camera                               
gmake[2]: *** No rule to make target '/home/greg/vcpkg/packages/realsense2_x64-linux/lib/libusb-1.0.a', needed by 'librealsense2_camera.so'.  Stop.
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/realsense2_camera.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< realsense2_camera [30.9s, exited with code 2]

Summary: 2 packages finished [39.1s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera

for asking help in issues.txt

MartyG-RealSense, 2023-04-04:
Hi @Russ76  The librealsense SDK can be installed using vcpkg with the instructions in the link below but I would not recommend attempting to do so unless vcpkg has to absolutely be used as the vcpkg install method often has problems.  Also, these instructions only install librealsense and not the RealSense ROS wrapper.

https://github.com/IntelRealSense/librealsense#building-librealsense---using-vcpkg

I would recommend first installing the librealsense SDK from packages if you are using kernel 5.13 or 5.15:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

And then afterwards installing the RealSense ROS2 wrapper for Humble from source code using `colcon build` on Ubuntu 22.04 Jammy:

https://github.com/IntelRealSense/realsense-ros#step-3-install-intel-realsense-ros2-wrapper-from-sources

Russ76, 2023-04-05:
Thanks for the info, will try what you suggest! It seemed like Intel suggested the vcpkg method, even though I hadn't used it before.

MartyG-RealSense, 2023-04-10:
Hi @Russ76  Do you require further assistance with this case, please?  Thanks!

Russ76, 2023-04-10:
Last night I was able to install the Realsense2 software. The binaries worked fine, Thanks!

MartyG-RealSense, 2023-04-11:
You are very welcome. It's great to hear that you were successful  :)  As you have achieved a solution, I will close this case.  Thanks again!

closed 2023-04-11 by MartyG-RealSense