realsenseai/realsense-ros #2682: enable l515 depth filter

robopassio, 2023-04-04:
Hi,

Is there anyway to enable filtering of depth data using ros command to obtain similar result like [here](https://github.com/IntelRealSense/librealsense/tree/development/examples/post-processing)?
There is a line in rs_rgbd.launch file:
```
      <arg name="filters"                  value="$(arg filters)"/>
```
is this related to my scope or what should I do to enable?

robopassio, 2023-04-04:
Thanks @MartyG-RealSense! is there any way to configurate parameters of the `temporal` filter during running? which ros topic I can communicate to?

MartyG-RealSense, 2023-04-04:
The Intel RealSense developer who created the ROS1 wrapper advises at https://github.com/IntelRealSense/realsense-ros/issues/742#issuecomment-720295533 how to get a list of parameters for the temporal filter.

closed 2023-04-10 by robopassio