realsenseai/realsense-ros #2683: Realsense Ros not working

TheNemo05, 2023-04-04:
Hello there,
wrt to this repository https://github.com/leggedrobotics/realsense-ros-rsl for ROS. Stored it in SRC of Catkin. build catkin with catkin_make, it shows this errors. 
[image]

MartyG-RealSense, 2023-04-04:
Hi @TheNemo05  Have you performed the installation step `Clone the latest Intel® RealSense™ ROS from here into 'catkin_ws/src/'` as a RealSense ROS user at the link below had the same error when they did not carry out this step.

https://answers.ros.org/question/304600/realsense-ros-wont-start-with-roslaunch-realsense2_camera-rs_cameralaunch/

As the leggedrobotics GitHub's fork of the RealSense ROS wrapper contain exactly the same instructions as the official wrapper site, the instructions do not now make sense to me as they are asking users to clone the wrapper from the GitHub address of the official wrapper instead of the leggedrobotics GitHub.  The actual git clone address of the leggedrobotics version of the wrapper is:

`https://github.com/leggedrobotics/realsense-ros-rsl.git`

TheNemo05, 2023-04-05:
followed the same procedure, these packages worked previously but now they are not working

MartyG-RealSense, 2023-04-05:
I do not have further suggestions for installing this non-Intel repository fork unfortunately.

closed 2023-04-23 by MartyG-RealSense