realsenseai/realsense-ros #2685: Image data interrupt

zrd1234, 2023-04-07:
Hello, could you please tell me that when I use roslaunch realsense2_camera rs_camera.launch, echo topic /camera/color/image_raw, the data run will stop after a while

zrd1234, 2023-04-07:
> Hi @zrd1234 Are there any messages in the ROS terminal that may indicate a problem or does the color stream just stop publishing messages without explanation?
> 
> How long does it typically run correctly for before the data is interrupted?
After about 2-3 minutes of execution, stop the screenshot as shown below：
[image]

MartyG-RealSense, 2023-04-07:
I can see 'IMU' in the background of the image and Not eno...' - is that message "Not enough inliers"?

Is your RealSense camera a model that is equipped with an IMU and is it being used during your ROS session?

zrd1234, 2023-04-10:
> 我可以在图像的背景中看到“IMU”，而不是 eno...' - 该消息是“没有足够的内脏”吗？
> 
> 您的实感摄像头是否是配备 IMU 的型号，是否在 ROS 会话期间使用？

Hello, I have found some problems in the last two days of testing. When I open imu synchronization, the image data will be interrupted. When I only open the image, there is no problem.

zrd1234, 2023-04-10:
> > 我可以在图像的背景中看到“IMU”，而不是 eno...' - 该消息是“没有足够的内脏”吗？
> > 您的实感摄像头是否是配备 IMU 的型号，是否在 ROS 会话期间使用？
> 
> Hello, I have found some problems in the last two days of testing. When I open imu synchronization, the image data will be interrupted. When I only open the image, there is no problem.

[image]

MartyG-RealSense, 2023-04-10:
The error **while stop_streaming()** has only previously been reported once in the past at https://github.com/IntelRealSense/realsense-ros/issues/756 and that case did not provide a reason for the error, though the error messages that came before 'while stop_streaming()' indicated that there was likely a conflict with the Linux kernel.

Are you using the ROS instruction **unite_imu_method**.  If you are, does the while stop_streaming error still occur if you do not use unite_imu_method.

zrd1234, 2023-04-10:
> The error **while stop_streaming()** has only previously been reported once in the past at #756 and that case did not provide a reason for the error, though the error messages that came before [...]

When I am not using, there is no such error

MartyG-RealSense, 2023-04-10:
If you do not use unite_imu_method then you will still receive the IMU streams but they will be in two separate topics **accel** and **gyro** instead of combined into a single topic called **imu**.

If you need both IMU topics to be combined into the **imu** topic then you could try setting unite_imu_method to **copy** instead of linear_interpolation.

zrd1234, 2023-04-11:
> linear_interpolation

I found that I used to be copy so I always interrupt, so when I changed it to linear_interpolation it didn't interrupt

MartyG-RealSense, 2023-04-11:
Is your interrupt problem now solved if you use linear_interpolation instead of copy, please?

closed 2023-04-23 by MartyG-RealSense