realsenseai/realsense-ros #2686: Inconsistent Frame Latency On Realsense D455

edgex004, 2023-04-07:
We are testing a multi-camera setup where synchronized frame capture is not important, but accurate frame timestamps and latency are critical.

While testing the latency of the D435 and D455, we noticed some behavior that we did not expect.

For both D435 and D455, the average depth image latency is ~9ms and the average RGB image latency is ~66ms.

However, the standard deviation of the latency is much higher when running a single D455 camera. Please see the logs below, collected using the [latency test tool](https://github.com/IntelRealSense/realsense-ros#latency-test-tool-and-launch-file):

Note: the default node frame_latency.cpp was modified to change the default topic from color/image_raw to depth/image_rect_raw

D435: 

ros2 launch realsense2_camera rs_intra_process_demo_launch.py serial_no:='"829212071045"' enable_depth:=true depth_module.profile:=848x480x15 rs_baseline:=50 clip_distance:=2.0 enable_color:=false rgb_camera.profile:=1920x1080x15 >d435_829212071045_depth.log

[d435_829212071045_depth.log](https://github.com/IntelRealSense/realsense-ros/files/11181827/d435_829212071045_depth.log)

D455:

ros2 launch realsense2_camera rs_intra_process_demo_launch.py serial_no:='"141322252733"' enable_depth:=true depth_module.profile:=848x480x15 rs_baseline:=95 clip_distance:=4.0 enable_color:=false rgb_camera.profile:=1280x720x15 >d455_141322252733_depth.log

[d455_141322252733_depth.log](https://github.com/IntelRealSense/realsense-ros/files/11181828/d455_141322252733_depth.log)

Both the RGB and depth images exhibit this behavior.

Per the logs, please note that the D435 depth latency converges to ~9ms. The D455 depth latency number continues to bounce between -20ms and 20ms.

One of the more interesting observations we have had is that when 2 D455 are connected, only one D455 will exhibit this behavior. The other D455 latency measurement will converge to ~9ms as expected.

For environment:

We have reproduced on both a Lenovo P15 and Intel NUC.
OS: Ubuntu 22.04
ROS: Rolling
RealSense ROS v4.51.1
Built with LibRealSense v2.53.1
Running with LibRealSense v2.53.1

MartyG-RealSense, 2023-04-08:
Hi @edgex004  Whilst ROS Rolling is supported for source-code builds of the RealSense ROS wrapper, we do not recommend using it.  This is because Rolling is not a stable ROS version because it is updated regularly.  If you are using Ubuntu 22.04 then **Humble** will be the ideal choice for use with the wrapper.

[image]

May I also ask whether you also experience the delays when using the rs_launch.py file instead of the much less commonly used rs_intra_process_demo_launch.py 

`ros2 launch realsense2_camera rs_launch.py`

edgex004, 2023-04-08:
@MartyG-RealSense we have seen the same behavior using the default launch files, but in that case we were manually calculating the latency using a subscriber node.

We can re-test using Humble and report back.

edgex004, 2023-04-08:
At the moment, we suspect librealsense is the source of the issue, but we are collecting more logs from that level before filing a ticket with librealsense.

MartyG-RealSense, 2023-04-08:
Thanks very much for the updates.  I look forward to your next report.  Good luck!

edgex004, 2023-04-10:
@MartyG-RealSense I filed a ticket directly in the librealsense repo [here](https://github.com/IntelRealSense/librealsense/issues/11667).

As you can see in that ticket, the latency discrepancy can be seen directly in the librealsense metadata.

MartyG-RealSense, 2023-04-12:
If latency is low when both D435 and D455 are used but high when D455 on its own is used, I wonder whether **Global Camera Time** (which is enabled by default on 400 Series cameras) is producing a common timestamp for the two cameras, as described at https://github.com/IntelRealSense/librealsense/pull/3909

You could try disabling global time when using the D455 on its own to see whether it affects latency.  To do this, please add to the end of your D455 launch instruction `global_time_enabled:=false`

edgex004, 2023-04-13:
@MartyG-RealSense `src/global_timestamp_reader.cpp` is certainly what we are paying closest attention to. We don't notice any change with `global_time_enabled:=false`. The metadata still shows `"clock_domain":"global_time"`.

MartyG-RealSense, 2023-04-13:
The  ros2-development wrapper has a multiple camera launch file called **rs_multi_camera_launch.py** that launches 2 attached cameras.  

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_multi_camera_launch.py

Does the latency still occur in one of two D455 cameras plugged in if you launch them using this file?

edgex004, 2023-04-13:
@MartyG-RealSense: yes we have tried that launch file. It still results in one camera with stable latency and another with highly variable latency.

MartyG-RealSense, 2023-04-14:
When you launch cameras with separate launch instructions instead of a single one like rs_multi_camera_launch.py, have you tried inputting the command for each camera in separate ROS terminals instead of all in the same terminal?   A unique name for each camera should also be defined.

TERMINAL 1: D435: 

ros2 launch realsense2_camera rs_intra_process_demo_launch.py serial_no:='"829212071045"' enable_depth:=true depth_module.profile:=848x480x15 rs_baseline:=50 clip_distance:=2.0 enable_color:=false rgb_camera.profile:=1920x1080x15 **camera_name1:=D435 device_type:=d435**

TERMINAL 2: D455:

ros2 launch realsense2_camera rs_intra_process_demo_launch.py serial_no:='"141322252733"' enable_depth:=true depth_module.profile:=848x480x15 rs_baseline:=95 clip_distance:=4.0 enable_color:=false rgb_camera.profile:=1280x720x15 **camera_name2:=D455 device_type:=d455**

MartyG-RealSense, 2023-04-19:
There was another case this week where the cameras were getting mixed, and the solution at https://github.com/IntelRealSense/realsense-ros/issues/2701#issuecomment-1512806710 was to use device_type1 and device_type2 instead of device_type.

closed 2023-05-01 by MartyG-RealSense