realsenseai/realsense-ros #2690: How to alternate emitter pattern,toggled on/off on per-frame basis in d435i?

kao-yang, 2023-04-12:
How to alternate emitter pattern,toggled on/off on per-frame basis in d435i?Just like realsense_viewer.
[image]

I want to use both stero images while emitter off (for SLAM) and depth images while emitter on (depth images are more accurate). I found this option in realsense_view, so how do I implement it in realsense_ros?

MartyG-RealSense, 2023-04-12:
Hi @kao-yang  In the RealSense ROS wrapper you should be able to enable per-frame alternating of the emitter by adding the command below to the end of your ROS launch instruction.

`emitter_on_off:=true`

If you are using ROS1 (Kinetic, Melodic, Noetic) then you could also try enabling the emitter on-off during runtime after launch with the command below.

`rosrun dynamic_reconfigure dynparam set /camera0/stereo_module emitter_on_off 1`

MartyG-RealSense, 2023-04-18:
Hi @kao-yang  Do you require further assistance with this case, please?  Thanks!

kao-yang, 2023-04-19:
> Hi @kao-yang Do you require further assistance with this case, please? Thanks!

It works!Thanks for your help!

closed 2023-04-19 by kao-yang

MartyG-RealSense, 2023-04-19:
That's excellent news that you succeeded.  Thanks very much for the update!