realsenseai/realsense-ros #2692: ros2 launch error

shivangi-futuristicbots, 2023-04-13:
Hi **@MartyG-RealSense,** 

i need to resolve some error that encounters when i run the realsense camera:

[image]

shivangi-futuristicbots, 2023-04-14:
Hi @MartyG-RealSense, Thanks for the reply but i want to ask how to install the SDK 2.51.1 firware 5.13.0.50 , 
i followed  the guildlines provided in the [](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) ibut there is no  version specification provided during the time of installation.

MartyG-RealSense, 2023-04-14:
If you are able to use the RealSense Viewer tool then you can install a specific firmware version by downloading it as a .bin file from the list on the firmware releases page.

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

Then go to the More option near the top of the Viewer's options side-panel and select the Update Firmware option from its drop-down menu. This will cause a file selector window to pop up. Select the firmware .bin folder from the location on your computer that you downloaded it to and then the firmware update will commence.

[image]

If you are not able to use the RealSense Viewer tool then you can instead use the **rs-fw-update** tool with a specific firmware .bin file.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update