realsenseai/realsense-ros #2695: [ERROR] (2023-04-15 00:34:33.010) Rtabmap.cpp:1337::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 6278 is ignored!

jos1102, 2023-04-14:
This is my first post , my apologies if this message is not well curated and Thank you in advance for helping !! Your help is very much appreciated.

Camera Model | { D435i }
Operating System & Version | Linux (Ubuntu 18.04.5)
Platform | intel i5 , ASUS

After I followed all the instruction and used the command
`roslaunch realsense2_camera opensource_tracking.launch`

Issue: I am getting the path of the camera in RViz (even though very noisy  )and also the real time feed of the cam , but the problem is when I am moving the camera the map is not getting created.

I got the following errors in between

[ERROR] [1681498182.734894088]: PluginlibFactory: The plugin for class 'octomap_rviz_plugin/ColorOccupancyGrid' failed to load.  Error: According to the loaded plugin descriptions the class octomap_rviz_plugin/ColorOccupancyGrid with base class type rviz::Display does not exist. Declared types are  rtabmap_ros/Info rtabmap_ros/MapCloud rtabmap_ros/MapGraph rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu

[ERROR] (2023-04-15 00:20:34.870) Rtabmap.cpp:1337::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 941 is ignored!

At first terminal was like this:

[ INFO] [1681498185.343414978]: Odom: quality=59, std dev=0.038966m|0.015492rad, update time=0.087784s
[ INFO] [1681498185.425335127]: Odom: quality=55, std dev=0.025254m|0.015492rad, update time=0.079772s
[ INFO] [1681498185.511115599]: Odom: quality=56, std dev=0.052089m|0.017144rad, update time=0.083995s
[ INFO] [1681498185.601580416]: Odom: quality=38, std dev=0.055716m|0.018188rad, update time=0.084843s
[ INFO] [1681498185.709734859]: Odom: quality=43, std dev=0.047812m|0.013634rad, update time=0.102271s
[ INFO] [1681498185.724626852]: rtabmap (3): Rate=1.00s, Limit=0.000s, Conversion=0.0019s, RTAB-Map=0.1206s, Maps update=0.0002s pub=0.0000s (local map=1, WM=1)

in between I got this:

[ WARN] (2023-04-15 00:20:41.504) OdometryF2M.cpp:561::computeTransform() Registration failed: "Too low inliers after bundle adjustment: 16<20" (guess=xyz=-0.002924,0.004860,0.003118 rpy=-0.001109,0.002106,-0.001927)
[ WARN] (2023-04-15 00:20:41.505) OdometryF2M.cpp:309::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.002924,0.004860,0.003118 rpy=-0.001109,0.002106,-0.001927), trying again without a guess.
[ WARN] [1681498241.552027796]: Failed to meet update rate! Took 0.0068535310000000003797
[ WARN] (2023-04-15 00:20:41.593) OdometryF2M.cpp:551::computeTransform() Trial with no guess still fail.
[ WARN] (2023-04-15 00:20:41.593) OdometryF2M.cpp:561::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=66) between -1 and 556" (guess=xyz=-0.002924,0.004860,0.003118 rpy=-0.001109,0.002106,-0.001927)

After some time the terminal was like this:

[ INFO] [1681498376.254611626]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.109153s
[ WARN] (2023-04-15 00:22:56.372) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=49) between -1 and 1696"
[ INFO] [1681498376.373308583]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.117213s
[ERROR] (2023-04-15 00:22:56.375) Rtabmap.cpp:1337::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 4082 is ignored!

jos1102, 2023-04-14:
I followed this https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i , I am using ROS1 melodic

MartyG-RealSense, 2023-04-14:
Hi @jos1102   Please try adding the commands below to the end of your opensource_tracking.launch instruction to enable the Accel and Gyro IMU topics, which are disabled by default in the RealSense ROS wrapper.

`enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

jos1102, 2023-04-15:
Thank you for you response , after selecting map cloud , 

**New Issue:** the map is getting created for some time and then after it stops creating and just shows the real time feed over the build map

when i type is command : enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation

it is saying that this command does not exist.

MartyG-RealSense, 2023-04-15:
The commands are placed on the end of the roslaunch instruction.

`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

jos1102, 2023-04-15:
yes I tried the command at the start the terminal was like this:
```

 WARN] [1681549764.442143706]: Failed to meet update rate! Took 0.0073949760000000001534
[ INFO] [1681549764.483520106]: Odom: quality=310, std dev=0.000430m|0.013634rad, update time=0.078764s
[ INFO] [1681549764.552390559]: Odom: quality=316, std dev=0.000680m|0.014651rad, update time=0.065002s
[ INFO] [1681549764.618382561]: Odom: quality=297, std dev=0.000578m|0.014651rad, update time=0.063974s
[ INFO] [1681549764.688700744]: Odom: quality=314, std dev=0.000346m|0.014651rad, update time=0.066467s
[ INFO] [1681549764.752931227]: Odom: quality=331, std dev=0.000687m|0.013634rad, update time=0.062592s
[ INFO] [1681549764.822047062]: Odom: quality=314, std dev=0.000779m|0.014651rad, update time=0.065356s
[ INFO] [1681549764.899012030]: Odom: quality=306, std dev=0.000327m|0.014651rad, update time=0.075215s
[ INFO] [1681549764.978757806]: Odom: quality=310, std dev=0.000475m|0.015089rad, update time=0.074290s
[ INFO] [1681549765.059422969]: Odom: quality=299, std dev=0.000672m|0.015089rad, update time=0.078700s
[ INFO] [1681549765.151615691]: Odom: quality=276, std dev=0.000571m|0.015089rad, update time=0.084797s
[ INFO] [1681549765.222212796]: Odom: quality=286, std dev=0.000285m|0.012320rad, update time=0.069038s
[ INFO] [1681549765.297857288]: Odom: quality=300, std dev=0.000342m|0.014170rad, update time=0.073449s
[ INFO] [1681549765.370657489]: Odom: quality=295, std dev=0.000330m|0.013027rad, update time=0.071301s
[ INFO] [1681549765.450644750]: Odom: quality=286, std dev=0.000486m|0.014651rad, update time=0.077138s
[ INFO] [1681549765.483734088]: rtabmap (3): Rate=1.00s, Limit=0.000s, Conversion=0.0019s, RTAB-Map=0.1104s, Maps update=0.0003s pub=0.0000s (local map=1, WM=1)
[ INFO] [1681549765.534105387]: Odom: quality=273, std dev=0.000310m|0.014170rad, update time=0.080926s
[ INFO] [1681549765.604757248]: Odom: quality=302, std dev=0.000491m|0.013634rad, update time=0.068521s
[ INFO] [1681549765.680495296]: Odom: quality=283, std dev=0.000335m|0.013027rad, update time=0.074020s
[ INFO] [1681549765.749382268]: Odom: quality=278, std dev=0.000352m|0.013634rad, update time=0.067046s
[ INFO] [1681549765.823835776]: Odom: quality=279, std dev=0.000459m|0.013027rad, update time=0.072483s
[ INFO] [1681549765.894372886]: Odom: quality=255, std dev=0.000628m|0.014651rad, update time=0.068108s
```

**after some time:**
```
[ WARN] (2023-04-15 14:40:44.156) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=4) between -1 and 746"
[ INFO] [1681549844.156535992]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.067347s
[ERROR] (2023-04-15 14:40:44.158) Rtabmap.cpp:1337::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 2369 is ignored!
[ INFO] [1681549844.158761064]: rtabmap (23): Rate=1.00s, Limit=0.000s, Conversion=0.0017s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=4, WM=4)
[ WARN] (2023-04-15 14:40:44.225) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=4) between -1 and 747"
[ INFO] [1681549844.226255647]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.067602s
[ WARN] (2023-04-15 14:40:44.296) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=9) between -1 and 748"
[ INFO] [1681549844.296897595]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.068668s
[ WARN] (2023-04-15 14:40:44.372) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=4) between -1 and 749"
[ INFO] [1681549844.372962582]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.074067s
[ WARN] (2023-04-15 14:40:44.438) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=6) between -1 and 750"
[ INFO] [1681549844.439092076]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.064556s
[ WARN] (2023-04-15 14:40:44.504) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=2) between -1 and 751"
[ INFO] [1681549844.505057111]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.064491s
[ WARN] (2023-04-15 14:40:44.572) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=10) between -1 and 752"
```

**warnings got more common**

jos1102, 2023-04-15:
And I wanted to know if this is possible https://www.youtube.com/watch?v=0L_K3DQey9I

the github link for the video.

https://github.com/indranildchandra/Intel-Realsense-SLAM-Robotics

Is the above one better than  https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i ?

issue I was facing to implement the second one is , I am using ROS1 melodic and in one command he is telling to downlownd ros-realsense official which is in ros2 . Is there any way through this.

Thank you

MartyG-RealSense, 2023-04-15:
Advice about the **Not enough inliers** warning is provided by the developer who created the RealSense ROS wrapper at https://github.com/IntelRealSense/realsense-ros/issues/1080 who advises adjusting the position of the camera to see whether the inlier warning goes away.

I do not know about the Intel-RealSense-SLAM-Robotics project and so cannot make a comparison with the D435i SLAM guide.  

The official wrapper page has the ROS2 wrapper on it because the ROS1 version of the wrapper is no longer updated.  You can still access the offical ROS1 wrapper on the **ros1-legacy** GiHub branch at the link below though.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

May I confirm that you have built the ROS1 wrapper please and not the ROS2 wrapper.  The opensource_tracking.launch file will not work with the ROS2 wrapper.

jos1102, 2023-04-15:
Sorry to bother you regarding the other project , he was not active that's why asking you.

for this one:  https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i  I have downloaded ROS1 wrapper

and for the other one I think ROS2 wrapper got downloaded  , it is possible to use ROS1 wrapper instead of ROS2 wrapper and everything else same from other project?

Thank you.

MartyG-RealSense, 2023-04-15:
SLAM is more difficult to implement on ROS2, so I would recommend using it in ROS1.

If you built the ROS2 wrapper from source with colcon build then just delete the entire **/ros2_ws/src/** catkin workspace folder to remove the ROS2 wrapper and then install the ROS1 wrapper.

If you are using Ubuntu 18.04 and want to try a ROS2 wrapper then it would be best to use the older **ros2-legacy** wrapper with ROS2 **Dashing**.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-legacy

MartyG-RealSense, 2023-04-20:
Hi @jos1102   Do you require further assistance with this case, please?  Thanks!

jos1102, 2023-04-20:
Thank You very much for your help, you may close the thread.

closed 2023-04-20 by jos1102

MartyG-RealSense, 2023-04-20:
You are very welcome, @jos1102 - thanks very much for the update!

ranjithlesoko, 2024-01-10:
@MartyG-RealSense  same error

MartyG-RealSense, 2024-01-10:
Hi @ranjithlesoko  Can you create a new issue and link it to this one, please?  Thanks!

ranjithlesoko, 2024-01-10:
solved

MartyG-RealSense, 2024-01-10:
@ranjithlesoko  That's great to hear that you achieved a solution.  Thanks very much for the update!

pseelam02, 2025-10-24:
What was the solution to this error