realsenseai/realsense-ros #2696: realsense2_camera not building because it cannot find librealsense2

Brac24, 2023-04-16:
I built librealsense2 on Ubuntu 18 for Jetson Nano from source because I want to use the native backend. I also have to build realsense2_camera for ROS 2 Foxy. But I get the following error when it tries to build realsense2_camera
[image]
There is a similar issue here unrelated to realsense https://github.com/colcon/colcon-core/issues/76

Is there a special way I need to build librealsense2 to be used with ROS 2? I can colcon build librealsense2 and then realsense2_camera will also build but I don't think that will build librealsense2 with native backend which I would like. Do I have to colcon build librealsense2 to be able to build realsense2_camera?

MartyG-RealSense, 2023-04-16:
Hi @Brac24  If you want to use the native backend then you will need to build librealsense from source code.  That will mean building the ROS2 wrapper from source code too afterwards.

The wrapper documentation recommends that Foxy is used with Ubuntu 20.04 rather than 18.04.  

[image]

For 18.04, ROS2 Dashing or Eloquent will be an appropriate choice to use with the **ros2-development** wrapper.

[image]

Brac24, 2023-04-16:
Yes I do understand but I am trying to use ROS 2 Foxy. I believe I would get the same error even if I choose a different ROS distribution. 
My question is why does the ROS 2 wrapper search for librealsense2 within the ROS2 workspace instead of the linux library paths such as /usr/local/lib or something?

MartyG-RealSense, 2023-04-17:
I see that your colcon build command is **colcon build --packages-select** and then a list of packages.  There is no need to specify packages when building the ROS wrapper from source code.  The build command is simply **colcon build**

https://github.com/IntelRealSense/realsense-ros#option-2-install-from-source

It looks as though the installation is looking for packages in the catkin workspace because of your --packages-select command, which is not required.

closed 2023-04-29 by MartyG-RealSense