realsenseai/realsense-ros #2697: Managing the Enabling and Disabling of Published Topics.

Llavrih, 2023-04-17:
Is there a method to selectively disable topics that are not required for my specific application while using ROS Noetic or potentially transitioning to ROS2? I am utilizing two D435f cameras, and the only topics necessary for my use case are /cam_1/depth/color/points and /cam_2/depth/color/points. The primary objective of this modification is to minimize CPU consumption.

MartyG-RealSense, 2023-04-17:
Hi @Llavrih  It is possible to **unsubscribe** from topics.

As you likely need depth and color enabled to publish the depth/color/points topics, and the Infrared and IMU topics are aready disabled by default in the ROS wrapper, there is not likely to be a significant gain to be made from unsubscribing from the remaining published topics, such as /diagnostics

You could reduce CPU usage if you had an Nvidia Jetson computing board, as RealSense supports acceleration of point cloud generation with **CUDA** support for the Nvidia graphics GPU on the Jetson, offloading work from the CPU onto the GPU.

Llavrih, 2023-04-17:
At the moment, I'm subscribed to only those two topics. I'm working with Python and Open3D to process point clouds, and I'm particularly interested in reducing my CPU usage. Unfortunately, Open3D doesn't support working on CUDA, so I can't take advantage of GPU acceleration.

Llavrih, 2023-04-17:
Okay thanks for help @MartyG-RealSense , I'm gonna try this.

MartyG-RealSense, 2023-04-23:
Hi @Llavrih  Do you require further assistance with this case, please?  Thanks!

Llavrih, 2023-04-23:
Hi @MartyG-RealSense. We can close the topic. However, this would be a nice feature especially for less powerful computers.

MartyG-RealSense, 2023-04-23:
I researched the issue of unsubscription further but was unable to find a simple method of unsubscribing, as it apparently usually involves writing a ROS node script in C++ or Python to do so.  For example:

https://itecnote.com/tecnote/c-unsubscribe-from-a-topic-and-subscribe-from-a-new-one-runtime-c-ros/

It seems that disabling a topic's **display** in RViz by un-ticking a checkbox also unsubscribes it.

http://docs.ros.org/en/fuerte/api/rviz/html/classrviz_1_1Display.html

closed 2023-04-23 by MartyG-RealSense