realsenseai/realsense-ros #2699: [d435i]Got nothing in the topic /camera/depth/image_rect_raw

Spring-Lee-cpu, 2023-04-17:
I'm using D435i to obtain depth information for my current work.

In my launch file ,I enabled depth measurement.

[image]

And I have the topic /camera/depth/image_rect_raw.

[image]

But it seems that there's noting carried in this topic .When I do `rostopic echo /camera/depth/image_rect_raw` ,nothing is outputted onto the screen .When I do `rocbag record -O test /camera/depth/image_rect_raw` ,no messages are recorded.I even can't see the depth image with `image_view`.

Any other msgs such as /color and /infra are correct.

I'd like to know why and HOW can I fix it.Thanks a lot!

Spring-Lee-cpu, 2023-04-21:
I'm using ROS1,and I just config as you said
but it still doesn't work
[image]

Spring-Lee-cpu, 2023-04-21:
This is my launch file ,is there anything wrong?
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="-1"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default="copy"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>

[... 56 lines cut ...]

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="wait_for_device_timeout"  value="$(arg wait_for_device_timeout)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
</launch>

Spring-Lee-cpu, 2023-04-21:
This is my launch file.Is there anything wrong?
------------------------------------------------------------------
[image]
[image]
[image]

MartyG-RealSense, 2023-04-21:
Setting Accel_FPS to 250 might cause a configuration failure if your camera was manufactured after 2022.  The 250 speed was originally supported for Accel in RealSense 400 Series cameras, but in 2022 onwards the D435i and D455 had their IMU component changed, and they then supported **200** accel instead of 250.

Spring-Lee-cpu, 2023-04-22:
Thank you for your reminder! It's a detail that deserves attention.
However ,I don't think it's the reason for my problem because I'm using the D435i manufactured before 2022.
My problem have been solved after I changed the parameters about the width of depth image to 848.
Thank you for your reply！

closed 2023-04-22 by Spring-Lee-cpu

MartyG-RealSense, 2023-04-22:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update!