realsenseai/realsense-ros #2702: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE

RoeschFlo, 2023-04-18:
Hello together,
i cant launch:  roslaunch realsense2_camera rs_camera.launch
OS: Ubuntu 20.04, ROS: noetic 

I couldnt find information with google that helped me 

florian@BBC:~/flo_ba_thesis_workspace$ source devel/setup.bash
florian@BBC:~/flo_ba_thesis_workspace$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/florian/.ros/log/0d5b1b50-ddcd-11ed-b0fd-d9a26b1b4e59/roslaunch-BBC-163507.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BBC:41295/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [163521]
process[camera/realsense2_camera-2]: started with pid [163522]
[ INFO] [1681812591.910246301]: Initializing nodelet with 32 worker threads.
[ INFO] [1681812591.974152242]: RealSense ROS v2.3.0
[ INFO] [1681812591.974173652]: Built with LibRealSense v2.50.0
[ INFO] [1681812591.974180912]: Running with LibRealSense v2.50.0
[ INFO] [1681812591.999198842]:  
[ INFO] [1681812592.173861514]: Device with serial number 750612070760 was found.

[ INFO] [1681812592.173879493]: Device with physical ID 6-3-2 was found.
[ INFO] [1681812592.173891473]: Device with name Intel RealSense D435 was found.
[ INFO] [1681812592.174082902]: Device with port number 6-3 was found.
[ INFO] [1681812592.174092902]: Device USB type: 3.2
[ INFO] [1681812592.174885998]: getParameters...
[ INFO] [1681812592.195859851]: setupDevice...
[ INFO] [1681812592.195873061]: JSON file is not provided
[ INFO] [1681812592.195881471]: ROS Node Namespace: camera
[ INFO] [1681812592.195889120]: Device Name: Intel RealSense D435
[ INFO] [1681812592.195897250]: Device Serial No: 750612070760
[ INFO] [1681812592.195904720]: Device physical port: 6-3-2
[ INFO] [1681812592.195910980]: Device FW version: 05.14.00.00
[ INFO] [1681812592.195919050]: Device Product ID: 0x0B07
[ INFO] [1681812592.195926970]: Enable PointCloud: Off
[ INFO] [1681812592.195934270]: Align Depth: Off
[ INFO] [1681812592.195941100]: Sync Mode: Off
[ INFO] [1681812592.195959700]: Device Sensors: 
[ INFO] [1681812592.278027710]: Stereo Module was found.
[ INFO] [1681812592.288634191]: RGB Camera was found.
[ INFO] [1681812592.288668441]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1681812592.288683051]: num_filters: 0
[ INFO] [1681812592.288690451]: Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/florian/flo_ba_thesis_workspace/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-1] process has died [pid 163521, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/florian/.ros/log/0d5b1b50-ddcd-11ed-b0fd-d9a26b1b4e59/camera-realsense2_camera_manager-1.log].
log file: /home/florian/.ros/log/0d5b1b50-ddcd-11ed-b0fd-d9a26b1b4e59/camera-realsense2_camera_manager-1*.log
^C[camera/realsense2_camera-2] killing on exit
[ WARN] [1681812599.165011605]: Couldn't find service realsense2_camera_manager/unload_nodelet, perhaps the manager is already shut down
shutting down processing monitor...
... shutting down processing monitor complete

MartyG-RealSense, 2023-04-18:
Hi @RoeschFlo  Please try installing ddynamic_reconfigure for Noetic using the instruction below to see whether it resolves the error

`apt-get update apt-get install ros-noetic-ddynamic-reconfigure`

RoeschFlo, 2023-04-18:
thank you for your answer :)
I am not sure, because we already have ddynamic-reconfigure as a ros package in our workspace:

florian@BBC:~/flo_ba_thesis_workspace$ catkin build ddynamic_reconfigure 
--------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/florian/flo_ba_thesis_workspace
--------------------------------------------------------------------------
Build Space:        [exists] /home/florian/flo_ba_thesis_workspace/build
Devel Space:        [exists] /home/florian/flo_ba_thesis_workspace/devel
Install Space:      [unused] /home/florian/flo_ba_thesis_workspace/install
Log Space:          [exists] /home/florian/flo_ba_thesis_workspace/logs
Source Space:       [exists] /home/florian/flo_ba_thesis_workspace/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
--------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------
[build] Found 89 packages in 0.0 seconds.                                      
[build] Package table is up to date.                                           
Starting  >>> ddynamic_reconfigure                                             
Finished  <<< ddynamic_reconfigure                [ 0.2 seconds ]              
[build] Summary: All 1 packages succeeded!                                     
[build]   Ignored:   88 packages were skipped or are skiplisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 0.3 seconds total.                                            
florian@BBC:~/flo_ba_thesis_workspace$ source devel/setup.bash 
florian@BBC:~/flo_ba_thesis_workspace$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/florian/.ros/log/d53d7a58-ddd4-11ed-b0fd-d9a26b1b4e59/roslaunch-BBC-164966.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BBC:39855/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
[... 92 lines cut ...]
[ INFO] [1681814170.583862332]: Device with physical ID 6-3-2 was found.
[ INFO] [1681814170.583875712]: Device with name Intel RealSense D435 was found.
[ INFO] [1681814170.584256600]: Device with port number 6-3 was found.
[ INFO] [1681814170.584278850]: Device USB type: 3.2
[ INFO] [1681814170.585552973]: getParameters...
[ INFO] [1681814170.605754736]: setupDevice...
[ INFO] [1681814170.605766196]: JSON file is not provided
[ INFO] [1681814170.605773796]: ROS Node Namespace: camera
[ INFO] [1681814170.605783616]: Device Name: Intel RealSense D435
[ INFO] [1681814170.605793956]: Device Serial No: 750612070760
[ INFO] [1681814170.605802846]: Device physical port: 6-3-2
[ INFO] [1681814170.605812416]: Device FW version: 05.14.00.00
[ INFO] [1681814170.605820866]: Device Product ID: 0x0B07
[ INFO] [1681814170.605827796]: Enable PointCloud: Off
[ INFO] [1681814170.605835636]: Align Depth: Off
[ INFO] [1681814170.605841716]: Sync Mode: Off
[ INFO] [1681814170.605863526]: Device Sensors: 
[ INFO] [1681814170.687655252]: Stereo Module was found.
[ INFO] [1681814170.697569670]: RGB Camera was found.
[ INFO] [1681814170.697591660]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1681814170.697603950]: num_filters: 0
[ INFO] [1681814170.697613260]: Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/florian/flo_ba_thesis_workspace/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-2] process has died [pid 164991, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/florian/.ros/log/d53d7a58-ddd4-11ed-b0fd-d9a26b1b4e59/camera-realsense2_camera_manager-2.log].
log file: /home/florian/.ros/log/d53d7a58-ddd4-11ed-b0fd-d9a26b1b4e59/camera-realsense2_camera_manager-2*.log

RoeschFlo, 2023-04-18:
Hi MartyG-RealSense, 

i allready have the latest version of ddynamic_reconfigure.

florian@BBC:~/flo_ba_thesis_workspace$ sudo apt-get install ros-noetic-ddynamic-reconfigure
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-noetic-ddynamic-reconfigure is already the newest version (0.3.2-1focal.20230215.221255).
The following packages were automatically installed and are no longer required:
  libxmlb1 python-dateutil python-pyparsing python-six python-yaml
  ubuntu-advantage-desktop-daemon
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 43 not upgraded.

MartyG-RealSense, 2023-04-18:
Which method did you use to install the ROS wrapper, please?  Did you install librealsense and the wrapper together at the same time from packages with Method 1 or install librealsense and the wrapper separately, building the wrapper from source code with Method 2

RoeschFlo, 2023-04-18:
I wasnt sure so i tried to remove the package and there was nothing to remove, so it must be the wrapper with source code

sudo apt remove ros-noetic-realsense2-camera
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'ros-noetic-realsense2-camera' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libxmlb1 python-dateutil python-pyparsing python-six python-yaml
  ubuntu-advantage-desktop-daemon
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 43 not upgraded.

MartyG-RealSense, 2023-04-18:
If you do not need the RealSense Viewer tool then a clean and easy way to install librealsense and the ROS wrapper from packages would be:

1.  Ensure that all RealSense related packages are uninstalled by using the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

2.  Check whether you have a ROS catkin workspace folder called **catkin_ws/src**.  If you do then you have a source code build of the ROS wrapper.  Delete the entire catkin_ws/src folder to remove the wrapper.

3.  Use the instruction below to install librealsense and the wrapper together at the same time from packages.  You do not need to manually install ddynamic_reconfigure with this method.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

RoeschFlo, 2023-04-18:
thanks for the good suggestion, but I would like to keep the wrapper and try to not to modify the code because its a larger project and if i chance something like this other users might get problems, because they have to chance everything on their systems as well

RoeschFlo, 2023-04-18:
something was definitive installed before: 

florian@BBC:~$ sudo find / -name librealsense*
find: ‘/run/user/1001/doc’: Permission denied
find: ‘/run/user/1001/gvfs’: Permission denied
find: ‘/run/user/121/gvfs’: Permission denied
/var/lib/dkms/librealsense2-dkms
/var/lib/dpkg/info/librealsense2-dev:amd64.md5sums
/var/lib/dpkg/info/librealsense2-dev:amd64.list
/var/lib/dpkg/info/librealsense2:amd64.md5sums
/var/lib/dpkg/info/librealsense2:amd64.shlibs
/var/lib/dpkg/info/librealsense2:amd64.symbols
/var/lib/dpkg/info/librealsense2:amd64.triggers
/var/lib/dpkg/info/librealsense2-dkms.md5sums
/var/lib/dpkg/info/librealsense2-dkms.postinst
/var/lib/dpkg/info/librealsense2-dkms.postrm
/var/lib/dpkg/info/librealsense2-dkms.prerm
/var/lib/dpkg/info/librealsense2-dkms.list
/var/lib/dpkg/info/librealsense2:amd64.list
/var/lib/dpkg/info/librealsense2-udev-rules:amd64.md5sums
/var/lib/dpkg/info/librealsense2-udev-rules:amd64.postinst
/var/lib/dpkg/info/librealsense2-udev-rules:amd64.list
/var/lib/dpkg/info/librealsense2-gl:amd64.md5sums
/var/lib/dpkg/info/librealsense2-gl:amd64.shlibs
/var/lib/dpkg/info/librealsense2-gl:amd64.symbols
/var/lib/dpkg/info/librealsense2-gl:amd64.triggers
/var/lib/dpkg/info/librealsense2-gl:amd64.list
/var/lib/dpkg/info/librealsense2-net:amd64.md5sums
/var/lib/dpkg/info/librealsense2-net:amd64.shlibs
/var/lib/dpkg/info/librealsense2-net:amd64.triggers
/var/lib/dpkg/info/librealsense2-net:amd64.list
/var/lib/dpkg/info/librealsense2-utils:amd64.md5sums
/var/lib/dpkg/info/librealsense2-utils:amd64.list
/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_InRelease
/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages
/var/cache/apt/archives/librealsense2-udev-rules_2.53.1-0~realsense0.8250_amd64.deb
/var/cache/apt/archives/librealsense2-dev_2.53.1-0~realsense0.8250_amd64.deb
/var/cache/apt/archives/librealsense2_2.53.1-0~realsense0.8250_amd64.deb
/var/cache/apt/archives/librealsense2-net_2.53.1-0~realsense0.8250_amd64.deb
/var/cache/apt/archives/librealsense2-gl_2.53.1-0~realsense0.8250_amd64.deb
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/include/librealsense2
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/include/librealsense2-gl
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/lib/librealsense2.so
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/lib/librealsense2.so.2.42
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/lib/librealsense2-gl.so
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/lib/librealsense-file.a
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/lib/librealsense2-gl.so.2.42.0
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/lib/librealsense2-gl.so.2.42
/home/vision/Work/coral/foxy/image/root/pyrealsense_install/lib/librealsense2.so.2.42.0
/home/vision/Work/coral/foxy/image/root/librealsense_install
/home/vision/Work/coral/foxy/image/root/librealsense_install/include/librealsense2
/home/vision/Work/coral/foxy/image/root/librealsense_install/include/librealsense2-gl
/home/vision/Work/coral/foxy/image/root/librealsense_install/lib/librealsense2.so
/home/vision/Work/coral/foxy/image/root/librealsense_install/lib/librealsense2.so.2.42
/home/vision/Work/coral/foxy/image/root/librealsense_install/lib/librealsense2-gl.so
/home/vision/Work/coral/foxy/image/root/librealsense_install/lib/librealsense-file.a
/home/vision/Work/coral/foxy/image/root/librealsense_install/lib/librealsense2-gl.so.2.42.0
/home/vision/Work/coral/foxy/image/root/librealsense_install/lib/librealsense2-gl.so.2.42
/home/vision/Work/coral/foxy/image/root/librealsense_install/lib/librealsense2.so.2.42.0
/home/vision/Work/AtWork/catkin_ws/src/ohm_atwork/sw_driver/sensors/realsense-ros/realsense2_camera/librealsense2_xenial.rdmanifest
/home/vision/Work/AtWork/catkin_ws/src/ohm_atwork/sw_driver/sensors/realsense-ros/realsense2_camera/librealsense2_bionic.rdmanifest
/home/florian/atwork_repo_orginal/ohm_atwork/sw_driver/sensors/realsense-ros/realsense2_camera/librealsense2_xenial.rdmanifest
/home/florian/atwork_repo_orginal/ohm_atwork/sw_driver/sensors/realsense-ros/realsense2_camera/librealsense2_bionic.rdmanifest
/home/florian/flo_ba_thesis_workspace_cpy18.4/realsense-ros/realsense2_camera/librealsense2_xenial.rdmanifest
/home/florian/flo_ba_thesis_workspace_cpy18.4/realsense-ros/realsense2_camera/librealsense2_bionic.rdmanifest
/home/florian/flo_ba_thesis_workspace/src/sw_driver/sensors/realsense-ros/realsense2_camera/librealsense2_xenial.rdmanifest
/home/florian/flo_ba_thesis_workspace/src/sw_driver/sensors/realsense-ros/realsense2_camera/librealsense2_bionic.rdmanifest
/home/florian/flo_ba_thesis_workspace/devel/lib/librealsense2_camera.so
/home/florian/flo_ba_thesis_workspace/devel/.private/realsense2_camera/lib/librealsense2_camera.so
/home/florian/flo_ba_thesis_workspace_orginal/src/realsense-ros/realsense2_camera/librealsense2_xenial.rdmanifest
[... 22 lines cut ...]
/opt/ros/galactic/lib/librealsense2_camera_msgs__rosidl_typesupport_fastrtps_c.so
/opt/ros/galactic/lib/librealsense2_camera_msgs__rosidl_typesupport_introspection_cpp.so
/opt/ros/noetic/include/librealsense2
/opt/ros/noetic/lib/x86_64-linux-gnu/librealsense2.so
/opt/ros/noetic/lib/x86_64-linux-gnu/librealsense2.so.2.50.0
/opt/ros/noetic/lib/x86_64-linux-gnu/librealsense-file.a
/opt/ros/noetic/lib/x86_64-linux-gnu/librealsense2.so.2.50
/usr/include/librealsense2
[... cut ...]
/usr/lib/x86_64-linux-gnu/librealsense2.so
/usr/lib/x86_64-linux-gnu/librealsense2-net.so.2.53
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.53
/usr/lib/x86_64-linux-gnu/librealsense-file.a
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.53.1
/usr/share/librealsense2
/usr/share/doc/librealsense2
/usr/share/doc/librealsense2-net
/usr/share/doc/librealsense2-udev-rules
/usr/share/doc/librealsense2-dev
/usr/share/doc/librealsense2-dkms
/usr/share/doc/librealsense2-utils
/usr/share/doc/librealsense2-gl

RoeschFlo, 2023-04-18:
I will make changes at the installation. If i change the wrapper the person who uses my source code will have problems on his  not updated system.

MartyG-RealSense, 2023-04-18:
If you change the librealsense version to 2.45.0 then you will need to build the 2.3.0 wrapper from source code again afterwards.  This is because the wrapper must be rebuilt every time that the librealsense version is changed.

RoeschFlo, 2023-04-19:
Thanks a lot.
 i have talked with my teamleader and the way with sudo apt-get install ros-noetic-realsense2-camera is okay to use and works. 
(there should be funktion in github were you can spend people that help a coffee on voluntary basis)

MartyG-RealSense, 2023-04-25:
Hi @RoeschFlo  Do you require further assistance with this case, please?  Thanks!

RoeschFlo, 2023-04-25:
no 
right now everthing seems to work. thank you very much

closed 2023-04-25 by RoeschFlo