realsenseai/realsense-ros #2703: Can D435i filter out 850nm infrared light?

kao-yang, 2023-04-21:
I am using two camera:Realsense D435i which closed emitter and Orbbec Gemini2 which emitting infrared structured light.
When I turn on two cameras,D435i's image effected by Gemini2's infrared structured light.
[image]
But after a few seconds, the infrared structured light in D435i's image disappear.
[image]
It seems that D435i can filter infrared structured light from Gemini2.Is that right?How should I set parameters to filter infrared structured light from Gemini2 at 850nm?

MartyG-RealSense, 2023-04-21:
Hi @kao-yang  If a RealSense camera's IR emitter is disabled and its projected dot pattern is removed, the camera can still see infrared patterns projected by other cameras or by external projectors.  The IR emitter of RealSense cameras also operates in the 850 nm range.  

The D435i camera model is not equipped with an infrared-blocking filter on its sensors.  It can see all light frequencies in the visible to near-infrared range.

I note that in your RealSense Viewer you have auto-exposure disabled.  Does the structured light from the Gemini2 still appear for a few seconds if auto-exposure is enabled, please?

If you are sensing at close range (between 7 cm and 50 cm) then the **D405** close-range camera model would not be able to see an external infrared pattern as its sensors are equipped with an IR-blocking **IR Cut** filter so that it cannot see frequencies in the 850 nm range, and so cannot see a projected IR pattern.

Purchasing and fitting an external **short pass** filter over the lenses on the outside of the D435i camera could help to block IR frequencies from other cameras, as described in Intel's guide to using optical filters with 400 Series cameras.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

kao-yang, 2023-04-21:
> 
if auto-exposure is enabled, the structured light from the Gemini2 still appear for a few seconds。What puzzles me is that since D435i can't filter out 850nm infrared light, why does the structural light disappear from the image after a few seconds?

kao-yang, 2023-04-21:
> Is the Gemini2 camera's projection disabling itself?

No,Gemini2 camera's projection is always enable.Will D435f slove this problem?Can D435f filter the ir light at 850nm?

MartyG-RealSense, 2023-04-21:
D435f does not block infrared light.  Its filter blocks **visible** light frequencies (the ones below 850 nm) and allows infrared frequencies, so the Gemini2's projection would still be visible.  D405 can block infrared, but it is a short-range camera with an ideal depth sensing range of 7 cm to 50 cm (half a meter).

kao-yang, 2023-04-21:
> 

https://user-images.githubusercontent.com/57090523/233601962-7965d98a-aba9-4a5e-9209-362f2a0e2718.mp4

This video shows that the the structured light from the Gemini2 disappeared in the begin.But in the end of the video,with the rotation of cameras,the structured light appear again,and disappear again in a few seconds.

MartyG-RealSense, 2023-04-21:
I examined your case carefully but without using filter products on the RealSense lenses to block infrared frequencies from being seen, the best option may be to compensate for the loss of depth detail when the structured light pattern is active.  Two ways of doing this are:

1.  Set the **Laser Power** option in Steeo Module > Controls to its maximum value of '360'.

2.  Expand open the **Spatial Filter** post-processing filter and set its **Holes filling mode** drop-down menu to 'Unlimited' so that it attempts to automatically fill in the gaps in the depth image caused by the structured light.

[image]

I cannot comment on why the Gemini2 camera may be behaving in this manner as I can only advise on RealSense camera technology.

kao-yang, 2023-04-22:
> I examined your case carefully but without using filter products on the RealSense lenses to block infrared frequencies from being seen, the best option may be to compensate for the loss of depth [...]

Thank you very much for your answer. I will try this setting.

closed 2023-05-04 by MartyG-RealSense