realsenseai/realsense-ros #2704: while doing ros launch using: roslaunch realsense2_camera rs_camera.launch, I am getting error :RLException: [rs_camera.launch] is neither a launch file in package [realsense2_camera] nor is [realsense2_camera] a launch file name The traceback for the exception was written to the log file.

drakshayanidesai, 2023-04-21:


drakshayanidesai, 2023-04-21:
I am running it on Ubuntu 20.04 and installed ros using following link:http://wiki.ros.org/noetic/Installation/Ubuntu

MartyG-RealSense, 2023-04-21:
Hi @drakshayanidesai  Use of this launch file requires the installation of the RealSense SDK (librealsense) and the RealSense ROS1 wrapper (which is compatible with ROS Kinetic, Melodic and Noetic).

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

Have you installed the RealSense ROS1 wrapper, please?

drakshayanidesai, 2023-04-21:
Yes

On Fri, 21 Apr 2023 19:14:28 +0530 MartyG-RealSense  wrote

Hi @drakshayanidesai  Use of this launch file requires the installation of the RealSense SDK 
(librealsense) and the RealSense ROS1 wrapper (which is compatible with ROS Kinetic, Melodic and Noetic).
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy
Have you installed the RealSense ROS1 wrapper, please?

MartyG-RealSense, 2023-04-21:
Did you build the ROS1 wrapper from source code?

If you did, were you in the catkin_ws/src/ directory when you performed the **git clone** instruction `git clone https://github.com/IntelRealSense/realsense-ros.git`

drakshayanidesai, 2023-04-21:
Yes I have done exactly the but same error as shown in attachment

On Fri, 21 Apr 2023 22:13:40 +0530 MartyG-RealSense  wrote

Did you build the ROS1 wrapper from source code?
If you did, were you in the catkin_ws/src/ directory when you performed the git clone instruction git 
clone https://github.com/IntelRealSense/realsense-ros.git

MartyG-RealSense, 2023-04-22:
After performing catkin_make install, did you source the bashrc file?

echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

drakshayanidesai, 2023-04-24:
These two steps also completed

Thanks

On Sat, 22 Apr 2023 15:57:39 +0530 MartyG-RealSense  wrote

After performing catkin_make install, did you source the bashrc file?
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc

source ~/.bashrc

MartyG-RealSense, 2023-04-24:
Sometimes something has gone wrong in the installation process that does not have a traceable cause and the only solution is to completely delete the wrapper and start the installation process again from the beginning.  This can be done by deleting the entire **/catkin_ws/src** catkin workspace folder.

**********

Instead of installing the wrapper from source code, it may be easier to build librealsense and the wrapper together at the same time from packages instead of installing librealsense and the wrapper separately.  This can be done with an apt-get installation with the **Method 1** instructions.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#method-1-the-ros-distribution

This method does not install the RealSense Viewer tool though, so would not be suitable if you require the Viewer.

MartyG-RealSense, 2023-05-06:
Case closed due to no further comments received.

closed 2023-05-06 by MartyG-RealSense

drakshayanidesai, 2023-06-10:
Hi 

I have recorded a bag file using rs-record for 120 seconds. The test.bag is generated. Now I am unable 
to play it back. And how can I extract information from it?

Regards
Drakshayani

On Sat, 06 May 2023 13:20:35 +0530 MartyG-RealSense  wrote
>
Case closed due to no further comments received.

—

--=_f7ea1b545d828f9dd4954555fdf8159b
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset="UTF-8"

Hi <br />
<br />
I have recorded a bag file using rs-record for 120 seconds. The test.bag is=
 generated. Now I am unable <br />
to play it back. And how can I extract information from it?<br />
<br />
<br />
Regards<br />
Drakshayani<br />
<br />

om> wrote<br />
><br />
Case closed due to no further comments received.<br />
<br />
=E2=80=94<br />
nse/realsense-<br />
***@***.***><br />
<br />
<br>
--=_f7ea1b545d828f9dd4954555fdf8159b--

MartyG-RealSense, 2023-06-10:
Are you using **rosbag play** to play back a bag recorded with rs-record, please?  If you are then bags recorded in RealSense SDK programs, such as rs-record, have incompatibities with ROS playback.  Likewise, bags recorded in ROS have incompatibilities with RealSense SDK programs.  A RealSense team member describes this at https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

You can record RealSense topics in ROS by performing a launch of the RealSense ROS wrapper and then using **rosbag record -a** to record all currently published ROS topics to a bag.  It can then be played back n ROS with **rosbag play**

http://wiki.ros.org/rosbag/Commandline#rosbag_record