realsenseai/realsense-ros #2705: Problem with command 'ros2 launch realsense2_camera rs_launch.py'

stelmik, 2023-04-21:
Hi,

I have problem with execution of command:

`ros2 launch realsense2_camera rs_launch.py`

I installed ROS2 Rolling on Ubuntu 22.04 with kernel 5.19.0-40-generic (the same problem exist with kernel 5.15.0-70-generic).
I compiled and installed from source Realsense SDK than I installed realsense_camera, realsense_camera_msgs and realsense2_description.
I connected camera Intel RealSense D405 (USB 3.0).

When I try to execute (I haven't changed any content of the files from above installations):

`ros2 launch realsense2_camera rs_launch.py`

it starts and finish with exit code -11. The application log is: 

```log
[INFO] [launch]: All log files can be found below /home/username/.ros/log/2023-04-21-21-20-19-040266-computer-6650
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [6651]
[realsense2_camera_node-1] [INFO] [1682104820.621445674] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1682104820.621592511] [camera.camera]: Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1682104820.621633994] [camera.camera]: Running with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1682104820.671594220] [camera.camera]: Device with serial number 11456789001 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1682104820.671651967] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:1c.7/0000:08:00.0/usb6/6-1/6-1.1/6-1.1.4/6-1.1.4.3/6-1.1.4.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1682104820.671676605] [camera.camera]: Device with name Intel RealSense D405 was found.
[realsense2_camera_node-1] [INFO] [1682104820.671975196] [camera.camera]: Device with port number 6-1.1.4.3 was found.
[realsense2_camera_node-1] [INFO] [1682104820.672010808] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1682104820.690765433] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1682104820.691293489] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1682104820.691328114] [camera.camera]: Device Name: Intel RealSense D405
[realsense2_camera_node-1] [INFO] [1682104820.691352106] [camera.camera]: Device Serial No: 11456789001
[realsense2_camera_node-1] [INFO] [1682104820.691380618] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:1c.7/0000:08:00.0/usb6/6-1/6-1.1/6-1.1.4/6-1.1.4.3/6-1.1.4.3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1682104820.691405010] [camera.camera]: Device FW version: 05.14.00.00
[realsense2_camera_node-1] [INFO] [1682104820.691425871] [camera.camera]: Device Product ID: 0x0B5B
[realsense2_camera_node-1] [INFO] [1682104820.691443993] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1682104820.735490732] [camera.camera]: Could not set param: depth_module.power_line_frequency with 3 Range: [0, 2]: parameter 'depth_module.power_line_frequency' could not be set: Parameter {depth_module.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1682104820.758175248] [camera.camera]: Set Exposure.1 to 7500
[realsense2_camera_node-1] [INFO] [1682104820.859060600] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1682104820.928908776] [camera.camera]: Stopping Sensor: Depth Module
[ERROR] [realsense2_camera_node-1]: process has died [pid 6651, exit code -11, cmd '/opt/ros/rolling/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_cbte4qow --params-file /tmp/launch_params_9g6ywy7z'].
```

From the last line 
`/opt/ros/rolling/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_cbte4qow --params-file /tmp/launch_params_9g6ywy7z`
it look like it create temporary file '/tmp/launch_params_cbte4qow' with content:

```yaml
/camera/camera:
  ros__parameters:
    accel_fps: 0
    align_depth.enable: false
    allow_no_texture_points: false
    camera_name: camera
    clip_distance: -2.0
    colorizer.enable: false
    config_file: ''
    decimation_filter.enable: false
    depth_module.exposure.1: 7500
    depth_module.exposure.2: 1
    depth_module.gain.1: 16
    depth_module.gain.2: 16
    depth_module.hdr_enabled: false
    depth_module.profile: 0,0,0
    device_type: ''
    diagnostics_period: 0.0
    enable_accel: false
    enable_color: true
    enable_confidence: true
    enable_depth: true
    enable_fisheye1: true
    enable_fisheye2: true
    enable_gyro: false
    enable_infra1: false
    enable_infra2: false
    enable_pose: true
    enable_sync: false
    gyro_fps: 0
    hdr_merge.enable: false
    infra_rgb: false
    initial_reset: false
    json_file_path: ''
    linear_accel_cov: 0.01
    log_level: info
    output: screen
    pointcloud.enable: false
    pointcloud.ordered_pc: false
    pointcloud.stream_filter: 2
    pointcloud.stream_index_filter: 0
    pose_fps: 200
    reconnect_timeout: 6.0
    rgb_camera.profile: 0,0,0
    rosbag_filename: ''
    serial_no: ''
    tf_publish_rate: 0.0
    unite_imu_method: 0
    usb_port_id: ''
    wait_for_device_timeout: -1.0
```

and create file '/tmp/launch_params_9g6ywy7z' with content:

```yaml
/camera/camera:
  ros__parameters: {}
```

Maybe somebody know where the problem is and how to solve it?
Thank you for any help.

MartyG-RealSense, 2023-04-22:
Hi @stelmik  The RealSense ROS wrapper's documentation recommends that ROS2 **Humble** is used with Ubuntu 22.04.

[image]

Whilst Rolling can work with the wrapper when the wrapper is built from source code, it is not recommended to use Rolling with it.  This is because Rolling is continually updated and so may introduce changes that cause breakages in the wrapper's Rolling support.

Did you install the wrapper from source by Git Cloning its repository and then using the **colcon build** command, as described in its instructions under the **Step 3: Install Intel® RealSense™ ROS2 wrapper** link of the wrapper documentation, please?

https://github.com/IntelRealSense/realsense-ros

The instructions are uncovered by left-clicking on the small arrow next to Step 3.

[image]

stelmik, 2023-04-22:
@MartyG-RealSense yes, I install the wrapper from source by Git Cloning its repository and then using the colcon build command, as described in its instructions under the Step 3: Install Intel® RealSense™ ROS2 wrapper -> Option 2: Install from source.
Finally I tried to download the package for Humble and prepare my package for ROS2 Rolling base on it. Then I install the prepared Ubuntu package.
Therefore I have `realsense_camera` in `/opt/ros/rolling/` too.
Both of the solutions not working. There is the same error message.

I try other command:

```shell
ros2 run realsense2_camera realsense2_camera_node
```
but there are error:
```log
...
[INFO] [1682196333.326644751] [camera]: getParameters...
[INFO] [1682196333.327043585] [camera]: JSON file is not provided
[INFO] [1682196333.327075797] [camera]: Device Name: Intel RealSense D405
[INFO] [1682196333.327098245] [camera]: Device Serial No: 11456789001
[INFO] [1682196333.327119075] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:1c.7/0000:08:00.0/usb6/6-1/6-1.1/6-1.1.4/6-1.1.4.3/6-1.1.4.3:1.0/video4linux/video0
[INFO] [1682196333.327141615] [camera]: Device FW version: 05.14.00.00
[INFO] [1682196333.327161493] [camera]: Device Product ID: 0x0B5B
[INFO] [1682196333.327179203] [camera]: Sync Mode: Off
[WARN] [1682196333.363839157] [camera]: Could not set param: depth_module.power_line_frequency with 3 Range: [0, 2]: parameter 'depth_module.power_line_frequency' could not be set: Parameter {depth_module.power_line_frequency} doesn't comply with integer range.
[INFO] [1682196333.544423389] [camera]: Stopping Sensor: Depth Module
[ros2run]: Segmentation fault
```

Look, I did the same installation on February 2023 and it worked then.
I have to reinstall Ubuntu system a week ago and now it doesn't work.

I think something have to change. Maybe exit code say something? The exit code is -11.

stelmik, 2023-04-22:
I checked and the line 231 in `rs_node_setup.cpp` source file is the problem in ROS2 Rolling:

Link:
https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/src/rs_node_setup.cpp#L231

It is:
```cpp
_image_publishers[sip] = std::make_shared<image_transport_publisher>(_node, image_raw.str(), qos);
```

When I change it to old version and compile by colcon build command then it work well.

Therefore my question should be: How to make `image_transport_publisher` to work on ROS2 Rolling well? :)

Have a nice day.

MartyG-RealSense, 2023-04-29:
Sincere apologies for the delay in responding further to your question about Rolling. I will highlight it to the RealSense ROS development team.

MartyG-RealSense, 2023-05-06:
Thanks very much @stelmik for your patience.  Whilst there is not currently a solution for the error that you are experiencing, the RealSense ROS development team will look at implementing measures to improve runtime testing of Rolling before a new wrapper version is released.

I have added an **Enhancement** label to this case to ensure that it is kept open whilst the development team's work on Rolling is ongoing.

stelmik, 2023-05-06:
@MartyG-RealSense thank you for your help.
Today I downloaded the librealsense and realsense-ros sources and recompiled then. It works very well.
I see the realsense-ros source has not changed, but the ROS2 Rolling packages for Ubuntu 22.04 have been updated twice in the last 2 weeks.
Therefore, in my opinion, the problem was in ROS 2 Rolling. You also wrote that Rolling is constantly updated and after each update it can cause problems.
I think you can close the issue as resolved or not connected with realsense-ros project. Thanks!

MartyG-RealSense, 2023-05-07:
You are very welcome, @stelmik - t's great to hear that you achieved a successful installation.  Thanks very much for the update.  As you suggest, I will close this case.  Thanks again!

closed 2023-05-07 by MartyG-RealSense