realsenseai/realsense-ros #2706: How to display detected object 3D coordinate marker in rviz using Realsense D455

TapleFlib, 2023-04-22:
How to display marker and get the position in 3D space of detected object using D455?
I want it to be like this video : https://www.youtube.com/watch?v=iJWHRW7sZW8
Are there example code for this?

MartyG-RealSense, 2023-04-23:
Hi @TapleFlib  You could use **darknet_ros** - which uses YOLO v3 - in combination with **darknet_ros_3d** to obtain the 3D coordinate.  darknet_ros is run first to obtain 2D information and then darknet_ros_3d is run seocndly to obtain 3D information.

TapleFlib, 2023-04-23:
I'll try Thankyou !

TapleFlib, 2023-04-23:
Is it enough to only use darknet_ros_3d without darknet_ros ? or both are required ? I'm planning to send the position of object to robot arm.

MartyG-RealSense, 2023-04-23:
I have not personally used darknet_ros but a RealSense user at https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/issues/15#issuecomment-634574521 stated that they were able to launch darknet_ros_3d without launching darknet_ros first.

TapleFlib, 2023-04-23:
I see, thank you !

closed 2023-04-23 by TapleFlib