realsenseai/realsense-ros #2708: Waiting for image in rviz using darknet_ros

TapleFlib, 2023-04-24:
Waiting for image when I already run the cam driver, darknet launch, and rviz.
launching files :

- camera : roslaunch realsense2_camera rs_camera.launch
- darknet : roslaunch darknet_ros yolo_v3.launch
- rviz : rviz

Ros : noetic

[image]

[image]

TapleFlib, 2023-04-24:
Just now it worked but only if I switch between /camera/color/image_raw and /darknet_ros/detection_image topic and it only run once with good fps, 0.1 fps for the rest 

[image]

MartyG-RealSense, 2023-04-25:
Hi @TapleFlib  I could only find a couple of issues where an FPS of 0.1 in darknet_ros is mentioned.  In the best of those references, it is suggested that CUDA graphics support might not be enabled in the makefile of darknet_ros.

[image]

MartyG-RealSense, 2023-05-02:
Hi @TapleFlib  Do you require further assistance with this case, please?  Thanks!

TapleFlib, 2023-05-02:
I'm using amd so this might not help me, I used my friend's laptop with nvidia and it didn't lag, thanykou !

closed 2023-05-02 by TapleFlib

MartyG-RealSense, 2023-05-02:
Thanks very much for the update.  RealSense 400 Series cameras should work with any Intel or Arm processor.  In the small number of cases where RealSense users reported using AMD processors, AMD Ryzen Threadripper processors were reported to be able to work with the cameras whilst non-Threadripper ones did not.

TapleFlib, 2023-05-02:
thanks for the info !