realsenseai/realsense-ros #2710: what is camera/depth_registered/points in realsense?

TapleFlib, 2023-04-25:
this is the default darknet_3d yaml file :
[image]
I use `rostopic list` to see my realsense D455 topic but it seems `/camera/depth_registered/points` is not there. What should I use instead of `/camera/depth_registered/points` ?

MartyG-RealSense, 2023-04-25:
Hi @TapleFlib  The topic **/camera/depth_registered/points** should be published if you launch the ROS1 wrapper with the **rs_rgbd** launch file.

`roslaunch realsense2_camera rs_rgbd.launch`

TapleFlib, 2023-04-25:
Ah I see, thank you so much.
By the way, what is the difference between the launch files? Why don't realsense only make one launch files that publish all topics?

MartyG-RealSense, 2023-04-25:
Intel provide a range of launch files that perform a dedicated function (such as multiple cameras, launching with a bag file instead of a live camera, SLAM navigation, Gazebo simulation, etc.  It would not be practical to provide all of those functions in a single launch file.

A key difference between the rs_camera and rs_rgbd launch files was originally that rs_camera published an **unorganized** point cloud whilst rs_rgbd published an **organized** point cloud.  Later though, the parameter **ordered_pc** was introduced in the ROS wrapper to enable rs_camera to publish an organized pointcloud instead of an unorganized one when ordered_pc = true.  

[image]

TapleFlib, 2023-04-25:
I see, any documentation on what these launch files does? I need info for rs_rgbd and rs_aligned_depth

MartyG-RealSense, 2023-04-25:
rs_aligned_depth has **align_depth** set to true by default so that additional topics aligned to depth are published.  It also enables the infra1 and infra2 infrared topics by default, whilst in rs_rgbd they are disabled by default.

There is no real reason to use rs_aligned_depth.launch as it is severely outdated (over 4 years old), align_depth is enabled by default in rs_rgbd too and you can easily enable the infra1 and infra2 topics in rs_rgbd if you want to by adding **enable_infra1:=true enable_infra2:=true** to the end of the roslaunch instruction.

You can examine the contents of each launch file here:

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy/realsense2_camera/launch

closed 2023-04-25 by TapleFlib