realsenseai/realsense-ros #2711: Core Dump if serial_no not given

danielhonies, 2023-04-25:
https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/src/realsense_node_factory.cpp 

Line 277 throws error: 

std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::reference std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::front() [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>; reference = char&]: Assertion '!empty()' failed.

if no serial number is given.

danielhonies, 2023-04-25:
ros2 run realsense2_camera realsense2_camera_node throws the error

ros2 run realsense2_camera realsense2_camera_node  --ros-args -p serial_no:=_123456789 works

However I think this should be fixed in code by checking for _serial_no to not be empty. This could be due to us using a newer compiler (gcc12).

danielhonies, 2023-04-25:
I can get a PR ready if you want.

danielhonies, 2023-04-25:
ros2 launch realsense2_camera rs_launch.py throws the same error

MartyG-RealSense, 2023-04-25:
I have not seen this problem reported before.

danielhonies, 2023-04-25:
Yes the behaviour for calling string.front() for empty strings might have changed with compilers recently and the default ros2 setup still uses an older compiler.

danielhonies, 2023-04-25:
as can be seen in the reference: https://en.cppreference.com/w/cpp/string/basic_string/front the behaviour is undefined for empty strings. Therefore we should check it.

MartyG-RealSense, 2023-04-25:
Thanks so much for the PR at https://github.com/IntelRealSense/realsense-ros/pull/2713  :)

As this issue is associated with the PR, I have added an **Enhancement** label to it so that the issue is kept open.

closed 2023-05-01 by Nir-Az

Nir-Az, 2023-05-01:
Hi all.
Issue have been merged.
Thanks for the fix.