realsenseai/realsense-ros #2719: Error when running rs_rgbd.launch on another package

TapleFlib, 2023-04-29:
I have 2 workspaces 1 for realsense and the other one for my project, I git clone realsense-ros to both of the workspaces , but when I run 
rs_rgbd.launch, it is normal on the realsense ws but it shows the following error on another ws :

`[camera/realsense2_camera_manager-2] process has died [pid 47559, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/tzuyu/.ros/log/a059623a-e66b-11ed-94fe-7b65ad546251/camera-realsense2_camera_manager-2.log].
log file: /home/tzuyu/.ros/log/a059623a-e66b-11ed-94fe-7b65ad546251/camera-realsense2_camera_manager-2*.log`

MartyG-RealSense, 2023-04-29:
Hi @TapleFlib  I would recommend only having one RealSense ROS catkin workspace if possible.  

If you do need to create more than one workspace, a guide at the link below advises that you cannot run multiple workspaces in the same session and need to switch between them with the **source** command.  The last workspace to be sourced is the one that is activated.

https://roboticsbackend.com/ros-multiple-catkin-workspaces/

TapleFlib, 2023-04-29:
I see, thankyou !

closed 2023-04-29 by TapleFlib