realsenseai/realsense-ros #2721: D455 instllation on rock-5b (kernel 5.10)  ubuntu 20.04

hemsinghb, 2023-05-01:
Please make available the installation package foe ubuntu 20.04 kernel 5.10 such that most of the users will get benefit out of this package. I have this D455 but no use as no support on kernel 5.10 no ros2 wrapper for arm64. Please ...

MartyG-RealSense, 2023-05-01:
Hi @hemsinghb  My understanding is that Arm64 build of the RealSense ROS wrapper is supported for ROS2 Foxy and Galactic (Ubuntu 20.04) and Humble (22.04), as mentioned by a member of Intel's RealSense ROS development team at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1280318586

For devices with the Rockchip (RK) chipset though, it is recommendable to build the librealsense SDK from source code with CMake using the **RSUSB backend** installation method.  RSUSB bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require kernel patching.  

The RSUSB source code installation procedure is described at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

Once librealsense has been successfully installed then you could afterwards build the ROS2 wrapper from source code.

********

An alternative approach may be to use the older **ros2-legacy** generation of the RealSense ROS2 wrapper and build librealsense and the wrapper together at the same time from packages using **Method 1**.  The packages of this older wrapper were based on RSUSB.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-legacy#method-1-the-ros-distribution

[image]

hemsinghb, 2023-05-01:
> Intel's RealSense ROS development team at IntelRealSense/librealsense#10439
> (comment)
> The RSUSB source code installation procedure is described at IntelRealSense/librealsense#9931

Thank you so much . One small issue  just tell me please ...
    What version of SDK2.0  matches with which version of ROS Wrapper for
ros2 foxy  ( ubuntu 20.04)

I have Ros2 FOXY  installed on  Ubuntu 20.04, I want use d455 for robot
navigation.

.. Thanks in advance, ...regards. hems.

MartyG-RealSense, 2023-05-01:
On the ROS wrapper releases page, the SDK version to match with a particular wrapper version is listed under the 'Supported RealSense SDK' heading of each version's information listing.

https://github.com/IntelRealSense/realsense-ros/releases

The current latest ROS2 wrapper version 4.51.1 is compatible with Foxy on Ubuntu 20.04 and SDK 2.51.1.