realsenseai/realsense-ros #2723: Pointcloud2 doesn

TapleFlib, 2023-05-01:


MartyG-RealSense, 2023-05-01:
Hi @TapleFlib  If ypur friend had already installed the wrapper with the sudo command then that will have installed both the librealsense SDK and the ROS wrapper from packages at the same time.  If you then installed the SDK and wrapper youself afterwards then it is possible that there are two SDK installations that are conflicting with each other.

If you also used the sudo command then you could try removing all RealSense related packages from the computer by inputting the command below into the Ubuntu terminal, and then repeating the SDK and wrapper installation.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

TapleFlib, 2023-05-01:
I'll try it thanks !

closed 2023-05-02 by TapleFlib