realsenseai/realsense-ros #2724: CMake couldn't find realsense2

mericgeren, 2023-05-04:
I am using Windows 10 and currently have ros noetic, Intel Realsense SDK 2.0 v2.50.0 installed. But, when i follow Step 2 of installation instructions for realsense-ros (with tag 2.3.2) at the step where i enter the command: `catkin_make clean` i get the following error message: 

> -- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:42 (find_package):
  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "C:/my_catkin_wp/build/CMakeFiles/CMakeOutput.log".
See also "C:/my_catkin_wp/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

could you help me on this issue please?

MartyG-RealSense, 2023-05-04:
Hi @mericgeren  I would recommend against installing the ROS wrapper on Windows if it is possible to use Linux instead.  It is possible to install the wrapper on Windows but the majority of attempts to do so are unsuccessful.

Using the direct version number of the wrapper (2.3.2) instead of a tag was the correct action on Windows.

When following the wrapper's Windows installation instructions, be careful to input each of the **cd ..** directory change commands so that you are in the correct directory when inputting the **catkin_make clean** command.  I believe that after the two cd .. commands are used, you should be in the **c:\catkin_ws\** directory (not \src) when inputting the catkin_make clean command.

MartyG-RealSense, 2023-05-04:
Yes, I mean inputting `git checkout 2.3.2`

mericgeren, 2023-05-04:
I just tried your suggestion. But, the error seems to persists.

MartyG-RealSense, 2023-05-04:
Does the same ocur if you put a V in front of the version number: `git checkout v2.3.2`

mericgeren, 2023-05-04:
When i put a V in fron of the version number i see this error: 

> error: pathspec 'v2.3.2' did not match any file(s) known to git

MartyG-RealSense, 2023-05-04:
Thank you.  Backtracking in the instructions a little, can you confirm that you:

1.  Cloned ddynamic_reconfigure into catkin_ws/src/

2.  Did `catkin_init_workspace` whilst in catkin_ws/src/

mericgeren, 2023-05-04:
1. Yes. I have cloned ddynamic_reconfigure into \src in my catkin workspace and i built it when i input `catkin_make clear` without any issue:

> 

> -- +++ processing catkin package: 'ddynamic_reconfigure'
> -- ==> add_subdirectory(ddynamic_reconfigure)
> -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy

2. Yes. I did whilst in \src of my catkin workspace:

`C:\my_catkin_wp\src>catkin_init_workspace
Creating symlink "C:\my_catkin_wp\src\CMakeLists.txt" pointing to "c:\opt\ros\noetic\x64\share\catkin\cmake\toplevel.cmake"`

MartyG-RealSense, 2023-05-04:
It appears that you have followed the entire procedure correctly and it still does not work.  There is nothing else that can be done, unfortunately.  As mentioned, installing the ROS wrapper on Windows usually does not succeed.

mericgeren, 2023-05-04:
Thank you for your help. Doesn't editing CMakeLists.txt or package.xml in a way which enables cmake to create config.cmake files for realsense can solve this issue?

Or do i need to create config.make files for realsense ground up? If so how can i do this?

MartyG-RealSense, 2023-05-04:
I could not find any references that indicate that editing CMAkeLists.txt will solve the problem.

A RealSense user at https://github.com/IntelRealSense/realsense-ros/issues/2280#issuecomment-1065210147 had the same problem on Windows with catkin_make clean and decided in the end not to use Windows.  I appreciate that this is not an option in your case.  

This year (2023) a RealSense user commented at https://github.com/IntelRealSense/realsense-ros/issues/2280#issuecomment-1411759588 that they were successful if they used the version number **2.2.22**.  So you could try the previous version number **2.3.1** instead of 2.3.2 to see if it makes a difference.

It is not known why this issue with Windows occurs and there are no other solutions for the ROS1 wrapper if it occurs.  A couple of ROS2 wrapper users got the RealSense ROS2 wrapper working with Windows at https://github.com/IntelRealSense/realsense-ros/issues/2587 and https://github.com/IntelRealSense/realsense-ros/issues/2660 after applyig numerous step by step fixes, which illustrates the high difficulty of achieving a successful wrapper installation on Windows.

mericgeren, 2023-05-04:
Thank you so much for your help. Currently trying 2.2.22 i will update on a success or a failiure.

mericgeren, 2023-05-04:
Hi, just tried all the steps from very beginning using ROS wrapper version 2.2.22 and Realsense SDK2.0 v2.42.0 exactly following steps without any mistake and unfortunatelly, i got the same error. But, i was able to input the command: `C:\my_catkin_wp>roslaunch realsense2_camera rs_camera.launch`

and i got the following messages afterwards:

> ... logging to C:\Users\User\.ros\log\e1a068fe-ea60-11ed-b60c-8ca982e040b0\roslaunch-DESKTOP-2KBLDP7-9396.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> started roslaunch server http://127.0.0.1:52491/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /camera/realsense2_camera/accel_fps: -1
>  * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
>  * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
>  * /camera/realsense2_camera/align_depth: False
>  * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/allow_no_texture_points: False
>  * /camera/realsense2_camera/base_frame_id: camera_link
>  * /camera/realsense2_camera/calib_odom_file:
>  * /camera/realsense2_camera/clip_distance: -2.0
>  * /camera/realsense2_camera/color_fps: -1
>  * /camera/realsense2_camera/color_frame_id: camera_color_frame
>  * /camera/realsense2_camera/color_height: -1
>  * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
>  * /camera/realsense2_camera/color_width: -1
>  * /camera/realsense2_camera/confidence_fps: -1
>  * /camera/realsense2_camera/confidence_height: -1
>  * /camera/realsense2_camera/confidence_width: -1
>  * /camera/realsense2_camera/depth_fps: -1
>  * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
>  * /camera/realsense2_camera/depth_height: -1
>  * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
>  * /camera/realsense2_camera/depth_width: -1
>  * /camera/realsense2_camera/device_type:
>  * /camera/realsense2_camera/enable_accel: False
>  * /camera/realsense2_camera/enable_color: True
>  * /camera/realsense2_camera/enable_confidence: True
>  * /camera/realsense2_camera/enable_depth: True
>  * /camera/realsense2_camera/enable_fisheye1: False
>  * /camera/realsense2_camera/enable_fisheye2: False
>  * /camera/realsense2_camera/enable_fisheye: False
>  * /camera/realsense2_camera/enable_gyro: False
>  * /camera/realsense2_camera/enable_infra1: False
>  * /camera/realsense2_camera/enable_infra2: False
>  * /camera/realsense2_camera/enable_infra: False
>  * /camera/realsense2_camera/enable_pointcloud: False
>  * /camera/realsense2_camera/enable_pose: False
>  * /camera/realsense2_camera/enable_sync: False
>  * /camera/realsense2_camera/filters:
>  * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
>  * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
>  * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
>  * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
>  * /camera/realsense2_camera/fisheye_fps: -1
>  * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
>  * /camera/realsense2_camera/fisheye_height: -1
>  * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
>  * /camera/realsense2_camera/fisheye_width: -1
>  * /camera/realsense2_camera/gyro_fps: -1
>  * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
>  * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
>  * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 26 lines cut ...]
>  * /rosversion: 1.15.9
> 
> NODES
>   /camera/
>     realsense2_camera (nodelet/nodelet)
>     realsense2_camera_manager (nodelet/nodelet)
> 
> auto-starting new master
> process[master]: started with pid [25632]
> ROS_MASTER_URI=http://127.0.0.1:11311
> setting /run_id to e1a068fe-ea60-11ed-b60c-8ca982e040b0
> process[rosout-1]: started with pid [34352]
> started core service [/rosout]
> process[camera/realsense2_camera_manager-2]: started with pid [32776]
> process[camera/realsense2_camera-3]: started with pid [1424]
> [ERROR] [1683193739.130972600]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet_topic_tools/test/test_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..
> [ERROR] [1683193739.131334700]: Skipped loading plugin with error: XML Document 'test_nodelet/test_nodelet.xml' has no Root Element. This likely means the XML is malformed or missing..
> [ INFO] [1683193739.156376700]: Initializing nodelet with 8 worker threads.
> [ERROR] [1683193740.033440300]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet_topic_tools/test/test_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..
[... cut ...]
> [FATAL] [1683193740.037245400]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
> ←[31m[camera/realsense2_camera-3] process has died [pid 1424, exit code 4294967295, cmd c:\opt\ros\noetic\x64\lib\nodelet\nodelet.exe load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=C:\Users\User\.ros\log\e1a068fe-ea60-11ed-b60c-8ca982e040b0\camera-realsense2_camera-3.log].
> log file: C:\Users\User\.ros\log\e1a068fe-ea60-11ed-b60c-8ca982e040b0\camera-realsense2_camera-3*.log←[0m

MartyG-RealSense, 2023-05-04:
The roslaunch log does not get to the point where the wrapper version and RealSense SDK version are detected before the first errors occur, which suggests to me that realsense2_camera (the wrapper) has not been built and so it cannot be loaded.

mericgeren, 2023-05-05:
Dear @MartyG-RealSense, 

Just switched to Linux 18.04LTS with kernel 5.3.0-59-generic and ROS Melodic. I have git cloned ddynamic and RealSense SDK 2.0 ( did git checkout 2.3.2 right after). Also, downloaded the zip file which contains the source code of librealsense v2.50.0 and extracted it directly to home directory. After that, created and moved into build directory at the root of librealsense. Then, using CMake 3.10.2 i built my project with the command: `cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
`. Then, followed steps of realsense-ros installation. When i input the command: `catkin_make clean` i got this error: 

> 
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at realsense-ros/realsense2_camera/CMakeLists.txt:45 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2"
  (requested version 2.50.0) with any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:49 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

> -- Configuring incomplete, errors occurred!
> See also "/home/baran/catkin_ws/build/CMakeFiles/CMakeOutput.log".
> See also "/home/baran/catkin_ws/build/CMakeFiles/CMakeError.log".
> Makefile:1554: recipe for target 'cmake_check_build_system' failed
> make: *** [cmake_check_build_system] Error 1
> Invoking "make cmake_check_build_system" failed

Can you help me on this issue please?

closed 2023-05-05 by mericgeren