realsenseai/realsense-ros #2725: How to know Serial number of Realsense Camera D405

kevinDrawn, 2023-05-04:
Hello. I want to know Serial number to run below code.

pipeline = rs.pipeline() 
serial_number = <your camera's serial number>
config = rs.config()
config.enable_device(serial_number)
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
pipeline.start(config)

How to know Serial number of D405 Camera?

There are two mystery number at the side of the camera. (refer to the picture)

[image]

MartyG-RealSense, 2023-05-04:
Hi @kevinDrawn  The serial number is the number next to Made in Thailand.  You can also see the serial number in the RealSense Viewer tool by clicking the 'Info' option at the top of the Viewer's options side-panel.

kevinDrawn, 2023-05-05:
Oh thank you a lot!!

MartyG-RealSense, 2023-05-05:
You are very welcome.  As you have the serial number, I will close this case.  Thanks!