realsenseai/realsense-ros #2727: topic /aligned_depth_to_color/image_raw is missing

alonsnir, 2023-05-05:
Ubuntu 22.04 with 5.15.0-71-generic kernel
Ros Humble
librealsense2 debian package installed from Intel servers
ROS2 wrapper Option 1: Install debian package from ROS servers

Now when I run as following, the /aligned_depth_to_color/image_raw topic is not available
```
ros2 run realsense2_camera realsense2_camera_node --ros-args --remap align_depth:=true

ros2 run realsense2_camera realsense2_camera_node --ros-args --remap align_depth:=true
[INFO] [1683288144.395186983] [camera]: RealSense ROS v4.51.1
[INFO] [1683288144.395513731] [camera]: Built with LibRealSense v2.51.1
[INFO] [1683288144.395571944] [camera]: Running with LibRealSense v2.51.1
[INFO] [1683288144.412555439] [camera]: Device with serial number 829212071017 was found.

[INFO] [1683288144.412598650] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video8 was found.
[INFO] [1683288144.412609056] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1683288144.412841731] [camera]: Device with port number 2-3 was found.
[INFO] [1683288144.412867225] [camera]: Device USB type: 3.2
[INFO] [1683288144.414596457] [camera]: getParameters...
[INFO] [1683288144.414812349] [camera]: JSON file is not provided
[INFO] [1683288144.414826454] [camera]: Device Name: Intel RealSense D435
[INFO] [1683288144.414834761] [camera]: Device Serial No: 829212071017
[INFO] [1683288144.414842061] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video8
[INFO] [1683288144.414849525] [camera]: Device FW version: 05.14.00.00
[INFO] [1683288144.414857121] [camera]: Device Product ID: 0x0B07
[INFO] [1683288144.414864407] [camera]: Sync Mode: Off
[ERROR] [1683288144.426225983] [camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ERROR] [1683288144.426542130] [camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ERROR] [1683288144.426848664] [camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ERROR] [1683288144.427154809] [camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[INFO] [1683288144.446846623] [camera]: Stopping Sensor: Stereo Module
[INFO] [1683288144.450432576] [camera]: Starting Sensor: Stereo Module
[INFO] [1683288144.464215260] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683288144.464278053] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683288144.464289105] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1683288144.464484809] [camera]: Stopping Sensor: RGB Camera
[INFO] [1683288144.522361847] [camera]: Starting Sensor: RGB Camera
[INFO] [1683288144.524914776] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[INFO] [1683288144.525411612] [camera]: RealSense Node Is Up!
[WARN] [1683288144.583592212] [camera]: 
 ```

```
ros2 topic list
/color/camera_info
/color/image_raw
/color/metadata
/depth/camera_info
/depth/image_rect_raw
/depth/metadata
/extrinsics/depth_to_color
/extrinsics/depth_to_infra1
/extrinsics/depth_to_infra2
/imu
/infra1/camera_info
/infra1/image_rect_raw
/infra1/metadata
/infra2/camera_info
/infra2/image_rect_raw
/infra2/metadata
/parameter_events
/rosout
/tf_static
```
As shown above aligned_depth_to_color is not listed

MartyG-RealSense, 2023-05-05:
Hi @alonsnir  The list of **An error has occurred** messages is likely occurring because you have firmware 5.14.0.0 installed in your camera but the SDK version is 2.51.1.  Firmware 5.14.0.0 is designed for SDK **2.53.1** and causes these errors if used with an older SDK.  The errors should disappear if you downgrade your firmware to version **5.13.0.50**, which is designed for SDK 2.51.1.

Are the alined topic published if you launch with **ros2 launch** instead of ros2 run?

`ros2 launch realsense2_camera rs_launch.py align_depth:=true`

alonsnir, 2023-05-05:
Hi @MartyG-RealSense, thanks for replying
Same error with the launch option

```
$ ros2 launch realsense2_camera rs_launch.py align_depth:=true
[INFO] [launch]: All log files can be found below /home/alon/.ros/log/2023-05-05-15-26-58-784739-alon-154478
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [154479]
[realsense2_camera_node-1] [INFO] [1683289619.199415208] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1683289619.199675490] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683289619.199734678] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683289619.241070508] [camera.camera]: Device with serial number 829212071017 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1683289619.241158193] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video8 was found.
[realsense2_camera_node-1] [INFO] [1683289619.241183327] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1683289619.241619307] [camera.camera]: Device with port number 2-3 was found.
[realsense2_camera_node-1] [INFO] [1683289619.241655851] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1683289619.244773776] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1683289619.245111075] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1683289619.245137979] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1683289619.245153513] [camera.camera]: Device Serial No: 829212071017
[realsense2_camera_node-1] [INFO] [1683289619.245168749] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video8
[realsense2_camera_node-1] [INFO] [1683289619.245184000] [camera.camera]: Device FW version: 05.14.00.00
[realsense2_camera_node-1] [INFO] [1683289619.245201000] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1683289619.245215319] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [ERROR] [1683289619.257751336] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [1683289619.258217432] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [1683289619.258697730] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [1683289619.259186160] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [INFO] [1683289619.262918315] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1683289619.272068210] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30
[realsense2_camera_node-1] [WARN] [1683289619.278247484] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1683289619.279204000] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683289619.281087744] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683289619.290216232] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1683289619.290607451] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683289619.300866854] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683289619.302863186] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1683289619.303247379] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1683289619.420932983] [camera.camera]: 

```

How do I downgrade firmaware to version 5.13.0.50?

MartyG-RealSense, 2023-05-05:
Do you have access to the RealSense Viewer tool?  If you do then you can download the firmware **.bin** file for 5.13.0.50 to your computer from the link below.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

[image]

Once the .bin is downloaded, go to the **More** option near the top of the Viewer's options side-panel and select **Update Firmware** from its drop-down menu.  A file selector will appear.  Navigate to the folder on your computer that the .bin file was downloaded to and select the .bin file.  The downgrade will then automatically begin.

alonsnir, 2023-05-05:
@MartyG-RealSense 

the error gone, but still no alignment

```
ros2 launch realsense2_camera rs_launch.py align_depth:=true
[INFO] [launch]: All log files can be found below /home/alon/.ros/log/2023-05-05-15-59-01-902708-alon-157121
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [157122]
[realsense2_camera_node-1] [INFO] [1683291542.293611927] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1683291542.293855079] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683291542.293909628] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683291542.318127228] [camera.camera]: Device with serial number 829212071017 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1683291542.318160685] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2 was found.
[realsense2_camera_node-1] [INFO] [1683291542.318193810] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1683291542.318394781] [camera.camera]: Device with port number 2-3 was found.
[realsense2_camera_node-1] [INFO] [1683291542.318427767] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1683291542.320107570] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1683291542.320342748] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1683291542.320359471] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1683291542.320368924] [camera.camera]: Device Serial No: 829212071017
[realsense2_camera_node-1] [INFO] [1683291542.320391624] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
[realsense2_camera_node-1] [INFO] [1683291542.320400399] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1683291542.320408302] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1683291542.320416199] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1683291542.334094059] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1683291542.341379490] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30
[realsense2_camera_node-1] [WARN] [1683291542.346337405] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1683291542.347172362] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683291542.349217415] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683291542.358569338] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1683291542.358793629] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683291542.369206318] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683291542.371281626] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1683291542.371633326] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1683291542.488945873] [camera.camera]: 
```
```
$ ros2 topic list
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/parameter_events
/rosout
/tf_static
```

```
$ ros2 run realsense2_camera realsense2_camera_node --ros-args --remap align_depth:=true
[INFO] [1683291818.477739469] [camera]: RealSense ROS v4.51.1
[INFO] [1683291818.477984979] [camera]: Built with LibRealSense v2.51.1
[INFO] [1683291818.478036027] [camera]: Running with LibRealSense v2.51.1
[INFO] [1683291818.514624376] [camera]: Device with serial number 829212071017 was found.

[INFO] [1683291818.514661061] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2 was found.
[INFO] [1683291818.514675915] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1683291818.514857287] [camera]: Device with port number 2-3 was found.
[INFO] [1683291818.514873987] [camera]: Device USB type: 3.2
[INFO] [1683291818.516523181] [camera]: getParameters...
[INFO] [1683291818.516761828] [camera]: JSON file is not provided
[INFO] [1683291818.516781056] [camera]: Device Name: Intel RealSense D435
[INFO] [1683291818.516794760] [camera]: Device Serial No: 829212071017
[INFO] [1683291818.516807111] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
[INFO] [1683291818.516819822] [camera]: Device FW version: 05.13.00.50
[... 11 lines cut ...]
[WARN] [1683291818.750610780] [camera]: 
```

```
$ ros2 topic list
/color/camera_info
/color/image_raw
/color/metadata
/depth/camera_info
/depth/image_rect_raw
/depth/metadata
/extrinsics/depth_to_color
/extrinsics/depth_to_infra1
/extrinsics/depth_to_infra2
/imu
/infra1/camera_info
/infra1/image_rect_raw
/infra1/metadata
/infra2/camera_info
/infra2/image_rect_raw
/infra2/metadata
/parameter_events
/rosout
/tf_static
```

MartyG-RealSense, 2023-05-05:
What happens if you change --remap for -p (parameter) in the ros2 run instruction.

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth:=true`

alonsnir, 2023-05-05:
@MartyG-RealSense, same result
```
$ ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth:=true
[INFO] [1683294979.118821373] [camera]: RealSense ROS v4.51.1
[INFO] [1683294979.118920667] [camera]: Built with LibRealSense v2.51.1
[INFO] [1683294979.118932958] [camera]: Running with LibRealSense v2.51.1
[INFO] [1683294979.130872095] [camera]: Device with serial number 829212071017 was found.

[INFO] [1683294979.130907525] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2 was found.
[INFO] [1683294979.130919778] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1683294979.131133313] [camera]: Device with port number 2-3 was found.
[INFO] [1683294979.131149194] [camera]: Device USB type: 3.2
[INFO] [1683294979.133103215] [camera]: getParameters...
[INFO] [1683294979.133316327] [camera]: JSON file is not provided
[INFO] [1683294979.133334517] [camera]: Device Name: Intel RealSense D435
[INFO] [1683294979.133345614] [camera]: Device Serial No: 829212071017
[INFO] [1683294979.133355625] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
[INFO] [1683294979.133365627] [camera]: Device FW version: 05.13.00.50
[INFO] [1683294979.133383028] [camera]: Device Product ID: 0x0B07
[INFO] [1683294979.133393027] [camera]: Sync Mode: Off
[INFO] [1683294979.160792745] [camera]: Stopping Sensor: Stereo Module
[INFO] [1683294979.215738829] [camera]: Starting Sensor: Stereo Module
[INFO] [1683294979.251686303] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683294979.251811857] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683294979.251858955] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1683294979.252663658] [camera]: Stopping Sensor: RGB Camera
[INFO] [1683294979.306751819] [camera]: Starting Sensor: RGB Camera
[INFO] [1683294979.313737553] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[INFO] [1683294979.314909811] [camera]: RealSense Node Is Up!
[WARN] [1683294979.370843437] [camera]: 
```
```
$ ros2 topic list
/color/camera_info
/color/image_raw
/color/metadata
/depth/camera_info
/depth/image_rect_raw
/depth/metadata
/extrinsics/depth_to_color
/extrinsics/depth_to_infra1
/extrinsics/depth_to_infra2
/imu
/infra1/camera_info
/infra1/image_rect_raw
/infra1/metadata
/infra2/camera_info
/infra2/image_rect_raw
/infra2/metadata
/parameter_events
/rosout
/tf_static
```

MartyG-RealSense, 2023-05-05:
And there is no difference if you reset the camera at launch with **initial_reset:=true** ?

**ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth:=true -p initial_reset:=true**

alonsnir, 2023-05-05:
@MartyG-RealSense 

initial_reset:=true ended with error

```
$ ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth:=true -p initial_reset:=true

[INFO] [1683303338.664052223] [camera]: Device Serial No: 829212071017
[INFO] [1683303338.664072935] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
[INFO] [1683303338.664091984] [camera]: Device FW version: 05.13.00.50
[INFO] [1683303338.664109110] [camera]: Device Product ID: 0x0B07
[INFO] [1683303338.664124353] [camera]: Sync Mode: Off
[INFO] [1683303338.694205478] [camera]: Stopping Sensor: Stereo Module
[INFO] [1683303338.746919449] [camera]: Starting Sensor: Stereo Module
[INFO] [1683303338.801524917] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683303338.801570048] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683303338.801580554] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1683303338.801770764] [camera]: Stopping Sensor: RGB Camera
[INFO] [1683303338.802672949] [camera]: Starting Sensor: RGB Camera
[INFO] [1683303338.804458204] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[INFO] [1683303338.804735504] [camera]: RealSense Node Is Up!
[WARN] [1683303338.868418038] [camera]: 
 05/05 19:15:40,494 ERROR [139642657560128] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
 05/05 19:15:40,494 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
 05/05 19:15:40,594 ERROR [139642657560128] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,594 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,695 ERROR [139642657560128] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,695 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,747 ERROR [139642967926336] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,747 ERROR [139642967926336] (error-handling.cpp:94) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,795 ERROR [139642657560128] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,796 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,896 ERROR [139642657560128] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,896 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,997 ERROR [139642657560128] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:40,997 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 05/05 19:15:41,097 ERROR [139642657560128] (librealsense-exception.h:52) acquire: Cannot open '/dev/video2 Last Error: No such file or directory
 05/05 19:15:41,097 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video2 Last Error: No such file or directory
 05/05 19:15:41,197 ERROR [139642657560128] (librealsense-exception.h:52) acquire: Cannot open '/dev/video2 Last Error: No such file or directory
 05/05 19:15:41,197 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video2 Last Error: No such file or directory
 05/05 19:15:41,297 ERROR [139642657560128] (librealsense-exception.h:52) acquire: Cannot open '/dev/video2 Last Error: No such file or directory
 05/05 19:15:41,297 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video2 Last Error: No such file or directory
 05/05 19:15:41,398 ERROR [139642657560128] (librealsense-exception.h:52) acquire: Cannot open '/dev/video2 Last Error: No such file or directory
 05/05 19:15:41,398 ERROR [139642657560128] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video2 Last Error: No such file or directory
[ERROR] [1683303341.489366129] [camera]: The device has been disconnected!
[INFO] [1683303341.489527754] [camera]: Stop Sensor: Stereo Module
[INFO] [1683303341.489553810] [camera]: Close Sensor. 
[INFO] [1683303341.490077107] [camera]: Close Sensor - Done. 
[INFO] [1683303341.490110890] [camera]: Stop Sensor: RGB Camera
[... cut ...]
 05/05 19:15:41,490 ERROR [139643269932608] (librealsense-exception.h:52) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 05/05 19:15:41,498 ERROR [139643269932608] (librealsense-exception.h:52) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 05/05 19:15:41,498 ERROR [139643269932608] (sensor.cpp:600) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
corrupted double-linked list
[ros2run]: Aborted

```

Now without the initial_reset, the node is up, but still same results for both ```ros2 launch``` and ```ros2 run```: no aligned_depth_to_color topic

alonsnir, 2023-05-06:
Hi @MartyG-RealSense 

I have followed your suggestion and uninstalled all RealSense packages and installed only the ROS wrapper. It didn't solve the issue yet.

```
$ ros2 run realsense2_camera realsense2_camera_node --ros-args --remap align_depth:=true
[INFO] [1683363822.130328401] [camera]: RealSense ROS v4.51.1
[INFO] [1683363822.130580102] [camera]: Built with LibRealSense v2.51.1
[INFO] [1683363822.130631431] [camera]: Running with LibRealSense v2.51.1
[INFO] [1683363822.147463985] [camera]: Device with serial number 829212071017 was found.

[INFO] [1683363822.147504990] [camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2 was found.
[INFO] [1683363822.147535437] [camera]: Device with name Intel RealSense D435 was found.
[INFO] [1683363822.147740261] [camera]: Device with port number 2-3 was found.
[INFO] [1683363822.147755460] [camera]: Device USB type: 3.2
[INFO] [1683363822.149509504] [camera]: getParameters...
[INFO] [1683363822.149737477] [camera]: JSON file is not provided
[INFO] [1683363822.149752682] [camera]: Device Name: Intel RealSense D435
[INFO] [1683363822.149761051] [camera]: Device Serial No: 829212071017
[INFO] [1683363822.149768463] [camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
[INFO] [1683363822.149776068] [camera]: Device FW version: 05.13.00.50
[INFO] [1683363822.149782991] [camera]: Device Product ID: 0x0B07
[INFO] [1683363822.149810633] [camera]: Sync Mode: Off
[WARN] [1683363822.165079035] [camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
 06/05 12:03:42,165 WARNING [140010863232576] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[INFO] [1683363822.174937012] [camera]: Stopping Sensor: Stereo Module
[INFO] [1683363822.178428915] [camera]: Starting Sensor: Stereo Module
[INFO] [1683363822.216896324] [camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683363822.216949111] [camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1683363822.216961508] [camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1683363822.217151488] [camera]: Stopping Sensor: RGB Camera
[INFO] [1683363822.268483358] [camera]: Starting Sensor: RGB Camera
[INFO] [1683363822.274831015] [camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[INFO] [1683363822.275876237] [camera]: RealSense Node Is Up!
[WARN] [1683363822.332718028] [camera]: 
```

```
$ ros2 topic list
/color/camera_info
/color/image_raw
/color/metadata
/depth/camera_info
/depth/image_rect_raw
/depth/metadata
/extrinsics/depth_to_color
/extrinsics/depth_to_infra1
/extrinsics/depth_to_infra2
/imu
/infra1/camera_info
/infra1/image_rect_raw
/infra1/metadata
/infra2/camera_info
/infra2/image_rect_raw
/infra2/metadata
/parameter_events
/rosout
/tf_static
```

MartyG-RealSense, 2023-05-06:
Please next try **align_depth.enable:=true** instead of align_depth:=true as that is the parameter name in the rs_launch.py file.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py#L54

alonsnir, 2023-05-06:
Hi @MartyG-RealSense this parameter did the trick :)

```
$ ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true
```

```
$ ros2 topic list
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
```

However ros2 run has failed
```
$ ros2 run realsense2_camera realsense2_camera_node --ros-args --remap align_depth.enable:=true
[ERROR] [1683399202.115807082] [rcl]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse remap rule: '--remap align_depth.enable:=true'. Error: Expected lexeme type 19, got 22 at index 11, at ./src/rcl/lexer_lookahead.c:244, at ./src/rcl/arguments.c:371
[ros2run]: Aborted
```

I guess we can close this issue though 

Thank you very much for helping out

MartyG-RealSense, 2023-05-06:
It's excellent to hear that adding .enable worked for you!

For ros2 run, you could try align_depth.enable with -p instead of --remap.

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth.enable:=true`

As you suggest, I will close the case.  Thanks very much for the update about your success!

closed 2023-05-06 by MartyG-RealSense

alonsnir, 2023-05-07:
@MartyG-RealSense I have tried the -p argument and just to confirm - it works properly

Thanks

MartyG-RealSense, 2023-05-07:
Thanks for the confirmation  :)

AlfretzNehemia, 2023-07-18:
Hi, right now I have try aligned_depth and it works. But, image from topic /aligned_depth_to_color/image_raw not shows color. It shows like depth/image_rect_raw.
[image]

AlfretzNehemia, 2023-07-18:
> Hi @AlfretzNehemia <https://github.com/AlfretzNehemia> What ROS launch
> instruction are you using, please?

ros2 launch realsense2_camera rs_launch.py align_depth:=true

MartyG-RealSense, 2023-07-18:
If you are using the 'ros2-development' wrapper, such as version 4.54.1, then the align command should be: 

align_depth.enable:=true