realsenseai/realsense-ros #2729: CMake couldn't find realsense2 in Linux

mericgeren, 2023-05-05:
Hi @MartyG-RealSense, 

Just switched to Linux 18.04LTS with kernel 5.3.0-59-generic and ROS Melodic. I have git cloned ddynamic and realsense-ros ( did git checkout 2.3.2 right after). Also, downloaded the zip file which contains the source code of librealsense v2.50.0 and extracted it directly to home directory. After that, created and moved into build directory at the root of librealsense. Then, using CMake 3.10.2 i built my project with the command: `cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
` As you suggested here: #6689. Then, followed steps of realsense-ros installation. When i input the command: `catkin_make clean` i got this error: 

> > 
> -- +++ processing catkin package: 'realsense2_camera'
> -- ==> add_subdirectory(realsense-ros/realsense2_camera)
> -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
> CMake Warning at realsense-ros/realsense2_camera/CMakeLists.txt:45 (find_package):
>   By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
>   has asked CMake to find a package configuration file provided by
>   "realsense2", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "realsense2"
>   (requested version 2.50.0) with any of the following names:
> 
>     realsense2Config.cmake
>     realsense2-config.cmake
> 
>   Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
>   "realsense2_DIR" to a directory containing one of the above files.  If
>   "realsense2" provides a separate development package or SDK, be sure it has
>   been installed.
> 
> 
> CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:49 (message):
>   
> 
>   
> 
>    Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/baran/catkin_ws/build/CMakeFiles/CMakeOutput.log".
> See also "/home/baran/catkin_ws/build/CMakeFiles/CMakeError.log".
> Makefile:1554: recipe for target 'cmake_check_build_system' failed
> make: *** [cmake_check_build_system] Error 1
> Invoking "make cmake_check_build_system" failed

Can you help me on this issue please?

MartyG-RealSense, 2023-05-05:
Hi @mericgeren  The error message states that **Intel RealSense SDK 2.0 is missing, please install it**.

After you used the CMake instruction to set up building:

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

There is a further step to input in the Build folder to compile librealsense that is not on the instructions at https://github.com/IntelRealSense/librealsense/issues/6689#issuecomment-648758973

`sudo make uninstall && make clean && make && sudo make install`

mericgeren, 2023-05-08:
Hi there,

Sorry for the late response. Just tried inputting `sudo make uninstall && make clean && make -j8 && sudo make install`. When i input that command i got a lot of warning messages with 

> warning: ISO C forbids conversion of object pointer to function pointer type [-Wpedantic]

But, after the completion of install i was able to open Realsense Viewer and view the stream coming from the camera. Then, i have entered `catkin_make clean` and run the command without any issues. Later, when i inputted ` catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release`  i got the following error: 

> ModuleNotFoundError: No module named 'em'
>     import em
> ModuleNotFoundError: No module named 'em'
>     import em
>     import em
> ModuleNotFoundError: No module named 'em'
> ModuleNotFoundError: No module named 'em'
> realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera_generate_messages_cpp.dir/build.make:72: recipe for target '/home/baran/catkin_ws/devel/include/realsense2_camera/Extrinsics.h' failed
> make[2]: *** [/home/baran/catkin_ws/devel/include/realsense2_camera/Extrinsics.h] Error 1
> make[2]: *** Waiting for unfinished jobs....
> realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera_generate_messages_cpp.dir/build.make:65: recipe for target '/home/baran/catkin_ws/devel/include/realsense2_camera/IMUInfo.h' failed
> make[2]: *** [/home/baran/catkin_ws/devel/include/realsense2_camera/IMUInfo.h] Error 1
> realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera_generate_messages_cpp.dir/build.make:79: recipe for target '/home/baran/catkin_ws/devel/include/realsense2_camera/Metadata.h' failed
> make[2]: *** [/home/baran/catkin_ws/devel/include/realsense2_camera/Metadata.h] Error 1
> realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera_generate_messages_cpp.dir/build.make:86: recipe for target '/home/baran/catkin_ws/devel/include/realsense2_camera/DeviceInfo.h' failed
> make[2]: *** [/home/baran/catkin_ws/devel/include/realsense2_camera/DeviceInfo.h] Error 1
> CMakeFiles/Makefile2:1997: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera_generate_messages_cpp.dir/all' failed
> make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera_generate_messages_cpp.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> [ 75%] Linking CXX shared library /home/baran/catkin_ws/devel/lib/libddynamic_reconfigure.so
> [ 75%] Built target ddynamic_reconfigure
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Error 2
> Invoking "make -j8 -l8" failed

MartyG-RealSense, 2023-05-08:
I have not seen the error **ModuleNotFoundError: No module named 'em'** before.  However, I found a case where a RealSense user who also experienced it resolved it by using the command **pip install empy**

[image]

mericgeren, 2023-05-08:
Installed empy with commands you have suggested. Now, when i input the command `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release` i got this error:

> CMakeFiles/Makefile2:1859: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
> make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Error 2
> Invoking "make -j8 -l8" failed

Thank you for your help.

MartyG-RealSense, 2023-05-08:
Using the -j parameter can cause a random crash of the build process on some computers, especially ones with a low-end configuration. You could try a smaller j-number such as -j4 or remove it and just use **make** instead of make -j8, though this results in a longer build time.  Also try it without -l8 as this parameter is usually not used in librealsense building.

mericgeren, 2023-05-08:
I just tried `sudo make uninstall && make clean && make -j4 && sudo make install` and `sudo make uninstall && make clean && make && sudo make install` after trying both when i input `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release` i get the same error.

mericgeren, 2023-05-08:
Hi there,

Deleted everything about ddynamic_reconfigure and relasense-ros in catkin workspace. Then, moved to build folder of librealsense installation and inputted the command `sudo make uninstall && make clean`. Then, inputted the command `sudo apt install ros-melodic-ddynamic-reconfigure` and i got this output:

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> The following NEW packages will be installed:
>   ros-melodic-ddynamic-reconfigure
> 0 upgraded, 1 newly installed, 0 to remove and 369 not upgraded.
> Need to get 65,4 kB of archives.
> After this operation, 272 kB of additional disk space will be used.
> Get:1 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-ddynamic-reconfigure amd64 0.3.2-1bionic.20221025.192508 [65,4 kB]
> Fetched 65,4 kB in 1s (66,3 kB/s)                          
> Selecting previously unselected package ros-melodic-ddynamic-reconfigure.
> dpkg: warning: files list file for package 'libopencv-samples' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-photo3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-stitching3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-contrib3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'python3-opencv' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-core3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-python' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-imgcodecs3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-videostab3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'opencv-licenses' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-objdetect3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-viz3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-dev' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-features2d3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-superres3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-ml3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-shape3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-imgproc3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-flann3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-video3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-calib3d3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-highgui3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-videoio3.2:amd64' missing; assuming package has no files currently installed
> (Reading database ... 373583 files and directories currently installed.)
> Preparing to unpack .../ros-melodic-ddynamic-reconfigure_0.3.2-1bionic.20221025.192508_amd64.deb ...
> Unpacking ros-melodic-ddynamic-reconfigure (0.3.2-1bionic.20221025.192508) ...
> Setting up ros-melodic-ddynamic-reconfigure (0.3.2-1bionic.20221025.192508) ...

MartyG-RealSense, 2023-05-08:
Which kind of computer are you installing on, please?  Is it a PC (desktop or laptop) or a single-board computer like Nvidia Jetson or Raspberry Pi?

mericgeren, 2023-05-08:
I am installing on a desktop pc.

MartyG-RealSense, 2023-05-08:
If you are installing librealsense and the wrapper separately from source code instead of installing them together at the same time from packages with **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** then once librealsense is installed, you should clone the ROS wrapper before installing ddynamic_reconfigure.  Did you perform this stage, please?

```
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..
```

mericgeren, 2023-05-08:
I haven't installed ROS wrapper yet. Now, trying installing ROS wrapper first.

mericgeren, 2023-05-08:
Just tried removing ddynamic_reconfigure with `sudo apt autoremove ros-melodic-ddynamic-reconfigure` and then, git cloning ROS wrapper with commands you mentioned above and then, installing ddynamic_reconfigure with command: `sudo apt install ros-melodic-ddynamic-reconfigure`. After inputting i got the follwing output:

> 
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> The following NEW packages will be installed:
>   ros-melodic-ddynamic-reconfigure
> 0 upgraded, 1 newly installed, 0 to remove and 369 not upgraded.
> Need to get 65,4 kB of archives.
> After this operation, 272 kB of additional disk space will be used.
> Get:1 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-ddynamic-reconfigure amd64 0.3.2-1bionic.20221025.192508 [65,4 kB]
> Fetched 65,4 kB in 2s (37,6 kB/s)                          
> Selecting previously unselected package ros-melodic-ddynamic-reconfigure.
> dpkg: warning: files list file for package 'libopencv-samples' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-photo3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-stitching3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-contrib3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'python3-opencv' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-core3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-python' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-imgcodecs3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-videostab3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'opencv-licenses' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-objdetect3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-viz3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-dev' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-features2d3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-superres3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-ml3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-shape3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-imgproc3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-flann3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-video3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-calib3d3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-highgui3.2:amd64' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-videoio3.2:amd64' missing; assuming package has no files currently installed
> (Reading database ... 373583 files and directories currently installed.)
> Preparing to unpack .../ros-melodic-ddynamic-reconfigure_0.3.2-1bionic.20221025.192508_amd64.deb ...
> Unpacking ros-melodic-ddynamic-reconfigure (0.3.2-1bionic.20221025.192508) ...
> Setting up ros-melodic-ddynamic-reconfigure (0.3.2-1bionic.20221025.192508) ...
>

mericgeren, 2023-05-08:
Managed to install most of the packages listed above. But, for packages have not installed yet. When try to install i get following messages:

> 
> package has no files currently installed
> dpkg: warning: files list file for package 'opencv-licenses' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv-dev' missing; assuming package has no files currently installed
> dpkg: warning: files list file for package 'libopencv' missing; assuming package has no files currently installed

MartyG-RealSense, 2023-05-08:
Are you able to install the packages manually with **sudo apt-get** instructions?

sudo apt-get install opencv-licenses
sudo apt-get install libopencv-dev
sudo apt-get install libopencv

mericgeren, 2023-05-08:
I wasn't able to. But, then, i deleted everything about realsense-ros in catkin workspace. Then, git cloned realsense-ros before i clone ddynamic_reconfigure and it fixed. But, just inputted `roslaunch realsense2_camera rs_camera.launch` and nothing happened.

Thank you all the help you have offered.

MartyG-RealSense, 2023-05-08:
You are very welcome.

Does it also not respond if you launch with **rs_rgbd.launch** instead of rs_camera.launch?  

To use rs_rgbd.launch, you should first install support for an RGBD launch with the command below.

`sudo apt-get install ros-melodic-rgbd-launch`

mericgeren, 2023-05-09:
After powering off and powering on the pc, i inputted `sudo apt-get install ros-melodic-rgbd-launch` and `oslaunch realsense2_camera rs_rgbd.launch`. Then, i got the following output:

> ... logging to /home/baran/.ros/log/d08075cc-ee3e-11ed-a7ed-309c234237f8/roslaunch-pppp-MS-7996-6140.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://127.0.0.1:35541/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /camera/realsense2_camera/accel_fps: -1
>  * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
>  * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
>  * /camera/realsense2_camera/align_depth: True
>  * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/allow_no_texture_points: False
>  * /camera/realsense2_camera/base_frame_id: camera_link
>  * /camera/realsense2_camera/calib_odom_file: 
>  * /camera/realsense2_camera/clip_distance: -1.0
>  * /camera/realsense2_camera/color_fps: -1
>  * /camera/realsense2_camera/color_frame_id: camera_color_frame
>  * /camera/realsense2_camera/color_height: -1
>  * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
>  * /camera/realsense2_camera/color_width: -1
>  * /camera/realsense2_camera/confidence_fps: 30
>  * /camera/realsense2_camera/confidence_height: 480
>  * /camera/realsense2_camera/confidence_width: 640
>  * /camera/realsense2_camera/depth_fps: -1
>  * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
>  * /camera/realsense2_camera/depth_height: -1
>  * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
>  * /camera/realsense2_camera/depth_width: -1
>  * /camera/realsense2_camera/device_type: 
>  * /camera/realsense2_camera/enable_accel: False
>  * /camera/realsense2_camera/enable_color: True
>  * /camera/realsense2_camera/enable_confidence: True
>  * /camera/realsense2_camera/enable_depth: True
>  * /camera/realsense2_camera/enable_fisheye1: False
>  * /camera/realsense2_camera/enable_fisheye2: False
>  * /camera/realsense2_camera/enable_fisheye: False
>  * /camera/realsense2_camera/enable_gyro: False
>  * /camera/realsense2_camera/enable_infra1: False
>  * /camera/realsense2_camera/enable_infra2: False
>  * /camera/realsense2_camera/enable_infra: False
>  * /camera/realsense2_camera/enable_pointcloud: False
>  * /camera/realsense2_camera/enable_pose: False
>  * /camera/realsense2_camera/enable_sync: True
>  * /camera/realsense2_camera/filters: 
>  * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
>  * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
>  * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
>  * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
>  * /camera/realsense2_camera/fisheye_fps: -1
>  * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
>  * /camera/realsense2_camera/fisheye_height: -1
>  * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
>  * /camera/realsense2_camera/fisheye_width: -1
>  * /camera/realsense2_camera/gyro_fps: -1
>  * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
>  * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
>  * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
>  * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 98 lines cut ...]
>  09/05 10:59:12,457 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,507 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,558 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,608 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,659 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,709 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,760 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,810 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,861 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:12,911 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
>  09/05 10:59:14,332 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:14,382 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:14,534 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
>  09/05 10:59:14,584 WARNING [140141955901184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

MartyG-RealSense, 2023-05-09:
The rs_rgbd launch file has depth-color alignment enabled by default, so please try disabling alignment with the launch instruction below to see whether it reduces the number of **control_transfer returned** warnings.

`roslaunch realsense2_camera rs_rgbd.launch align_depth:=false`

mericgeren, 2023-05-09:
Just tried inputting <roslaunch realsense2_camera rs_rgbd.launch align_depth:=false>. The pc doesn't freeze. But, this time i still get same error messages i got when i tried <roslaunch realsense2_camera rs_rgbd.launch>

MartyG-RealSense, 2023-05-09:
Is there any data displayed in RViz?  Sometimes streams will perform normally in RViz even when control_transfer returned warnings are generating continuously in the ROS log.

mericgeren, 2023-05-09:
When trying to ***Save Config*** a pop-up window appears which displays the message: "Failed to open /opt/ros/melodic/share/rviz/default.rviz for writing."

MartyG-RealSense, 2023-05-09:
The link below suggests dealing with this error by selecting **Save Config As** instead of Save Config and saving a configuration to a directory inside your catkin_ws folder.

https://get-help.robotigniteacademy.com/t/cannot-save-rviz-config-file/13091

MartyG-RealSense, 2023-05-10:
That's great to hear that you found a solution!

MartyG-RealSense, 2023-05-10:
You are very welcome.  I don't mind at all.  :)  Thanks very much for the update!