realsenseai/realsense-ros #2730: Low fps on D345i

nadavis, 2023-05-07:
Hi,
Camera: D435i
OS: raspberry pi 4B arm64 : ubuntu 22.04 server
laptop : ubuntu 22.04 desktop
ROS version : both ROS2 Humble

i ran the ros2 launch realsense2_camera rs_launch.py with defult config and i am getting low fps around 10 on my laptop and on RPI i am getting 1fps.
what fps i should expect?

any advice?

moreover i am getting the follow error after i launched the ros script.

>>>ros2 launch robot rs_launch.py 
[INFO] [launch]: All log files can be found below /home/nadav/.ros/log/2023-05-07-13-36-00-719276-XPS-49027
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [49028]
[realsense2_camera_node-1] [INFO] [1683455761.554884875] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1683455761.554962378] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683455761.554967965] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1]  07/05 13:36:01,555 WARNING [140329974838848] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  07/05 13:36:01,555 WARNING [140329974838848] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  07/05 13:36:01,555 WARNING [140329974838848] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1] [INFO] [1683455761.573883258] [camera.camera]: Device with serial number 153122077556 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1683455761.573968417] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/video4linux/video4 was found.
[realsense2_camera_node-1] [INFO] [1683455761.573989777] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1683455761.574452132] [camera.camera]: Device with port number 3-2 was found.
[realsense2_camera_node-1] [INFO] [1683455761.574472697] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1683455761.574488704] [camera.camera]: Device 153122077556 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1]  07/05 13:36:01,572 WARNING [140329974838848] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  07/05 13:36:01,572 WARNING [140329974838848] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1] [INFO] [1683455761.577585987] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1683455761.577807150] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1683455761.577824245] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1683455761.577835615] [camera.camera]: Device Serial No: 153122077556
[realsense2_camera_node-1] [INFO] [1683455761.577845844] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/video4linux/video4
[realsense2_camera_node-1] [INFO] [1683455761.577857067] [camera.camera]: Device FW version: 05.14.00.00
[realsense2_camera_node-1] [INFO] [1683455761.577866883] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1683455761.577875542] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1]  07/05 13:36:01,572 WARNING [140329974838848] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1] [ERROR] [1683455761.596006770] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [1683455761.596625805] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [1683455761.597158609] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [1683455761.597750951] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [INFO] [1683455761.599017955] [camera.camera]: Set ROS param depth_module.profile to default: 640x480x15
[realsense2_camera_node-1] [WARN] [1683455761.600408252] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1683455761.601486276] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1683455761.601663295] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1683455761.601699382] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [WARN] [1683455761.604642438] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1683455761.605262576] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683455761.607008882] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683455761.660875849] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1683455761.661404645] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683455761.664304928] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683455761.669581680] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1683455761.670293681] [camera.camera]: RealSense Node Is Up!

MartyG-RealSense, 2023-05-07:
Hi @nadavis  Whilst slower performance on Raspberry Pi should be expected, you should be able to achieve at least 15 FPS on a laptop.

The Auto Exposure Limit and the Auto Gain Limit errors are occuring because your camera firmware driver is 5.14.0.0 but your SDK version is 2.51.1.  The 5.14.0.0 firmware is designed for SDK **2.53.1** and will cause these errors if used on an older SDK.  Downgrading your firmware version to **5.13.0.50** should resolve the errors.  You can then check whether performance has improved afterwards.

nadavis, 2023-05-07:
Many thanks for your quick replay!
i update the firmware as you recommended
[image]
Now when i run on laptop i got 15 fps but with some warning as follow. but when i run of rpi there is no improvement, still 1 fps and less, moreover i got huge delay any thought?

[INFO] [launch]: All log files can be found below /home/nadav/.ros/log/2023-05-07-14-30-55-503615-XPS-53940
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [53941]
[realsense2_camera_node-1] [INFO] [1683459055.994308234] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1683459055.994384234] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683459055.994389664] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1]  07/05 14:30:55,994 WARNING [140709586335296] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  07/05 14:30:55,994 WARNING [140709586335296] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  07/05 14:30:55,994 WARNING [140709586335296] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1] [INFO] [1683459056.011075519] [camera.camera]: Device with serial number 153122077556 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1683459056.011138424] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/video4linux/video4 was found.
[realsense2_camera_node-1] [INFO] [1683459056.011153021] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1683459056.011473871] [camera.camera]: Device with port number 3-2 was found.
[realsense2_camera_node-1] [INFO] [1683459056.011489240] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1683459056.011502027] [camera.camera]: Device 153122077556 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1]  07/05 14:30:56,010 WARNING [140709586335296] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  07/05 14:30:56,010 WARNING [140709586335296] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1] [INFO] [1683459056.013781471] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1683459056.013944088] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1683459056.013956225] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1683459056.013965403] [camera.camera]: Device Serial No: 153122077556
[realsense2_camera_node-1] [INFO] [1683459056.013973315] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/video4linux/video4
[realsense2_camera_node-1] [INFO] [1683459056.013982183] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1683459056.013989813] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1683459056.013996845] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1]  07/05 14:30:56,010 WARNING [140709586335296] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1] [INFO] [1683459056.037666259] [camera.camera]: Set ROS param depth_module.profile to default: 640x480x15
[realsense2_camera_node-1] [WARN] [1683459056.039556610] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1683459056.040981904] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1683459056.041225409] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1683459056.041254781] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [WARN] [1683459056.044694611] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1683459056.045308952] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683459056.047053177] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683459056.100728926] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1683459056.101288874] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683459056.103205132] [camera.camera]: Starting Sensor: RGB Camera
[... cut ...]
[realsense2_camera_node-1] [WARN] [1683459056.287001790] [camera.camera]: 
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [1683459194.085875475] [rclcpp]: signal_handler(signum=2)
[realsense2_camera_node-1] [INFO] [1683459194.090973012] [camera.camera]: Stop Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683459194.091028381] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1683459194.093979041] [camera.camera]: Close Sensor - Done.

nadavis, 2023-05-07:
BTW i tried to disable the depth camera and still very low fps...

MartyG-RealSense, 2023-05-07:
Most of the **WARN** messages are ones that can be ignored, except for the busnum/devnum ones.  This is a very rare error that can indicate that the USB port is supplying the camera with insufficient power.  It only usually occurs if the USB port on the computer that the camera is attached to is a **USB OTG** micro-sized port rather than a full-size USB port. 

Is the above log from the Pi 4?  If it is then a Pi 4 has a USB 3 port and so the USB connection should not be being detected as the slower USB 2.1.  It could be mis-detected as USB 2 if the camera is connected to a USB 2 hub or you are using your own choice of USB cable instead of the official one supplied with the camera and your cable is a USB 2 one instead of a USB 3 cable.

You could try adding **initial_reset:=true** to your ros2 launch instruction to reset the camera at launch to see whether the camera is detected as a USB 3 conneciton.

nadavis, 2023-05-07:
the following is the output from rpi
i used the cable i got with the camera on rpi (on my laptop my own cable)
but still very low performance, i set me launch file with initial reset:=true , but it does not help
what fps i should expect from rpi? and what is the delay i should expect from rpi?

[INFO] [launch]: All log files can be found below /home/nadav/.ros/log/2023-05-07-15-16-41-562755-nadav-rpi-3944
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [3945]
[realsense2_camera_node-1] [INFO] [1683461802.420455273] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1683461802.420750432] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683461802.420821087] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683461802.520773681] [camera.camera]: Device with serial number 153122077556 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1683461802.520976479] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1683461802.521040282] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1683461802.522391506] [camera.camera]: Device with port number 1-1.1 was found.
[realsense2_camera_node-1] [INFO] [1683461802.522539391] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1683461802.522600471] [camera.camera]: Device 153122077556 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1683461802.522672756] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1683461808.668348438] [camera.camera]: Device with serial number 153122077556 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1683461808.668516509] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1683461808.668578441] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1683461808.669593204] [camera.camera]: Device with port number 1-1.1 was found.
[realsense2_camera_node-1] [INFO] [1683461808.669751978] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1683461808.669812780] [camera.camera]: Device 153122077556 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1683461808.679265139] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1683461808.680279254] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1683461808.680387172] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1683461808.680441843] [camera.camera]: Device Serial No: 153122077556
[realsense2_camera_node-1] [INFO] [1683461808.680493756] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1683461808.680542816] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1683461808.680586950] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1683461808.680629398] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1683461808.738401746] [camera.camera]: Set ROS param depth_module.profile to default: 640x480x15
[realsense2_camera_node-1] [WARN] [1683461808.754868325] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1683461808.761483934] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1683461808.763152407] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1683461808.763400153] [camera.camera]: Set ROS param accel_fps to default: 100

MartyG-RealSense, 2023-05-07:
At 640x480 resolution I would hope for 15 FPS for depth and color on a Pi 4.  It is not clear why the USB connection is being detected as 2.1 if the camera is plugged into the USB 3 port on the Pi.

Which method did you use to install the SDK and wrapper on Pi and Ubuntu 22.04 please?

nadavis, 2023-05-07:
Many thanks, 
i chanched the cable. and now by sure is usb3 as you can see in the output.
but i did not get any frame now
i saw that the topic exist, but the frames not show in rviz 

[INFO] [launch]: All log files can be found below /home/nadav/.ros/log/2023-05-07-16-51-06-765081-nadav-rpi-2730
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [2731]
[realsense2_camera_node-1] [INFO] [1683467469.432376713] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1683467469.432916617] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683467469.433062358] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1683467469.576599005] [camera.camera]: Device with serial number 153122077556 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1683467469.576782029] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1683467469.576844649] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1683467469.578209160] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1683467469.578367644] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1683467469.617551532] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1683467469.621805087] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1683467469.621977572] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1683467469.622076085] [camera.camera]: Device Serial No: 153122077556
[realsense2_camera_node-1] [INFO] [1683467469.622155596] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1683467469.622236236] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1683467469.622311171] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1683467469.622383736] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1683467469.699575212] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1683467469.720716284] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1]  07/05 16:51:09,726 WARNING [281472602532064] (backend-v4l2.cpp:1444) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1683467469.746692376] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1683467469.749679784] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1683467469.750025774] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1683467469.796445272] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683467469.825575794] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1683467469.889106840] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1683467469.889565325] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683467469.895997924] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1683467469.913888831] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1683467469.916582831] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1683467469.958335573] [camera.camera]:

MartyG-RealSense, 2023-05-07:
How does it perform if you launch with **ros2 run** instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node`

nadavis, 2023-05-08:
it seem the fps is very low, this is the output i got:
[image]

MartyG-RealSense, 2023-05-08:
You could test whether it is the color stream that is dragging down the FPS by adding **enable_color:=false** to your launch instruction so that only depth is enabled.

nadavis, 2023-05-08:
Unfortunately this is not help
[image]

BTW 
Unfortunately this is effecting just on ros2 launch and not ros2 run. so now i start getting 1 fps instead  [WARNING: topic [/camera/depth/image_rect_raw]]
Any idea what can cause to low performance?

MartyG-RealSense, 2023-05-08:
The causes of very slow performance on Pi  for some RealSense ROS users are not clear, unfortunately.  A Pi user at https://github.com/IntelRealSense/realsense-ros/issues/1858 with a Pi and PC had similar FPS to you (4 on Pi and 14 on PC).

In that case I highlighted how a user at https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-839616982 achieved improved performance by setting RViz to use compressed data.

nadavis, 2023-05-08:
I do not have 4fps, i have less than 1, more likely 1 frame to 3 second. And huge delay.
is it common?

MartyG-RealSense, 2023-05-08:
FPS sometimes goes as low as 3 FPS but it is not common for it to be as low as you are experiencing.  When talking about huge delay I am reminded or a case at https://github.com/IntelRealSense/realsense-ros/issues/1581 where huge delay was resolved by setting **global_time_enabled** to False (it is true by default on RealSense 400 Series camera models).

MartyG-RealSense, 2023-05-11:
Yes, that is the correct instruction.  So it may be that global time being true is not a factor in your particular cases's delay.

MartyG-RealSense, 2023-05-16:
Hi @nadavis  Do you require further assistance with this case, please?  Thanks!

nadavis, 2023-05-16:
Thank you so much for your invaluable assistance! I genuinely appreciate your help...

MartyG-RealSense, 2023-05-21:
You are very welcome!

As you have not requested further assistance, I will close this case.  Thanks again!

closed 2023-05-21 by MartyG-RealSense