realsenseai/realsense-ros #2731: How to make the subscribe node to receive "/camera/depth/image_rect_raw"?

kevinDrawn, 2023-05-08:
Hello. I have question about the topic named '/camera/depth/image_rect_raw' 
This topic is published when i launch this 'rs_camera.launch'.

 I want to receive the topic and transfer to depth_frame for mapping x,y pixel coordinate to the world coordinate (x,y,z).

ex) The mapping function is it. (Depth -> Relative Coordinate). This functon's parameter requires the depth_frame.  
`def get_relative_xyz(depth_frame, pixel_x, pixel_y, depth_scale=10):
    depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
    depth_pixel = [pixel_x, pixel_y]
    depth_image_point = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_pixel, depth_frame.get_distance(pixel_x, pixel_y))
    depth_relative_point = np.array([depth_image_point[0]*depth_scale, depth_image_point[1]*depth_scale, depth_image_point[2]*depth_scale])
    return depth_relative_point` 

And This is my code.

`#!/usr/bin/env python
#-*- coding:utf-8 -*- 

import rospy
from sensor_msgs.msg import Image
from cv_bridge import CvBridge
import cv2
import pyrealsense2 as rs
import numpy as np

x = 250
y = 250
i = 0
depth_relative_point=0

bridge = CvBridge()

pipeline = rs.pipeline()
serial_number = "128422271045" # it is the Serial number of my Depth camera.
config = rs.config()
config.enable_device(serial_number)
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30) #launch file 
pipeline.start(config)

def Image_callback(image):
	global x, y, depth_relative_point
	depth_array = np.array(image.data, dtype=np.uint16)
	depth_frame = rs.frame(depth_array)
	depth_relative_point = get_relative_xyz(depth_frame,x,y,depth_scale=10)
	print(depth_relative_point[0])
		
def get_relative_xyz(depth_frame, pixel_x, pixel_y, depth_scale=10):
    depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
    depth_pixel = [pixel_x, pixel_y]
    depth_image_point = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_pixel, depth_frame.get_distance(pixel_x, pixel_y))
    depth_relative_point = np.array([depth_image_point[0]*depth_scale, depth_image_point[1]*depth_scale, depth_image_point[2]*depth_scale])
    return depth_relative_point
			

rospy.init_node('sub')
rospy.Subscriber('/camera/depth/image_rect_raw',Image, Image_callback) 
	
rospy.spin()

pipeline.stop()
cv2.destroyAllWindows()		
`

Whenever  I run this code, No device error  occurred. 

`Traceback (most recent call last):
  File "/home/park/catkin_ws/src/robot/src/sub.py", line 27, in <module>
    pipeline.start(config)
RuntimeError: No device connected`

Realsense Camera D405.

kevinDrawn, 2023-05-12:
Thank you a lot!

MartyG-RealSense, 2023-05-12:
You are very welcome!  :)

closed 2023-05-25 by MartyG-RealSense