realsenseai/realsense-ros #2736: Humble: installed Linorobot2_ws  on Ros2 humble, buut no support for linorobot2_hardware.

hemsinghb, 2023-05-12:
As there is no ROS2 distro : humble supporting linorobot2_hardware ,  I tried with galactice hardware which is on github , but failed ,  while launching slam.launch ( ERROR : lookup would require extrapolation into the future. Requested time xxxxxx 
 but the latest data is at yyyyy   , when looking up transform from trame to frame [map]) error is appearing and on rviz2 map robot not moving , but rotation is updating on rviz2. Please guide further.

MartyG-RealSense, 2023-05-12:
Hi @hemsinghb  This support forum is for RealSense depth cameras and using them with ROS.  If you are not using RealSense cameras then I recommend posting your question on the official **ROS Answers** support forum at the link below.  Thanks!

https://answers.ros.org/questions/

hemsinghb, 2023-05-16:
ok , noted. Thx.

closed 2023-05-16 by MartyG-RealSense