realsenseai/realsense-ros #2737: D405 and ROS1

BasmaAB, 2023-05-12:
I'm trying to use ROS1 with D405 but I got the following error:

[FATAL] [1683904824.861121730]: Unsupported device! Product ID: 0x0B5B
[camera/realsense2_camera_manager-1] process has died [pid 66274, exit code 1, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/basma/.ros/log/7e5eb514-f0d8-11ed-850f-23e21b29427c/camera-realsense2_camera_manager-1.log].
log file: /home/basma/.ros/log/7e5eb514-f0d8-11ed-850f-23e21b29427c/camera-realsense2_camera_manager-1*.log

MartyG-RealSense, 2023-05-12:
Hi @BasmaAB  The RealSense ROS1 wrapper does not support D405 by default but a RealSense user created a custom version of the wrapper with D405 support that can be built instead of the official Intel wrapper.  Details can be found at https://github.com/IntelRealSense/realsense-ros/issues/2661#issuecomment-1474932181

The RealSense ROS2 wrapper **ros2-development** has D405 support.

BasmaAB, 2023-05-12:
I have question about how to update the file (constants.h) as there is no permission to do so.

MartyG-RealSense, 2023-05-12:
The creator of the custom wrapper advises at https://github.com/IntelRealSense/realsense-ros/pull/2297#issuecomment-1477025502 that constants.h is located at **include/realsense2_camera/constants.h**

In the directory structure of the official ROS1 wrapper the file is located at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy/realsense2_camera/include/realsense2_camera

BasmaAB, 2023-05-12:
I had error in CMake after downloading the GitHub clone. is there is any other things that I need to download?

MartyG-RealSense, 2023-05-13:
If you are using the GitHub clone that already has D405 support built into it then you should not need to download it as you will not need to make edits to its source code.  Instead, it should be sufficient just to clone it.

https://github.com/rjwb1/realsense-ros#step-2-install-intel-realsense-ros-from-sources

The instructions though are a copy of the official Intel ones rather than tailored to the rjwb1 fork.  So I would think that you would follow the source code installation instructions but use the clone instruction below to clone the rjwb1 fork instead of the official wrapper.

`git clone https://github.com/rjwb1/realsense-ros.git`

BasmaAB, 2023-05-15:
Hello @MartyG-RealSense . I still get the same error ([FATAL] [1684144583.821993612]: Unsupported device! Product ID: 0x0B5B) is there is anything I can do?

MartyG-RealSense, 2023-05-15:
If building the modified wrapper or performing the change to the 'constants.h' file described at https://github.com/IntelRealSense/realsense-ros/issues/2582#issuecomment-1371133133 do not work then there is nothing else that can be done except to use the ROS2 wrapper ros2-development instead of ROS1, unfortunately.

BasmaAB, 2023-05-15:
Noted. Thank you so much.

closed 2023-05-15 by BasmaAB

zhaianbang, 2025-06-30:
> Hello [@MartyG-RealSense](https://github.com/MartyG-RealSense) . I still get the same error ([FATAL] [1684144583.821993612]: Unsupported device! Product ID: 0x0B5B) is there is anything I can do?您好。我仍然收到相同的错误 ([FATAL] [1684144583.821993612]: 不支持的设备！产品 ID：0x0B5B)，请问该怎么办？

mkdir -p your_file_path/catkin_ws/src
cd your_file_path/catkin_ws/src/

git clone --branch development https://github.com/rjwb1/realsense-ros.git 

catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install

echo "source your_file_path/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

Tested and verified. If you encounter any issues, try executing all the above commands twice.
亲测有效，如果有问题，执行两次上面的所有指令。

MartyG-RealSense, 2025-06-30:
Thanks so much @zhaianbang for sharing your experience of what worked for you!

vocdex, 2025-06-30:
@zhaianbang Thanks! How exactly do you launch d405 camera after catkin build?

zhaianbang, 2025-07-01:
> [@zhaianbang](https://github.com/zhaianbang) Thanks! How exactly do you launch d405 camera after catkin build?

roslaunch realsense2_camera rs_camera.launch align_depth:=true color_width:=640 color_height:=480 depth_width:=640 depth_height:=480 depth_fps:=30  color_fps:=30