realsenseai/realsense-ros #2738: realsense d435i unsupported device!

fomboo, 2023-05-13:
I want to subscribe to topics infra1 and infra2, but they are not in rostopic list when I run roslaunch, how to solve the problem?
//
rosdistro: noetic 
ubuntu20.04
RealSense ROS v2.3.2
LibRealSense v2.50.0
Running with LibRealSense v2.50.0
//
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 76 lines cut ...]
[ INFO] [1683966934.795997226]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1683966934.796058519]: num_filters: 0
[ INFO] [1683966934.796079046]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1683966937.954528150]: Done Setting Dynamic reconfig parameters.
[ INFO] [1683966937.955171671]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ WARN] [1683966937.955234103]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ WARN] [1683966937.955284891]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ INFO] [1683966937.956404317]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1683966937.956440484]: gyro stream is enabled - fps: 200
[ INFO] [1683966937.956454655]: accel stream is enabled - fps: 63
[ INFO] [1683966937.956478829]: setupPublishers...
[ INFO] [1683966937.958563245]: Expected frequency for depth = 30.00000
[ INFO] [1683966937.984087729]: Expected frequency for color = 30.00000
[ INFO] [1683966938.002864624]: Start publisher IMU
[ INFO] [1683966938.003639699]: setupStreams...
 13/05 04:35:38,151 WARNING [140128695817984] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1683966938.427173558]: 
[ WARN] [1683966938.427265169]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 13/05 04:35:38,536 WARNING [140128467019520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1683966938.586494915]: SELECTED BASE:Depth, 0
[ INFO] [1683966938.593885904]: RealSense Node Is Up!

MartyG-RealSense, 2023-05-13:
Hi @fomboo  The infrared topics are disabled by default in the RealSense ROS wrapper.  You can enable them by adding to your roslaunch instruction the commands below.

**enable_infra1:=true enable_infra2:=true**

For example:

`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true`

fomboo, 2023-05-13:
UMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: True
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
[... 68 lines cut ...]
[ INFO] [1683983259.385104147]: num_filters: 0
[ INFO] [1683983259.385137178]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1683983262.739109568]: Done Setting Dynamic reconfig parameters.
[ INFO] [1683983262.739682983]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ WARN] [1683983262.739756416]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ WARN] [1683983262.739803315]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ INFO] [1683983262.740878431]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1683983262.740915212]: gyro stream is enabled - fps: 200
[ INFO] [1683983262.740931581]: accel stream is enabled - fps: 63
[ INFO] [1683983262.740956725]: setupPublishers...
[ INFO] [1683983262.743301431]: Expected frequency for depth = 30.00000
[ INFO] [1683983262.773589031]: Expected frequency for color = 30.00000
[ INFO] [1683983262.790984169]: Start publisher IMU
[ INFO] [1683983262.791642955]: setupStreams...
 13/05 09:07:42,931 WARNING [140459467532032] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1683983263.247671788]: 
[... cut ...] , error: Resource temporarily unavailable, number: 11
[ INFO] [1683983263.379761508]: SELECTED BASE:Depth, 0
[ INFO] [1683983263.390360843]: RealSense Node Is Up!
 13/05 09:07:43,451 WARNING [140459238217472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/05 09:07:43,502 WARNING [140459238217472] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

fomboo, 2023-05-13:
~$ rostopic list
/camera/accel/imu_info
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/gyro/imu_info
/camera/imu
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

fomboo, 2023-05-13:
: (

fomboo, 2023-05-13:
dont know why stream configuration is not supported LOL 
[ WARN] [1683983262.739756416]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ WARN] [1683983262.739803315]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ INFO] [1683983262.740878431]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8

fomboo, 2023-05-13:
Dont know why stream configuration is not supported
[ WARN] [1683983262.739756416]: Given stream configuration is not supported by the device! Stream: Infrared, Stream Index: 1, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ WARN] [1683983262.739803315]: Given stream configuration is not supported by the device! Stream: Infrared, Stream Index: 2, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ INFO] [1683983262.740878431]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8

fomboo, 2023-05-13:
[image]

MartyG-RealSense, 2023-05-13:
Next, please add to your roslaunch instruction **infra_rgb:=false** to make sure that the launch is not trying to use the infrared sensors for RGB instead of infrared.

fomboo, 2023-05-14:
thank uuuuuuuuuuuuuuu   QAQ
[image]

MartyG-RealSense, 2023-05-14:
You are very welcome.  :)

fomboo, 2023-05-14:
Emm one more question，my vins_fusion node is closed ，how to activate it？

[image]
[image]

MartyG-RealSense, 2023-05-14:
I am not familiar with fast_drone_250, unfortunately.  

https://github.com/ZJU-FAST-Lab/Fast-Drone-250

It does have an English language instruction guide though that covers the same steps as the instructions on the front page of the fast_drone_250 repository.

https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/master/readme_en.md

It does not explain though how to use fast_drone_250.launch.

There is a guide to using a RealSense camera with VINS and fast_drone_250 here:

https://blog.csdn.net/weixin_41967369/article/details/125183678

I have posted below images of an English translated version captured from my browser.

[image]

[image]

[image]

fomboo, 2023-05-14:
okkk, thank u anyway : )

closed 2023-05-14 by fomboo