realsenseai/realsense-ros #2740: Colour mismatching while extracting .bag images

SingCheng, 2023-05-13:
Hi,

I am new to handling the bag file, I did use the [bag_to_images.py](https://gist.github.com/wngreene/835cda68ddd9c5416defce876a4d7dd9) to extract images from bag file.

However, when I review the images, I did find that the colour is mismatched

Original:
[image]

After extracted:
[image]

Therefore, I check the info about the bag file and I saw
[image]

For the code that I have used, I provided a few options for colour or pixel depth conversions, is any option suitable for us below?

- mono8: CV_8UC1, grayscale image
- mono16: CV_16UC1, 16-bit grayscale image
- bgr8: CV_8UC3, color image with blue-green-red color order
- rgb8: CV_8UC3, color image with red-green-blue color order
- bgra8: CV_8UC4, BGR colour image with an alpha channel
- rgba8: CV_8UC4, RGB colour image with an alpha channel

Furthermore, **if none of the above options is suitable, are we required to use the metadata as 'supplementary' data?**

Cheers

MartyG-RealSense, 2023-05-13:
Hi @SingCheng  When using a RealSense RGB image with OpenCV, the **BGR8** format should be used for the RealSense RGB stream instead of RGB8 or the color will be reversed, producing effects like the one that you experienced.  This is because BGR8 is the default color space for OpenCV.

MartyG-RealSense, 2023-05-21:
Hi @SingCheng  Do you require further assistance with this case, please?  Thanks!

SingCheng, 2023-05-21:
Hi @MartyG-RealSense, the problems have been solved, thank you.

closed 2023-05-21 by SingCheng