realsenseai/realsense-ros #2741: The z-axis of pointcloud2 in the camera_link frame when displayed in rviz

TongZhangTHU, 2023-05-14:
Hello,

Why is the z-axis of pointcloud2 (from /camera/depth/color/points topic) in the camera_link frame pointing upwards instead of towards the camera's front（which I think should be) when displayed in rviz？

MartyG-RealSense, 2023-05-14:
Hi @TongZhangTHU  The ROS standard reference frame is for Z to be up, whilst on RealSense cameras Z is forward.  The RealSense ROS wrapper converts RealSense's coordinate system to the ROS standard.

TongZhangTHU, 2023-05-14:
@MartyG-RealSense Thanks for your reply. But I still don't know whether the camera_link frame should be (X- right, Y- down, Z- forward) or (X- forward, Y- left, Z- up)? 

If the camera_link frame is RealSense's coordinate system, why when I use camera_link frame as fixed frame in rviz,  is still shows  Z to be up?

MartyG-RealSense, 2023-05-14:
The camera data of the RealSense SDK librealsense) which the RealSense ROs compatibility wrapper is not in the ROS standard format and so the wrapper translates the SDK's data into a ROS-compatible structure.  So you do not need to think about the camera's Z-depth being on the forward axis, as the ROS wrapper automatically makes the adjustments required for the data to be correctly understood by ROS.

TongZhangTHU, 2023-05-14:
@MartyG-RealSense 

The reason I ask this question is that after I calibrated the RealSense camera extrinsics relative to the robot,  I visualized the point cloud in the robot frame with rviz /camera/depth/color/points topic (I have published the extrinsics TF), but found that the point cloud is in the wrong position and looks rotated in some way,  I suspect that it is caused by the wrong coordinate system. I wonder if you have any suggestions for my problem?

TongZhangTHU, 2023-05-14:
[image]

TongZhangTHU, 2023-05-14:
The iamge of the problem is as follows
[image]

MartyG-RealSense, 2023-05-14:
My understanding is that camera_link should correspond to the center of the camera's left infrared sensor (which is the origin point of depth) and base_link correspond to the tripod screw-thread hole on the bottom of the camera where a camera can be mounted onto a robot arm, as discussed at https://github.com/IntelRealSense/realsense-ros/issues/2700

MartyG-RealSense, 2023-05-15:
If you have launched with the pointcloud filter enabled and Fixed Frame is set to **camera_link** in Rviz - like in the example in the official ROS1 wrapper documentation at the link below - then the pointcloud should be published to /camera/depth/color/points and displayed correctly in Rviz.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#point-cloud

TongZhangTHU, 2023-05-16:
@MartyG-RealSense I find I can transform the PointCloud from (x, y, z) to (-y,-z, x) while not changing the frame so that it can be visualized as I expected. 

My implementation is to subscribe /camera/depth/color/points topic and then do the transformation for each point and then publish a transformed topic, but I found this runs slowly. 

Is there a better way to achieve the above requirements? Do you have any idea how to transform it efficiently?

Thank you for your help!

MartyG-RealSense, 2023-05-21:
Hi @TongZhangTHU  Do you require further assistance with this case, please?  Thanks!

TongZhangTHU, 2023-05-21:
@MartyG-RealSense I have found a way to solve the problems. Thank you for your kind help!

closed 2023-05-21 by TongZhangTHU

MartyG-RealSense, 2023-05-21:
You are very welcome, @TongZhangTHU - I am pleased to hear that you achieved a solution.  Thanks very much for the update!