realsenseai/realsense-ros #2743: cannot run rtabmap with RealSense D455

c56589796, 2023-05-17:
i have problems running rtabmap with my realsense d455

here is my launch file

<launch>
    
    <arg name="device_type"    		      default="D455"/>	
    <arg name="serial_no"    			  default="" />
    <arg name="camera"                    default="camera"/>		
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="device_type"             value="$(arg device_type)"/>
        <arg name="serial_no"               value="$(arg serial_no)"/>
        <arg name="camera"                  value="$(arg camera)"/>
        <arg name="clip_distance"           value="$(arg clip_distance)"/>
        <arg name="enable_pointcloud"       value="true"/>
	<arg name="initial_reset"	    value="true"/>
    </include>

    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">           
 	 <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/$(arg camera)/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="$(arg camera)_link"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/camera/gyro/imu_info"/>
            <arg name="rgb_topic"          value="/$(arg camera)/color/image_raw"/>
            <arg name="camera_info_topic"  value="/$(arg camera)/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>
</launch>

in rs_camera.launch   I have enabled accel and gyro

and terminator constanstly tell me that:

17/05 14:33:56,317 WARNING [140055687440128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1684305238.375165693]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /camera/gyro/imu_info \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info
 17/05 14:33:59,369 WARNING [140055687440128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1684305243.375312084]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /camera/gyro/imu_info \
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info

when I rostopic hz, all can give a output but /camera/gyro/imu_info

any one can give any help will be great, thank u very much!

MartyG-RealSense, 2023-05-17:
Hi, what computer / computing device are you using, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc).  Thanks!

c56589796, 2023-05-17:
@MartyG-RealSense hello,I 'm using desktop.Thanks！

MartyG-RealSense, 2023-05-17:
Do you see any improvement if you add the commands below onto the end of your rs_camera.launch instruction?

`enable_sync:=true initial_reset:=true unite_imu_method:=linear_interpolation`

c56589796, 2023-05-17:
@MartyG-RealSense No,still telling me the same mistake:
/rtabmap/rtabmap subscribed to (approx sync):
/camera/gyro/imu_info
/camera/color/image_raw
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info

And I cannot find odom and map in rqt_tf_tree.Did I subscribe the false odom topic?Thanks!

MartyG-RealSense, 2023-05-17:
The log above looks okay aside from the **control_transfer returned** warnings that can indicate a serious communication prolem between computer and camera if they are generating continuously.

Do they still occur if you use **unite_imu_method:=copy** instead of setting it to linear_interpolation

MartyG-RealSense, 2023-05-17:
A continuously generating control_transfer returned warning is usually a bad thing.  Sometimes though, streams still perform normally even though the warning is constantly appearing.  

What happens if you only launch with **opensource_tracking.launch** (which includes rtabmap instructions) and do not use the rtabmap launch instruction?

`roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

c56589796, 2023-05-18:
@MartyG-RealSense Hello,I tried and now I can get a map in rviz, but still come up with some warning

`NODES
  /
    ImuFilter (imu_filter_madgwick/imu_filter_node)
    points_xyzrgb (nodelet/nodelet)
    rviz (rviz/rviz)
    ukf_se (robot_localization/ukf_localization_node)
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)

auto-starting new master
process[master]: started with pid [2726]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bd0676da-f51f-11ed-856f-e454e89a3e91
process[rosout-1]: started with pid [2737]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [2741]
process[camera/realsense2_camera-3]: started with pid [2742]
process[ImuFilter-4]: started with pid [2743]
process[rtabmap/rgbd_odometry-5]: started with pid [2744]
process[rtabmap/rtabmap-6]: started with pid [2745]
process[rviz-7]: started with pid [2746]
process[points_xyzrgb-8]: started with pid [2747]
process[ukf_se-9]: started with pid [2748]
type is rtabmap_ros/point_cloud_xyzrgb
[ WARN] [1684375221.817766447]: Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities
[ INFO] [1684375222.377450865]: Initializing nodelet with 4 worker threads.
[ WARN] [1684375226.690955790]: Failed to meet update rate! Took 0.0088892250000000005983
[ WARN] [1684375226.946884925]: Failed to meet update rate! Took 0.0081668110000000012499
[ WARN] [1684375227.260922892]: Failed to meet update rate! Took 0.008886306000000000066
[ INFO] [1684375227.542745609]: RealSense ROS v2.3.2
[ INFO] [1684375227.542774782]: Built with LibRealSense v2.50.0
[ INFO] [1684375227.542787410]: Running with LibRealSense v2.50.0
[ INFO] [1684375227.763317281]:  
 18/05 10:00:27,811 WARNING [140190250907392] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[ WARN] [1684375227.846936362]: Failed to meet update rate! Took 0.011549232000000001289
[ WARN] [1684375227.847159188]: Failed to meet update rate! Took 0.0084756610000000006527
 18/05 10:00:27,864 WARNING [140190250907392] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[ INFO] [1684375227.941710222]: Device with serial number 135122250381 was found.

[ INFO] [1684375227.941751708]: Device with physical ID 2-1-2 was found.
[ INFO] [1684375227.941772288]: Device with name Intel RealSense D455 was found.
[ INFO] [1684375227.942157821]: Device with port number 2-1 was found.
[ INFO] [1684375227.942181757]: Device USB type: 3.2
[ INFO] [1684375228.052914151]: getParameters...
[ INFO] [1684375228.096425909]: setupDevice...
[ INFO] [1684375228.096449760]: JSON file is not provided
[ INFO] [1684375228.096459669]: ROS Node Namespace: camera
[ INFO] [1684375228.096476345]: Device Name: Intel RealSense D455
[ INFO] [1684375228.096484392]: Device Serial No: 135122250381
[ INFO] [1684375228.096492133]: Device physical port: 2-1-2
[ INFO] [1684375228.096499022]: Device FW version: 05.14.00.00
[ INFO] [1684375228.096506130]: Device Product ID: 0x0B5C
[ INFO] [1684375228.096512993]: Enable PointCloud: On
[ INFO] [1684375228.096519526]: Align Depth: On
[ INFO] [1684375228.096526116]: Sync Mode: On
[ INFO] [1684375228.096569472]: Device Sensors: 
[ INFO] [1684375228.181325452]: Stereo Module was found.
[ INFO] [1684375228.191032219]: RGB Camera was found.
[ INFO] [1684375228.191192615]: Motion Module was found.
[ INFO] [1684375228.191234679]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1684375228.191533811]: Add Filter: pointcloud
[ INFO] [1684375228.192021414]: num_filters: 2
[ INFO] [1684375228.192039740]: Setting Dynamic reconfig parameters.
[... 319 lines cut ...]
[ WARN] (2023-05-18 10:00:42.470) OdometryF2M.cpp:309::computeTransform() Failed to find a transformation with the provided guess (xyz=0.000223,0.000906,-0.000232 rpy=-0.000658,-0.000664,-0.000876), trying again without a guess.
[ WARN] (2023-05-18 10:00:42.520) OdometryF2M.cpp:551::computeTransform() Trial with no guess still fail.
[ WARN] (2023-05-18 10:00:42.520) OdometryF2M.cpp:561::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=3) between -1 and 78" (guess=xyz=0.000223,0.000906,-0.000232 rpy=-0.000658,-0.000664,-0.000876)
[ INFO] [1684375242.520774766]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.108198s
^C[ INFO] [1684375242.693545522]: Odom: quality=78, std dev=0.001109m|0.018867rad, update time=0.169709s
[points_xyzrgb-8] killing on exit
[ukf_se-9] killing on exit
[rtabmap/rtabmap-6] killing on exit
[rviz-7] killing on exit
[rtabmap/rgbd_odometry-5] killing on exit
[camera/realsense2_camera-3] killing on exit
[ImuFilter-4] killing on exit
[ WARN] [1684375242.742895773]: Failed to meet update rate! Took 0.0088882850000000009766
[camera/realsense2_camera_manager-2] killing on exit
[ WARN] (2023-05-18 10:00:42.772) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=8) between -1 and 80"
rtabmap: Saving database/long-term memory... (located at /home/wjyzyw/.ros/rtabmap.db)
rtabmap: 2D occupancy grid map saved.
rtabmap: Saving database/long-term memory...done! (located at /home/wjyzyw/.ros/rtabmap.db, 3 MB)
[rosout-1] killing on exit
^C[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

Is this a normal situation?Or what can I do to get some improvements?Thanks!

MartyG-RealSense, 2023-05-18:
There is very little documentation about the **Failed to meet update rate** warning.  When ROS does generate this error though, it sometimes provides this advice:

"Try decreasing the rate, limiting sensor output frequency, or limiting the number of sensors".

In regard to **Registration failed: "Not enough inliers 0/20**, the Intel RealSense team member who created the RealSense ROS1 wrapper advised adjusting the position of the camera to see whether the inlier warning goes away, and also to provide an "interesting" scene for the camera (a location that has detail in it for the camera to observe and is not too empty).

c56589796, 2023-05-22:
@MartyG-RealSense  Thanks! I will try your advice

MartyG-RealSense, 2023-05-27:
Hi @c56589796  Do you have an update about this case that you can provide, please?  Thanks!

c56589796, 2023-05-29:
@MartyG-RealSense Thanks! I have no question now.

closed 2023-05-29 by c56589796

MartyG-RealSense, 2025-07-08:
Hi @A25310  I do not have knowledge of Isaac Sim, unfortunately.  But when attempting SLAM with Isaac ROS, **Visual SLAM** is often used instead of rtabmap.

https://nvidia-isaac-ros.github.io/concepts/visual_slam/cuvslam/tutorial_realsense.html