realsenseai/realsense-ros #2745: IMU with d434i providing continuous rotational velocity wrt Z axis

janaardhanan, 2023-05-18:
Hi,

I am using RealSense D435i camera for IMU. I have calibrated the camera using the `rs-imu-calibration.py`  program and the following result was achieved in calibration.
```
norm (raw data  ): 9.735610
norm (fixed data): 9.806497 A good calibration will be near 9.806650
```
On checking the IMU with RVIZ using the following launch file and ekf parameters, The camera seems to be providing a continuous and small value of yaw velocity which indicates that the camera is in continuous rotation while physically it is perfectly still without any disturbance.

**Launch File:**
```
<launch>

  <arg name="align_depth"               default="false"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
  <arg name="enable_gyro"               default="true"/>
  <arg name="enable_accel"              default="true"/>

  <arg name="enable_color"              default="false"/>
  <arg name="publish_odom_tf"           default="false"/>  
  <arg name="enable_depth"              default="false"/>

  <arg name="urdf_file" default="$(find realsense2_description)/urdf/test_d435i_camera.urdf.xacro" />
  <param name="robot_description" command="$(find xacro)/xacro --inorder $(arg urdf_file)" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
        type="robot_state_publisher">
    <param name="publish_frequency" value="30.0"/>
  </node>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="enable_color"             value="$(arg enable_color)"/>
    <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>  
    <arg name="enable_depth"             value="$(arg enable_depth)"/>
    <arg name="align_depth"              value="$(arg align_depth)"/>
    <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
    <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
    <arg name="enable_accel"             value="$(arg enable_accel)"/>
  </include>
  
  <node pkg="imu_filter_madgwick" type="imu_filter_node"
      name="imu_filter">
    <param name="use_mag" value="False" />
    <param name="publish_tf" value="False" />
    <param name="fixed_frame" value="odom"/>
    <param name="world_frame" value="enu" />
    <param name="remove_gravity_vector" value="false" /> 

    <remap from="imu/data_raw" to="camera/imu" />
    <remap from="imu/data" to="filtered/imu" />
  </node>

  <node name="ekf_localization" pkg="robot_localization" type="ekf_localization_node" >
    <rosparam command="load" file="$(find bot)/config/imu_ekf.yaml" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find bot)/config/rs_imu.rviz"/>

</launch>
```

**imu_ekf.yaml:**

```
frequency: 30.0
silent_tf_failure: false
sensor_timeout: 0.1
two_d_mode: false
transform_time_offset: 0.0
transform_timeout: 0.0
print_diagnostics: true
debug: false
debug_out_file: /path/to/debug/file.txt
permit_corrected_publication: false
publish_acceleration: false
publish_tf: true

odom_frame: odom            
base_link_frame: base_link  
world_frame: odom           

imu0: filtered/imu
imu0_config: [false, false, false,
              true,  true,  true,
              false, false, false,
              false,  false,  false,
              false,  false,  false]
imu0_nodelay: false

imu0_queue_size: 2

process_noise_covariance: [0.05, 0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0.05, 0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0.06, 0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0.03, 0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0.03, 0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0.06, 0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0.025, 0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0.025, 0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0.04, 0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0.01, 0,    0, 
![Screenshot from 2023-05-18 15-02-55](https://github.com/IntelRealSense/realsense-ros/assets/60310128/28c56c27-270b-493e-9387-db6bc1d85730)
![Screenshot from 2023-05-18 15-04-24](https://github.com/IntelRealSense/realsense-ros/assets/60310128/304f1f78-3d93-41f9-ab29-77933bd5908f)
![Screenshot from 2023-05-18 15-18-27](https://github.com/IntelRealSense/realsense-ros/assets/60310128/0afdc55f-cfd7-4d97-b497-ab15658389e9)
   0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0.01, 0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0.02, 0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0.01, 0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0.01, 0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0.015]

initial_estimate_covariance: [1e-9, 0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    1e-9, 0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    1e-9, 0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    1e-9, 0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    1e-9, 0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    1e-9, 0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    1e-9, 0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    1e-9, 0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    1e-9, 0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    1e-9,  0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     1e-9,  0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     1e-9,  0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     1e-9, 0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    1e-9, 0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    1e-9]
``` 
**Screenshots:**

[image]
[image]
[image]

Please help me out on how to solve this issue.

MartyG-RealSense, 2023-05-18:
Hi @janaardhanan  When a RealSense IMU is uncalibrated, the Y-accel will typically be non-zero even when the camera is stationary on a flat surface.  Applying a calibration to the IMU allows the RealSense SDK to apply correction to the raw data to make it 'fixed'.  

Does adding the command **enable_motion_correction:=true** to your roslaunch instruction improve the stability of the data, please?

janaardhanan, 2023-05-19:
Hi @MartyG-RealSense Thank you for your quick reply.

I have am unable to locate the argument **enable_motion_correction** in any of the launch files. However, I have tried adding the command to my launch command like ```roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:="linear_interpolation" enable_motion_correction:=true``` but there is no difference in the resut.

Additionally I have calibrated the IMU again using `rs-imu-calibration.py` program in the RealSense SDK by altering the following lines
```
if self.imu_sensor.supports(rs.option.enable_motion_correction):
self.imu_sensor.set_option(rs.option.enable_motion_correction, 0)
```
to 
```
if self.imu_sensor.supports(rs.option.enable_motion_correction):
self.imu_sensor.set_option(rs.option.enable_motion_correction, 1)
``` 
which again gave no difference in the output.

The result of calibration is as follows:
```
norm (raw data  ): 9.805112
norm (fixed data): 9.806385 A good calibration will be near 9.806650

```

Kindly let me know where I went wrong and please guide me on which launch file should i add the command **enable_motion_correction:=true**.

MartyG-RealSense, 2023-05-19:
Not all controllable parameters are defined in the launch file by default and yet can still be configured.  **inter_cam_sync_mode** is an examples of this.  If a parameter is supported in the Dynamic Reconfigure interface in ROS1 for setting values during runtime after launch then a parameter with the same name can also be added to a launch file as a **rosparam**.  For example for inter_cam_sync_mode:

```
<rosparam> 
/camera/stereo_module/inter_cam_sync_mode : 1 
</rosparam>
```

Information about the **Dynamic Reconfigure** interface can be found at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#set-camera-controls-using-dynamic-reconfigure-params

There is not an available reference for defining enable_motion_correction as a launch file rosparam but I believe that it would be something like this:

```
<rosparam> 
/camera/motion_module/enable_motion_correction : true 
</rosparam>
```

You should be able to confirm whether enable_motion_correction is a configurable parameter by opening the Dynamic Reconfigure interface with **rosrun rqt_reconfigure rqt_reconfigure** and selecting the **motion_module** category in its side-panel to list the available motion module options.

janaardhanan, 2023-05-19:
Thank you for the clear explanation. 
Although I have made the **enable_motion_correction** parameter true, the results remain the same which can be seen in the image.

Also note that the calibration of IMU has been done, the result of calibration is as follows:
```
[[ 1.00542561 -0.01185965  0.05846351]
 [ 0.01502802  1.00759506 -0.0184912 ]
 [-0.05134756  0.01537133  1.00551046]
 [ 0.08031734  0.005698    0.06446475]]
residuals: [ 1.8379236   1.35883992 11.5456381 ]
rank: 4
singular: [437.37113373 434.84478885 433.98188624  77.44639355]
norm (raw data  ): 9.735610
norm (fixed data): 9.806497 A good calibration will be near 9.806650
Would you like to write the results to the camera? (Y/N)Y
Writing calibration to device.
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  241222077005
  Firmware version:  05.14.00.00
SUCCESS: saved calibration to camera.
Done.

```

[image]

janaardhanan, 2023-05-20:
Hello,
I have tried to observe the topic `/camera/accel/imu_info` but it provides only a single message (top right terminal in the image). But the values in the topic `/camera/imu` seems to fluctuating (left side terminal in the image).

[image]

MartyG-RealSense, 2023-05-20:
What is the fluctuation like if you set unite_imu_method to **copy** instead of linear_interpolation?

janaardhanan, 2023-05-20:
I don't see any big difference after change in unite_imu_method to copy.

[image]

MartyG-RealSense, 2023-05-20:
Which librealsense version are you using, please?  Firmware driver 5.14.0.0 is only compatible with SDK 2.53.1, so if you are using librealsense 2.50.0 or 2.51.1 then you should downgrade the firmware to **5.13.0.50**. 

If you change the Frames setting in RViz from base_link to camera_link, does the fluctuation still occur?

janaardhanan, 2023-05-20:
The versions of firmware and librealsense match correctly as seen in the screenshot. 

There is still no difference when the frame setting is changed from base_link and camera_link.

[image]

MartyG-RealSense, 2023-05-20:
It is important to match ROS wrapper versions as closely as possible with a specific librealsense version listed in the release notes for that particular wrapper version.  

The ROS1 wrapper (final version 2.3.2 as development of it has ceased and it is no longer updated) should ideally use librealsense 2.50.0 or 2.51.1.  As newer librealsense versions are released, incompatiility between them and the 2.3.2 ROS1 wrapper is likely to increase.

What is the stability of the IMU values like when viewing them in the Viewer?

janaardhanan, 2023-05-23:
Hello,

I have downgraded my librealsense version to **2.51.1** and my camera firmware version to **5.13.0.50**. The output still seems to be the same. i have added the bag files of the **motion module** data recorded using **realsense viewer**. Please let me know if the readings are as stable as it needs to be or should it be improved by any methods.

[realsense_bagfile.zip](https://github.com/IntelRealSense/realsense-ros/files/11543430/realsense_bagfile.zip)

janaardhanan, 2023-05-24:
Hi, Thank you for your observation. When I run `realsense2_camera/launch/rs_camera.launch` with IMU options using ROS, There is a constant `angular_velocity: Z` of **around -0.005** which is causing the rotation behavior. Please let me know if this is normal or it needs to be rectified.
I have attached the readings of `rostopic echo /camera/imu`.

[imu_data.txt](https://github.com/IntelRealSense/realsense-ros/files/11551678/imu_data.txt)

MartyG-RealSense, 2023-05-24:
I believe that you could only eliminate this small amount of drift in the gyro Z value if a perfect calibration could be achieved (which is unlikely), as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/4391#issuecomment-515894672

MartyG-RealSense, 2023-05-30:
Hi @janaardhanan  Do you require further assistance with this case, please?  Thanks!

janaardhanan, 2023-06-01:
Hello @MartyG-RealSense.

I have added control parameters and am using message `/cmd_vel` to minimise the same effect of rotation.

Thank you for your assistance. I do not have any further questions.

MartyG-RealSense, 2023-06-01:
It's great to hear that you achieved a solution.  As you do not have further questions, I will close this case.  Thanks very much for the update!

closed 2023-06-01 by MartyG-RealSense