realsenseai/realsense-ros #2747: aligned_depth fps too low

13randNEW, 2023-05-21:
color and depth images fps are around 50-60 fps which is expected. However the aligned depth images fps is about 10 fps. This is too low. I can see error like `error: Resource temporarily unavailable`. Could you help me with this issue? Thank you.
```
started roslaunch server http://ubuntu:40961/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 60
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 95 lines cut ...]
 21/05 19:19:55,704 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:55,755 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:55,807 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:55,858 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:56,010 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:56,061 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:56,213 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:56,264 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:56,416 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:19:56,467 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:21:16,468 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:21:24,532 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:21:24,583 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 21/05 19:21:29,642 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:21:29,693 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:22:23,838 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 21/05 19:22:23,889 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 21/05 19:22:24,941 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/05 19:22:24,992 WARNING [281472376213872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
Platform:
Jetson AGX Orin
Ubuntu 20.04
L4T 35.3.1
Jetpack 5.1.1

MartyG-RealSense, 2023-05-22:
Hi @13randNEW  Do you get better performance with the 2.3.2 ROS1 wrapper if you perform an **rs_rgbd launch instead of an rs_camera one?  

`roslaunch realsense2_camera rs_rgbd.launch`

Before using this launch file, you must install support for an RGBD launch (which has alignment enabled by default) using the instruction below in the Ubuntu terminal:

`sudo apt-get install ros-noetic-rgbd-launch`

13randNEW, 2023-05-22:
Hi, I installed ROS1 Noetic support for an RGBD launch and ran it with `roslaunch realsense2_camera rs_rgbd.launch device_type:=d435 align_depth:=true enable_sync:=true color_fps:=60 color_width:=640 color_height:=480 depth_fps:=60 depth_width:=640 depth_height:=480`. 
The aligned depth is still very slow. Error message still shown.
[image]
```
22/05 14:04:43,272 WARNING [281471730303344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/05 14:04:43,349 WARNING [281471730303344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/05 14:04:43,429 WARNING [281471730303344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/05 14:04:43,479 WARNING [281471730303344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 22/05 14:04:43,509 ERROR [281471384478064] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 22/05 14:08:28,031 WARNING [281471730303344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

13randNEW, 2023-05-25:
Thank you. But I think I installed the librealsense SDK v2.50.0 from the source code with CUDA enabled. `cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install
` What else could possibly affect the fps of aligned depth?

MartyG-RealSense, 2023-05-25:
There is a known issue with alignment on Jetson specifically where FPS can drop very low but perform normally when align_depth = false - see https://github.com/IntelRealSense/realsense-ros/issues/2200 for more information.

The most reliable solution found was to use the older librealsense version **2.43.0** and firmware version **5.12.12.100** or alternatively to use rs_rgbd.launch (which you have already tried).

This issue will not be fixed in the ROS1 wrapper because development of that wrapper has ceased.  There have not been reports of the alignment lag issue occurring with the ROS2 wrapper **ros2-development**.

MartyG-RealSense, 2023-05-30:
hi @13randNEW  Do you require further assistance with this case, please?  Thanks!

13randNEW, 2023-05-30:
Hi @MartyG-RealSense . I was using ros2-development but the fps for color is about 15 when the profile is expected to be 640,480,60. I found ros1 wrapper won't have this problem. However, aligned depth is another problem when I turned to ros1. I am still stuck. When I downgraded to older librealsense version **2.43.0** and firmware version **5.12.12.100**, I got even lower fps for color to be 30 but expecting 60. aligned depth was still ~10 fps.

MartyG-RealSense, 2023-05-31:
As a pointcloud in the ROS wrapper is based on aligned_depth topics, do you get the aligned depth topic if you set pointcloud.enable:=true but do not set align_depth.enable:=true and if you do, what is its FPS?

ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true

********

Alternatively, how does alignment perform if you launch with **ros2 run** instead of ros2 launch?

ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth.enable:=true

closed 2023-06-12 by MartyG-RealSense