realsenseai/realsense-ros #2748: Launch Configuration for multiple cameras

husseinalijaafar, 2023-05-23:
Hi, I was struggling a bit with the following 

I want to launch two cameras connected to the same SBC, an L515 and a D435i. Currently I do:
```
ros2 launch realsense2_camera rs_multi_camera_launch.py camera_name1:=D435i camera_name2:=L515 device_type2:=l5. device_type1:=d4
```
Now I also want to apply parameter such that
- L515 only publishes depth (no RGB)
- D435i only publishes RGB (no depth)

Traditionally I would use enable_depth:=true etc, however, how would it work with the name spaces

husseinalijaafar, 2023-05-23:
It seems the solution was rather straight forward.
```
ros2 param set /D435i/D435i enable_depth false && ros2 param set /L515/L515 enable_color false
ros2 param set /D435i/D435i enable_accel false && ros2 param set /L515/L515 enable_accel false
ros2 param set /D435i/D435i enable_gyro false && ros2 param set /L515/L515 enable_gyro false
```

husseinalijaafar, 2023-05-23:
How would I go about changing the resolution of the cameras?
When setting the parameter after launch, I have this issue: [WARN] [1684869335.279869558] [D435i.D435i]: re-enable the stream for the change to take effect.

husseinalijaafar, 2023-05-30:
HI Marty,
Thank you. I was able to actually just set the resolutions seperately using different local launch files, as it allowed me to configure the parameters easier.

I just made a copy of rs_launch.py and modified it for my needs.

closed 2023-05-30 by husseinalijaafar

MartyG-RealSense, 2023-05-30:
That's great to hear that you achieved a solution.  Thanks very much for the update!