realsenseai/realsense-ros #2750: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically. [D435i]

10179013, 2023-05-25:
D435i、ubutnu18.04
When I run：
`roslaunch realsense2_camera rs_camera.launch \
    unite_imu_method:="linear_interpolation" \
    enable_gyro:=true \
     enable_accel:=true`
There is a warning: 
frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
How can I solve it?

MartyG-RealSense, 2023-05-25:
Hello, you reported this issue 2 weeks ago on another case.  If you are still using librealsense 2.50.0 and firmware driver 5.14.0.0. have you tried the advice at https://github.com/IntelRealSense/realsense-ros/issues/2733#issuecomment-1540111195 to downgrade your firmware, please?  Thanks!

MartyG-RealSense, 2023-05-31:
Do you require further assistance with this case, please?  Thanks!

10179013, 2023-06-01:
Excuse me, please
1. Where can I find out how to downgrade the camera firmware of my d435i from 5.14.0.0 to 5.13.0.50 under ubuntu18.04.
2. I am under ubuntu18.04 and ros1. Is it possible to upgrade librealsense 2.50.0 to librealsense 2.53.1? If so where can I find the steps to do it
Many thanks

MartyG-RealSense, 2023-06-01:
1.  You can downgrade the firmware in the realsense-viewer tool if you have access to it.

Download the Viewer bin file for 5.14.0.0 to your computer from the link below.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

Go to the More option near the top of the Viewer's options side-panel and select Update Firmware from its drop-down menu.

A file navigation box will pop up.  Navigate to the location on your computer where you downloaded the .bin file to and select the file.  The firmware update should now automatically begin.

If you do not have access to realsense-viewer then you can use the downloaded firmware .bin file to downgrade using the **rs-fw-update** tool at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

2.  It is possible to upgrade from 2.50.0 to 2.53.1 but if you are using ROS then I would not recommend it.  This is because development of the RealSense ROS1 wrapper has ceased (ROS2 continues) and so there is not a ROS1 wrapper version designed for 2.53.1.  Instead, the 2.50.0 version that you are using now is the most compatible.

10179013, 2023-06-02:
Sorry to bother you, I have downgraded my camera miner version to version 5.13, but there are still two error reports, what should I do next? (The D435i and data cable I use are all original factory)

` frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.`
` Hardware Notification:Depth stream start failure,1.68569e+12,Error,Hardware Error`

```
smile@smile-X3-S:~$ roslaunch realsense2_camera rs_camera_jason_mapx2_may.launch
... logging to /home/smile/.ros/log/f75cf390-010e-11ee-b420-3c58c26f458c/roslaunch-smile-X3-S-16999.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://smile-X3-S:39739/

SUMMARY
========

PARAMETERS
 * /camera/color_sensor/global_time_enabled: True
 * /camera/motion_module/global_time_enabled: True
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 116 lines cut ...]
 02/06 14:30:37,394 ERROR [140183253391104] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 02/06 14:30:37,414 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,464 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,515 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,565 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,616 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,666 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,717 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,729 ERROR [140182623872768] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131
 02/06 14:30:37,767 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:37,818 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
[... cut ...]
 02/06 14:30:38,273 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:38,526 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
 02/06 14:30:38,576 WARNING [140184158983936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Invalid argument, number: 22
[ WARN] [1685687438.629649960]: Hardware Notification:Depth stream start failure,1.68569e+12,Error,Hardware Error

```

10179013, 2023-06-02:
I have downgrade to [Signed_Image_UVC_5_13_0_50.bin](https://www.intelrealsense.com/download/19295/) and
[Signed_Image_UVC_5_13_0_55.bin](https://www.intelrealsense.com/d400_series_fw_5_13_0_55/), both of them don't work

MartyG-RealSense, 2023-06-02:
The **Invalid argument, number: 22** type of 'control_transfer returned' warning has not been previously reported on this support forum.

Does the problem still occur if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch?

10179013, 2023-06-08:
I add initial_reset:=true , but it still `frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically`
Sorry to bother you again, is there any other way to fix it?

my launch file is:
```
<launch>
  <arg name="emitter_enable"   	  default="0"/>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
[... 289 lines cut ...]
 08/06 16:53:29,373 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:29,424 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:29,475 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:29,525 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:29,662 ERROR [140216090081024] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 08/06 16:53:29,778 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:29,829 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:29,980 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:30,031 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:30,182 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:30,232 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:53:31,328 ERROR [140216090081024] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 08/06 16:53:31,330 ERROR [140216064902912] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131
 08/06 16:53:37,324 ERROR [140216090081024] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 08/06 16:53:38,656 ERROR [140216090081024] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 08/06 16:53:38,660 ERROR [140216064902912] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131
[... cut ...]
 08/06 16:55:51,097 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:56:00,153 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/06 16:56:02,205 WARNING [140216824215296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

MartyG-RealSense, 2023-06-08:
Do the streams publish correctly in RViz after launch despite the hardware clock warning?

It is as not important to have Global Time enabled in ROS as it is in librealsense as ROS provides its own time drift correction, as advised at https://github.com/IntelRealSense/realsense-ros/issues/796#issuecomment-497997643 by the RealSense team member who created the ROS1 wrapper.

closed 2023-06-21 by MartyG-RealSense