realsenseai/realsense-ros #2753: D435i - "/camera/infra, gyro, accel" topics not showing in ROS

robotom, 2023-05-25:
Hi, I went through and installed the ROS1 Realsense package using the method found here: https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

The camera enumerates and I can view the raw stream in rqt. My screen will even rotate if I rotate the camera, so something appears to be working in there. 

My camera is definitely a D435i. It's written on the sticker underneath. I've listed my available topics below: 

<details>
<summary>My 'rostopic list' </summary>

/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/clicked_point
/cmd_vel
/cmd_vel/enabled
/cmd_vel/inCommand/desired
/cmd_vel/teleop/drive
/cmd_vel/teleop/flip
/cmd_vel/teleop/valve
/diagnostics
/initialpose
/joy
/joy/set_feedback
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

</details>

The official documentation says to run this command: `roslaunch realsense2_camera rs_camera.launch` and should then see the below topics: 

<details>
<summary>Expected partial topics list</summary>

    /camera/color/camera_info
    /camera/color/image_raw
    /camera/color/metadata
    /camera/depth/camera_info
    /camera/depth/image_rect_raw
    /camera/depth/metadata
    /camera/extrinsics/depth_to_color
    /camera/extrinsics/depth_to_infra1
    /camera/extrinsics/depth_to_infra2
    /camera/infra1/camera_info
    /camera/infra1/image_rect_raw
    /camera/infra2/camera_info
    /camera/infra2/image_rect_raw
    /camera/gyro/imu_info
    /camera/gyro/metadata
    /camera/gyro/sample
    /camera/accel/imu_info
    /camera/accel/metadata
    /camera/accel/sample
    /diagnostics

</details>

Any ideas? 
Thank you.

MartyG-RealSense, 2023-05-26:
Hi @robotom  The IMU topics (accel and gyro) are disabled by default in the RealSense ROS wrapper.  To enable them, please try the roslaunch command below.

`roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

The unite_imu_method parameter cmbines the two separate gyro and accel topics into a single /imu topic.

robotom, 2023-05-26:
Thanks! Works well! It would appear that  including those arguments after rs_camera.launch means that they will only be available for that session. If I go into the launch file, could I make this change permanent so I don't have to specify those arguments every time I launch the package?  I'm changed those one you mentioned to "true" in the launch file though they have disappeared again from the publisher. Thanks again!

robotom, 2023-05-26:
I was modifying the correct file (in the Catkin Workspace as you say) but foolishly forgot to source my directory. Should really add that to my .bashrc. Works flawlessly now. Pointcloud is working really well too. Thanks for all of your help!

MartyG-RealSense, 2023-05-26:
You are very welcome.  It's great to hear that you were successful  :)

robotom, 2023-05-26:
I did have one other question out of curiosity. Here are some streams that are available in the launch file: 

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

If I enable all four (infra, infra1, infra2 and infra_rgb) none of them show up in rostopic list. But if I leave it as specified above, I get infra1 and infra2 working flawlessly. Am I to assume that this has more to do with the onboard processing and which lenses they're utilizing?

MartyG-RealSense, 2023-05-26:
If you are using infra1 and infra2 then you should not set infra and infra_rgb to true, as infra and infra_rgb are related to streaming RGB color from the left infrared sensor (infra1).  The D435i camera model does not support that feature.

[image]

robotom, 2023-05-31:
Thank you. This is very helpful!

closed 2023-05-31 by robotom