realsenseai/realsense-ros #2754: Lots of "Failed to read busnum/devnum" warnings polluting the terminal

martinakos, 2023-05-26:
I'm running on ros2 humble. I launch the camera with 

`ros2 launch realsense2_camera rs_launch.py `

and I get this output

```
[INFO] [launch]: All log files can be found below /ros_logs/dev/2023-05-26-10-24-49-424022-UK64049-33086
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [33087]
[realsense2_camera_node-1] [INFO] [1685096690.046262963] [camera.camera]: RealSense ROS v4.51.1 [Fn init, Ln 241]
[realsense2_camera_node-1] [INFO] [1685096690.046446139] [camera.camera]: Built with LibRealSense v2.51.1 [Fn init, Ln 242]
[realsense2_camera_node-1] [INFO] [1685096690.046482376] [camera.camera]: Running with LibRealSense v2.51.1 [Fn init, Ln 243]
[realsense2_camera_node-1]  26/05 10:24:50,048 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  26/05 10:24:50,048 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.10.auto
[realsense2_camera_node-1]  26/05 10:24:50,048 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  26/05 10:24:50,048 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.15.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.8.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.14.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.7.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.13.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.6.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.12.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1]  26/05 10:24:50,049 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.11.auto
[realsense2_camera_node-1]  26/05 10:24:50,072 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.10.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.15.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.8.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.14.auto
[realsense2_camera_node-1] [INFO] [1685096690.074825755] [camera.camera]: Device with serial number 040322073861 was found.
[realsense2_camera_node-1]  [Fn getDevice, Ln 91]
[realsense2_camera_node-1] [INFO] [1685096690.074905676] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:1c.0/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb6/6-1/6-1:1.0/video4linux/video4 was found. [Fn getDevice, Ln 94]
[realsense2_camera_node-1] [INFO] [1685096690.074928594] [camera.camera]: Device with name Intel RealSense D435I was found. [Fn getDevice, Ln 96]
[realsense2_camera_node-1] [INFO] [1685096690.075468299] [camera.camera]: Device with port number 6-1 was found. [Fn getDevice, Ln 114]
[realsense2_camera_node-1] [INFO] [1685096690.075494133] [camera.camera]: Device USB type: 3.2 [Fn getDevice, Ln 166]
[realsense2_camera_node-1] [INFO] [1685096690.078390265] [camera.camera]: getParameters... [Fn getParameters, Ln 12]
[realsense2_camera_node-1] [INFO] [1685096690.078915543] [camera.camera]: JSON file is not provided [Fn setAvailableSensors, Ln 81]
[realsense2_camera_node-1] [INFO] [1685096690.078949690] [camera.camera]: Device Name: Intel RealSense D435I [Fn setAvailableSensors, Ln 84]
[realsense2_camera_node-1] [INFO] [1685096690.078971189] [camera.camera]: Device Serial No: 040322073861 [Fn setAvailableSensors, Ln 87]
[realsense2_camera_node-1] [INFO] [1685096690.078989994] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:1c.0/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb6/6-1/6-1:1.0/video4linux/video4 [Fn setAvailableSensors, Ln 91]
[realsense2_camera_node-1] [INFO] [1685096690.079009250] [camera.camera]: Device FW version: 05.13.00.50 [Fn setAvailableSensors, Ln 94]
[realsense2_camera_node-1] [INFO] [1685096690.079021009] [camera.camera]: Device Product ID: 0x0B3A [Fn setAvailableSensors, Ln 97]
[realsense2_camera_node-1] [INFO] [1685096690.079031009] [camera.camera]: Sync Mode: Off [Fn setAvailableSensors, Ln 99]
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.7.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.13.auto
[realsense2_camera_node-1]  26/05 10:24:50,073 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.6.auto
[realsense2_camera_node-1]  26/05 10:24:50,074 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.12.auto
[realsense2_camera_node-1]  26/05 10:24:50,074 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1]  26/05 10:24:50,074 WARNING [139757807978048] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.11.auto
[... cut ...]  other messages I'm interested in. The driver seems to be working normally though. 

Note I'm running humble inside a docker with Ubuntu 22.04 LTS. In my host I have Ubuntu 20.04 with ROS Noetic and if I run 

`roslaunch realsense2_camera rs_camera.launch` 

with the same camera connected I don't get any of the warnings above.

I've checked other issues with this warning message but none of their suggestions worked to eliminate this warning. Do you have any suggestions to eliminate this warning?

MartyG-RealSense, 2023-05-26:
Hi @martinakos  **Failed to read busnum/devnum** is a very rarely occurring warning that has typically been associated with using RealSense on a single-board computer where the USB port is a **USB OTG** micro-sized port instead of a full-sized port.  As you experience this warning with ROS2 Humble but not with ROS Noetic though, this suggests that the problem is not related to your computer hardware.

Do the warnings still occur in Humble if you reset the camera at launch with initial_reset:=true

`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

martinakos, 2023-05-26:
>Do the warnings still occur in Humble if you reset the camera at launch with initial_reset:=true
>ros2 launch realsense2_camera rs_launch.py initial_reset:=true

Yes, they do.

martinakos, 2023-05-26:
I have Noetic installed in ubuntu 20.04 in my laptop (the host). I run ros2 humble  on a docker because it requires ubuntu 22.04 and I can't install it in the host OS.

MartyG-RealSense, 2023-05-26:
There was a Humble case a few weeks ago at https://github.com/IntelRealSense/realsense-ros/issues/2730 where a RealSense ROS user was experiencing the busnum/devnum message.  Setting RViz to use compressed data - see https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-839616982 - improved their performance.

MartyG-RealSense, 2023-05-26:
Would it be possible for you to change the logging level to mute WARNING type messages, or do you need to look for other types of WARNING message when checking the log?

martinakos, 2023-05-30:
I'd prefer not have to mute all warning messages. do you know how to mute that warning message only?

MartyG-RealSense, 2023-05-31:
I have highlighted your question to my Intel RealSense colleagues.  Thanks very much for your patience.

MartyG-RealSense, 2023-06-01:
Hi @martinakos  My colleagues agreed with my suggestion to edit the source code to remove the busnum/devnum message, and added that it would be safe to ignore the warning as you did not find that it impacted the actual performance.

In regard to metadata extraction, they said that you can use the Realsense Viewer’s metadata window to confirm if the desired metadata attribute(s) is supported. or use the following example code for reference.
 
https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp#L71

martinakos, 2023-06-01:
Ok. thanks. 
Regarding the metadata fields I'll replay in the other issue.

MartyG-RealSense, 2023-06-07:
Hi @martinakos  Do you require further assistance with this case, please?

Version 2.54.1 of librealsense is now available at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

The release notes, with an extensive list of enhancements and bug-fixes, are available here:

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2541

martinakos, 2023-06-07:
I'm using the latest and still getting the warning messages. 

```
[INFO] [realsense2_camera_node-1]: process started with pid [32109]
[realsense2_camera_node-1] [INFO] [1686133087.792960565] [d435i.d435i]: RealSense ROS v4.51.1 [Fn init, Ln 241]
[realsense2_camera_node-1] [INFO] [1686133087.793105086] [d435i.d435i]: Built with LibRealSense v2.51.1 [Fn init, Ln 242]
[realsense2_camera_node-1] [INFO] [1686133087.793130792] [d435i.d435i]: Running with LibRealSense v2.51.1 [Fn init, Ln 243]
[realsense2_camera_node-1]  07/06 10:18:07,793 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  07/06 10:18:07,793 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.10.auto
[realsense2_camera_node-1]  07/06 10:18:07,793 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  07/06 10:18:07,793 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.15.auto
[realsense2_camera_node-1]  07/06 10:18:07,793 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
[realsense2_camera_node-1]  07/06 10:18:07,793 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.8.auto
[realsense2_camera_node-1]  07/06 10:18:07,794 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.14.auto
[realsense2_camera_node-1]  07/06 10:18:07,794 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.7.auto
[realsense2_camera_node-1]  07/06 10:18:07,794 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.13.auto
[realsense2_camera_node-1]  07/06 10:18:07,794 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.6.auto
[realsense2_camera_node-1]  07/06 10:18:07,794 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.12.auto
[realsense2_camera_node-1]  07/06 10:18:07,794 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1] [INFO] [1686133087.808945196] [d435i.d435i]: Device with serial number 040322073861 was found.
[realsense2_camera_node-1]  [Fn getDevice, Ln 91]
[realsense2_camera_node-1] [INFO] [1686133087.808999073] [d435i.d435i]: Device with physical ID /sys/devices/pci0000:00/0000:00:1c.0/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb6/6-2/6-2:1.0/video4linux/video4 was found. [Fn getDevice, Ln 94]
[realsense2_camera_node-1] [INFO] [1686133087.809009519] [d435i.d435i]: Device with name Intel RealSense D435I was found. [Fn getDevice, Ln 96]
[realsense2_camera_node-1] [INFO] [1686133087.809183531] [d435i.d435i]: Device with port number 6-2 was found. [Fn getDevice, Ln 114]
[realsense2_camera_node-1] [INFO] [1686133087.809198357] [d435i.d435i]: Device USB type: 3.2 [Fn getDevice, Ln 166]
[realsense2_camera_node-1] [INFO] [1686133087.809208766] [d435i.d435i]: Resetting device... [Fn getDevice, Ln 187]
[realsense2_camera_node-1]  07/06 10:18:07,794 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.11.auto
[realsense2_camera_node-1]  07/06 10:18:07,807 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  07/06 10:18:07,807 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.10.auto
[realsense2_camera_node-1]  07/06 10:18:07,807 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  07/06 10:18:07,807 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.15.auto
[realsense2_camera_node-1]  07/06 10:18:07,808 WARNING [140359120184896] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
[... cut ...]
[realsense2_camera_node-1] [INFO] [1686133094.042594469] [d435i.d435i]: Device physical port: /sys/devices/pci0000:00/0000:00:1c.0/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb6/6-2/6-2:1.0/video4linux/video4 [Fn setAvailableSensors, Ln 91]
[realsense2_camera_node-1] [INFO] [1686133094.042603286] [d435i.d435i]: Device FW version: 05.15.00.02 [Fn setAvailableSensors, Ln 94]
[realsense2_camera_node-1] [INFO] [1686133094.042611088] [d435i.d435i]: Device Product ID: 0x0B3A [Fn setAvailableSensors, Ln 97]
[realsense2_camera_node-1] [INFO] [1686133094.042617809] [d435i.d435i]: Sync Mode: On [Fn setAvailableSensors, Ln 99]
```

closed 2023-06-07 by martinakos