realsenseai/realsense-ros #2756: Calibration of L515 RGB Sensor and AWR1642 Radar

Daya-AWR1642, 2023-05-29:
Hi!
I'm working on sensor fusion project using only L515 RGB Sensor and AWR1642 radar.Using ROS I can launch both sensor but problem lying in calibration of those two sensor.

can you please provide resource for Calibration of L515 RGB Sensor and radar

MartyG-RealSense, 2023-05-29:
Hi @Daya-AWR1642  Intel's guide to inter-connecting multiple cameras in ROS suggests a general method for manually calibrating cameras together using RViz and pointclouds, calibrating one camera at a time.  I have quoted the information below.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

The two cameras are referred to as cam_1 and cam_2.

*******

1.  Set cam_2 in the field view of cam_1.  Keep in RViz display only the point cloud of cam_1 and the axis notation (TF) of cam_2. Now it’s possible to watch the axis notation of cam_2 and change its x,y,z parameters until it’s located right on top of the camera as seen in cam_1’s point cloud.  You can even try and put some distinct object close by and directly in line of sight of cam_2 and try to modify its orientation parameters to align the axis image so it will point to the object as seen in cam_1’s point cloud.

2.  Now, display also the point cloud of cam_2.  Put a board to be seen by both cameras. Modify cam_2’s parameters to rotate the board until it is aligned parallel to the board as seen from cam_1, then translate it, rotate again and so on, iteratively, until they are exactly aligned.  Use the board to understand what parameter to modify and your face for fine tuning.

MartyG-RealSense, 2023-06-06:
Hi @Daya-AWR1642  Do you require further assistance with this case, please?  Thanks!

Daya-AWR1642, 2023-06-06:
may be yeah but not now, some work is going on radar side after that i need to write script for spatial connection of radar and camera at that time I need help. 

As of now I'll close the issue if I need I'll reopen it.

closed 2023-06-06 by Daya-AWR1642