realsenseai/realsense-ros #2758: Realtime Monitoring

hsynklceng, 2023-05-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.13.0.50| 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.0.2                                     | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            | 2.50      | 

### Issue Description
*Question*
I want to make realtime monitoring over network I found some ethernet example in realsense web page , I just cannot decide that can it be runable in global network instead local network? Because I need to reach realsense streams into another IP which is not local in network.

MartyG-RealSense, 2023-05-30:
Hi @hsynklceng  There is a project on the main Intel GitHub site that demonstrates display of a RealSense camera in an HTML5 browser window.

https://github.com/intel/depth-camera-web-demo 

It does not support RealSense 400 Series cameras by default, although the project's Development page explains how to add support for new RealSense camera models and depth and color resolutions not in the demo's default configuration.

https://github.com/intel/depth-camera-web-demo/blob/master/DEVELOPMENT.md 

*********

Alternatively, you could explore the possibility of using a web-based remote access tool such as Google's **Chrome Remote Desktop** to access a distantly located PC with an attached RealSense camera over an internet connection and run programs on that distant PC.

https://remotedesktop.google.com/?pli=1

MartyG-RealSense, 2023-06-06:
Hi @hsynklceng  Do you require further assistance with this case, please?  Thanks!

tugbakara, 2023-06-07:
Actually, I achieved my purpose but I want to send my image as gray because of latency, can I set any param to take gray image over the network in ROS?

MartyG-RealSense, 2023-06-07:
When publishing an RGB image with the RealSense ROS wrapper, the only available format is color RGB8 and not monochrome RGB.  The **infra1** infrared image should be monochrome though and will resemble an RGB image.

closed 2023-06-14 by hsynklceng