realsenseai/realsense-ros #2760: Use of depth_module.emitter_on_off in ROS2 Humble

martinakos, 2023-05-30:
In ROS1 I've been using the **emitter_on_off** feature to be able to get a stereo pair (infra1, infra2) with laser off (for VIO), and a depth image with laser on (for dense depth image based scene reconstruction). I use the rather involved approach explained here https://github.com/IntelRealSense/realsense-ros/issues/2231
The approach involved checking the **frame_laser_power** in /d435i/infra1/metadata to deinterlace the images with laser on/laser off.

Now I'm trying to do the same in ROS2 Humble. I can start the camera with **depth_module.emitter_on_off:=true** and I can see in rviz2 that the infra1 image showing with the laser pattern on/off (although at about 1Hz despite I launched the camera at 60Hz??). Then if I check either the topics /d435i/infra1/metadata or /d435i/depth/metadata I can't find the **frame_laser_power** entry in their **json_data** field.  
How am I supposed to deinterlace the laser on/laser off frames without the **frame_laser_power** entry?

MartyG-RealSense, 2023-05-30:
Hi @martinakos  I will highlight your question to the RealSense ROS team.  Thanks very much for your patience.

MartyG-RealSense, 2023-05-31:
Hi @martinakos  My Intel RealSense colleagues on the ROS team provided the following advice.

********

When “emitter_on_off” is turned on, the IR toggles the laser on and off.  When activated on the Viewer, you can also see it toggle on the metadata.

If the user cannot see it then it may be that they always catch the frames when the laser is off (and the laser power is 0) or maybe they don’t see all the metadata in their “ros2 topic echo” command.  I think it has a character limit.

martinakos, 2023-05-31:
Hi. I'm not sure what you mean with the Viewer. Is that the realsense-viewer? that's not connected to ROS, is it? 

I visualise the metadata using the echo_metadata.py for both the infra1 and the depth topics and there is no **frame_laser_power** field.

martinakos, 2023-05-31:
A minimum example show what metadata I get is:

In one terminal launch:
`ros2 launch realsense2_camera rs_launch.py
`
in another terminal launch:
`rviz2 (and show the depth topic)
`
in another terminal launch:
`rqt  (open the parameter reconfigure plugin and tick on depth_module.emitter_on_off)
`
in another terminal run:
`python3 echo_metadata.py /camera/depth/metadata
`
all the metadata I get is:
```
header:
stamp:
  secs: 1685521804 
  nsecs: 4048384
frame_number:4106      
clock_domain:global_time
frame_timestamp:1685521804004.0483
hw_timestamp:929497443 
time_of_arrival:1685521803978
backend_timestamp:1685521803971
actual_fps:29        
raw_frame_size:814080   
``` 

there is no **frame_laser_power**

My actual camera launch is a bit different than the rs_launch.py but this example is minimum and shows that there is no **frame_laser_power** field in the depth metadata

MartyG-RealSense, 2023-05-31:
Thanks very much @martinakos for the additional information for my colleagues to review.

martinakos, 2023-06-01:
As suggested I've tested the Emitter On Off from the realsense-viewer and show the metadata. See snapshot below:
[image]
I don't see the the **frame_laser_power** field

MartyG-RealSense, 2023-06-01:
I note that the Python instruction that you used to read the metadata topic was:

`python3 echo_metadata.py /camera/depth/metadata `

The metadata section of the ROS2 wrapper's documentation recommends to use this:

`python3 src/realsense-ros/realsense2_camera/scripts/echo_metadada.py /camera/depth/metadata`

Does this result in the same information as the instruction that you use?

The wrapper's metadata script is called **echo_metadada.py** and not echo_metadata.py.

martinakos, 2023-06-01:
yes, it's the same script, I just run it from the same folder and unconsciously corrected the spelling when writing it here.

I find easier to run this:
`ros2 topic echo /d435i/infra1/metadata --field json_data
`
which gives the metadata without truncation:
```
{"frame_number":12931,"clock_domain":"global_time","frame_timestamp":1685628004207.989990,"hw_timestamp":221774058,"time_of_arrival":1685628004195,"backend_timestamp":1685628004188,"actual_fps":58,"raw_frame_size":814080}
```

MartyG-RealSense, 2023-06-02:
My colleagues provided further feedback about the absence of frame_laser_power.  They said that if you cannot find frame_laser_power or frame_laser_power_mode in the RealSense Viewer's metadata overlay, it is an indication of a missing kernel patch (v4l2 backend) to enable these fields.   Applying the kernel patch will enable these fields.  

If you are running an unsupported kernel, you could perhaps try building librealsense from source code with CMake with the RSUSB Backend method so that hardware metadata support is included in the SDK.

If you have a supported kernel version and librealsense is built from packages instead of source code then the kernel patch is already included in the packages and so hardware metadata support is automatically provided.

Once these fields are available in the metadata, it will show up in the ROS2 message.

martinakos, 2023-06-05:
I'm using the kernel version 5.15.0-72-generic, which it's supported by librealsense2.  As I mentioned, I'm testing this in ros2 humble inside a docker container. librealsense2 has been installed from packages, including the librealsense2-dkms package, from the  Debian repositories. version 2.53.1. The physical port used by the camera shows it's using video4linux. 

In the host I have exactly the same versions for kernel, librealsense, installed from Debian repos, and physical port used by the camera (showing it's using video4linux). However, in the host I can see the  frame_laser_power field in the metadata and inside the docker I don't see that field in the metadata. 

One difference is that I'm using ros1 in the host with the legacy realsense2_camera and in the container I'm using ros2 with the latest realsense2_camera package.  Is there any other difference that could account for not seeing that metadata field inside the container?

MartyG-RealSense, 2023-06-05:
The advice of my colleagues is that the frame_laser_power metadata should be available if metadata support is present, and it will be enabled if you have built librealsense from packages.  The key difference in your case is that ROS1 (where it works) is running in the host and ROS2 (where it doesn't work) is running in the docker container.

Have you tested ROS2 on a host to confirm whether or not it is a container related issue?

closed 2023-06-06 by martinakos

MartyG-RealSense, 2023-06-06:
I will pass your feedback to my colleagues who advised on your case.  Thanks very much for the update!