realsenseai/realsense-ros #2761: Why namespace is fixed as node_name

hyunseok-yang, 2023-06-01:
I'd like to apply namespace in realsense node, but it fixed a namesapce as a node name by default.
Is this right apporach...?
Or  if there is a any intention?

MartyG-RealSense, 2023-06-01:
Hi @hyunseok-yang  Are you trying to set a custom namespace for your camera in your launch of the RealSense ROS wrapper?  If you are then you can do this by setting the **camera** parameter in the ROS1 wrapper:

`roslaunch realsense2_camera rs_camera.launch camera:=cam_1`

For ROS2 I believe that it would be something like this:

`ros2 launch realsense2_camera rs_launch.py camera_name1:=cam1`

If you are trying to do something other than set a namespace in the launch then please provide further information.  Thanks!

hyunseok-yang, 2023-06-02:
> Hi @hyunseok-yang Are you trying to set a custom namespace for your camera in your launch of the RealSense ROS wrapper? If you are then you can do this by setting the **camera** parameter in the [...]

Hi, 
I think this should be modified like this.
`namespace=LaunchConfiguration("camera_name"),` ==> `namespace=LaunchConfiguration("namespace"),`

```python
          launch_ros.actions.Node(
                package='realsense2_camera',
                namespace=LaunchConfiguration("namespace") ,
                name=LaunchConfiguration("camera_name"),
```

MartyG-RealSense, 2023-06-02:
Thanks very much @hyunseok-yang 

The only example that I have of setting camera_name in ROS2 that I have, which I based my own suggestion on, is the launch filr **rs_multi_camera_launch.py**

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_multi_camera_launch.py

hyunseok-yang, 2023-06-03:
> Thanks very much @hyunseok-yang
> 
> The only example that I have of setting camera_name in ROS2 that I have, which I based my own suggestion on, is the launch file **rs_multi_camera_launch.py**
> 
> https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_multi_camera_launch.py

All I need was actually topic names from realsensenso node as `/{namespace}/{camera_name}/blah~blah`.

So I would like to sugget like in new PR(https://github.com/IntelRealSense/realsense-ros/pull/2764).

MartyG-RealSense, 2023-06-03:
Thanks so much @hyunseok-yang for your PR!

As this issue is associated with your PR, I have added an **Enhancement** label to it so that it is kept open whilst your PR is active.  Thanks again!

hyunseok-yang, 2023-06-04:
> Thanks so much @hyunseok-yang for your PR!
> 
> As this issue is associated with your PR, I have added an **Enhancement** label to it so that it is kept open whilst your PR is active. Thanks again!

Do I need to assign someone else for that PR?

MartyG-RealSense, 2023-06-04:
No you don't need to assign reviewers or assign yourself.  The Intel RealSense developer team should assign a reviewer themselves at a future date.

hyunseok-yang, 2023-06-05:
> No you don't need to assign reviewers or assign yourself. The Intel RealSense developer team should assign a reviewer themselves at a future date.

Ok Thanks! have a good day!

MartyG-RealSense, 2026-10-31:
Case closed due to the PR https://github.com/realsenseai/realsense-ros/pull/2764 linked to this issue having been merged in August 2023.

closed 2026-10-31 by MartyG-RealSense