realsenseai/realsense-ros #2762: D405 raises error for fps >30: [realsense2_camera_node-1]  01/06 12:00:06,720 ERROR [281471864519072] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

nidhicj, 2023-06-01:
Hi,

I ran into an issue: 
[image]

This is when I am using D405 camera with Jetson Orin - Auvidea Carrier board. I ran a basic ros2 launch.py file that can changed according our project but after much editing it is quite same as realsense ros2 launch.py file.
The problem arises when I increase the fps to 90. Whenever fps >30, the uvc-streamer error pops up. For the application I am working on, fps cannot be reduced, ideally I need it at 90. I tried to troubleshoot by cooling down the camera and the jetson orin - Auvidea carrier board. The error persists on depth module more than rgb images. I am using the official 1m cable from realsense and did not see errors due to cable.
Have enabled depth_module.enable_auto_exposure as well. Have gone through issue: [https://github.com/IntelRealSense/realsense-ros/issues/2486](url)

Details about the hardware being used:
Linux: 
Description: Ubuntu 20.04.6 LTS
Release: 20.04
Codename: focal
PC: Jetson Orin X230D- Auvidea Carrier board
rosdistro: humble
camera: Intel Realsense D405
RealSense ROS wrapper:  v4.51.1
Built with LibRealSense SDK:  v2.51.0
Realsense camera firmware: v5.13.0.50

How to tackle this problem?
Any help is great thanks :)

MartyG-RealSense, 2023-06-01:
Hi @nidhicj  Do the depth and color streams publish correctly despite the ERROR messages?  If they do then you could consider muting the ERROR messages by changing the logging level.  For example, using the following command in the Ubuntu terminal:

`export LRS_LOG_LEVEL="WARNING"`

nidhicj, 2023-06-01:
Hi @MartyG-RealSense , No, after checkking the /diagnostics topic, the actual frequency of the depth module goes to 0
[image]

MartyG-RealSense, 2023-06-01:
Does the same problem occur if launching using the instruction below?

`ros2 launch realsense2_camera rs_launch.py enable_sync:=true rgb_camera.profile:=640,480,90 depth_module.profile:=640,480,90`

nidhicj, 2023-06-01:
Hi @MartyG-RealSense , 

Yes. I tweaked with resolutions and fps both. Additional point to add is that I was running rviz2 in another terminal to view the /image topics. I noticed the actual frequency from the /diagnostics topic goes down only when rviz is running. Is that connected somehow would you say? 
Like I go this error:
[realsense2_camera_node-1]  01/06 15:02:48,037 ERROR [281472390539680] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
only when I was running rviz on another terminal. After killing rviz, the actual frequency gained back in the range of 90 fps.

Let me know if you find any observations if rviz is the culprit

Thanks

MartyG-RealSense, 2023-06-01:
I have handled some cases in the past where streams perform normally til RViz is opened and then they slowed down greatly. 

Are you launching RViz with 'ros2 run rviz rviz'.  If you are, have you tried launching RViz2 with the command **"ros2 run rviz2 rviz2"** to see whether the FPS still slows?

nidhicj, 2023-06-02:
I have not particularly used this command but just rviz2 in ros2 environment.

Turns out rviz when operating from an external laptop (ssh into jetson), was creating an issue. But still rviz slowed down the operation :/

Thanks

MartyG-RealSense, 2023-06-02:
If using RViz over ssh causes you a problem then you could investiate whether something called **X11 Forwarding** is slowing RViz down, as discussed at the link below.

https://answers.ros.org/question/323569/is-there-a-way-to-fix-rviz-lag/

Apparently, adding -x to your ssh command disables X11 forwarding, such as **ssh -x**

closed 2023-06-13 by MartyG-RealSense