realsenseai/realsense-ros #2763: How should I trigger the D405 and the D435i synchronously in ros1？

kao-yang, 2023-06-02:
I have read white paper of [Multi-Camera configurations](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration),but the D405 cam seems no synchronous interface.

MartyG-RealSense, 2023-06-02:
Hi @kao-yang  You are correct, the D405 camera model does not have a set of hardware sync pins, unfortunately.  

On RealSense 400 Series cameras such as D405 and D435i, **Global Time** is enabled by default though and it will calculate a common timestamp between multiple cameras.  Please see https://github.com/IntelRealSense/librealsense/pull/3909 for further details.  

Global Time requires hardware metadata support to be enabled.  It is enabled if building the librealsense SDK from packages or from source code with CMake with the **RSUSB backend** installation method.  If building from source code and not using the RSUSB installation method, a kernel patch script should be applied to add hardware metadata support.

MartyG-RealSense, 2023-06-08:
Hi @kao-yang  Do you require further assistance with this case, please?  Thanks!

kao-yang, 2023-06-09:
Thanks.As d405 can not triggered with d435i synchronously,I may do this in program.

MartyG-RealSense, 2023-06-09:
Thank you.  Do you need help with developing such a program, please?

closed 2023-06-16 by MartyG-RealSense