realsenseai/realsense-ros #2765: After my D435i starts roslaunch, it will frequently cause ubuntu18.04 to freeze (you can only force restart the computer)

10179013, 2023-06-03:
Because I need to use binocular, color, IMU. And the image needs 640 480, is it caused by the roslaunch file I configured? How should I fix it?

The roslaunch file 、 the terminal prompt after running and my ubuntu system error log are as follows

1.roslaunch file 
```
<launch>
  <arg name="emitter_enable"   	  default="0"/>

  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
[... 256 lines cut ...]
[ INFO] [1685692493.910809429]: Expected frequency for infra1 = 30.00000
[ INFO] [1685692493.921934292]: Expected frequency for infra2 = 30.00000
[ INFO] [1685692493.932607205]: Expected frequency for color = 30.00000
[ INFO] [1685692493.943062017]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1685692493.953237958]: Start publisher IMU
[ INFO] [1685692493.954282338]: setupStreams...
 02/06 15:54:54,195 WARNING [140539061044992] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1685692494.273598483]: 
[ WARN] [1685692494.273778606]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 02/06 15:54:54,379 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1685692494.429806314]: SELECTED BASE:Depth, 0
[ INFO] [1685692494.443155212]: RealSense Node Is Up!
 02/06 15:54:54,608 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:54:54,658 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:54:54,910 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:54:54,960 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:54:55,011 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:54:55,163 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:54:55,314 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/06 15:54:55,365 WARNING [140538829330176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

3.error
[image]

MartyG-RealSense, 2023-06-03:
We can test whether there is a problem in your custom launch file by launching with rs_camera.launch instead and setting your parameters in a long launch instruction instead of inside the file.  These parameters will override the ones that are set in the launch file.

`roslaunch realsense2_camera rs_camera.launch emitter_enabled:=true align_depth:=true filters:=pointcloud enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 infra1_width:=640 infra1_height:=480 infra1_fps=30 infra2_width:=640 infra2_height:=480 infra2_fps=30`

If your D435i was manufactured before April 2022 then use 250 for accel and 400 for gyro.  

`accel_fps:=250 gyro_fps:=400`

If it was manufactured after April 2022 then use 200 for Accel and 400 for Gyro.  

`accel_fps:=200 gyro_fps:=400`

This is because the IMU component in D435i and D455 was changed after April 2022 and supports different Accel and Gyro values to the older units.

A quick way to check supported IMU FPS values on your particular camera is to use the RealSense Viewer tool and look at the listed FPS for Accel and Gyro in the Motion Module category of the Viewer's options side-panel.  The serial number of the camera can also give a clue.  Units manufactured in 2020 start with '0', 2021 units start with '1' and 2022 will start with '2', etc

10179013, 2023-06-05:
sorry to bother you again.
My d435i was produced in 2018, so I executed the following command, and I encountered some problems and need to ask you：
1. Now the `/camera/imu` frequency is 400, But I need 200， I know I can change it by ros tool. Could we just let  /camera/imu frequency be published as 200?
2. How can I solve the `frame's time domain is HARDWARE_CLOCK? Timestamps may reset periodically.`I have downgraded the d435i framework to suit my realsenceros, but it doesn't work.
3. How can I solve the` control_transfer returned an error, index: 768, error: Resource temporarily unavailable, number: 11`, what is this problem? How to fixed it?
4. My` /camera/depth/image_rect_raw `does not work normally, and I can’t see it under rviz. How should I start it?

`roslaunch realsense2_camera rs_camera.launch emitter_enabled:=true align_depth:=true filters:=pointcloud enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 infra1_width:=640 infra1_height:=480 infra1_fps:=30 infra2_width:=640 infra2_height:=480 infra2_fps:=30 accel_fps:=250 gyro_fps:=400`

[image]

full terminal prompt：

```
smile@smile-X3-S:~$ roslaunch realsense2_camera rs_camera.launch emitter_enabled:=true align_depth:=true filters:=pointcloud enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30 infra1_width:=640 infra1_height:=480 infra1_fps:=30 infra2_width:=640 infra2_height:=480 infra2_fps:=30 accel_fps:=250 gyro_fps:=400
... logging to /home/smile/.ros/log/2eca6534-0343-11ee-aaaa-3c58c26f458c/roslaunch-smile-X3-S-580.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://smile-X3-S:36973/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
[... 171 lines cut ...]
 05/06 09:58:21,803 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/06 09:58:26,856 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/06 09:58:48,915 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/06 09:58:54,968 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/06 09:59:24,031 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 05/06 10:03:03,922 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/06 10:03:19,980 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 05/06 10:03:27,033 WARNING [140578123724544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

closed 2023-06-16 by MartyG-RealSense